espressif/arduino-esp32 #11187: Ethernet Events doesn't call callback function

sergshadrap, 2025-03-27:
### Board

WT32-Eth01 Ethernet module

### Device Description

Just plain board WT32-Eth01 Ethernet. 

### Hardware Configuration

WT32-Eth01 Ethernet board with Ethernet cable connected. Powered over FTDI232 from laptop USB.

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921600

### Description

I use Ethernet connection for my library, it represents different kind of devices with Ethernet connection. I used to using WT32-Eth01 board for this purpose. Previously I used the v 3.0.7 release and everything was perfect (tested on multiple devices) , recently I've tried  to move for 3.1.3 release and seems that the Events callback function stop working.  There is nothing special I registered callback function over intermediate function (due to library usage) . The registration process says "true" , but no call for the function.  The system Events goes fine and I see that the board is initialized with Ethernet and got an IP. Seems like a brake between system and application layer.

### Sketch

```cpp
if(!networkEvents.onEvent(MppServer::onEventStatic)) Serial.println("ETH Event function hasn't registered!");
        else Serial.println("ETH Event function registered!"); // That is callback intermediate registration   of onEventStatic() .

void MppServer::onEventStatic(arduino_event_id_t event) { 
   mppserver.MyonEvent(event); 
} // That is my static function 

 void MppServer::MyonEvent(arduino_event_id_t event) { 
      if (event == ARDUINO_EVENT_ETH_GOT_IP) {
...............
}  // that is real function
```

### Debug Message

```plain
20:51:43.442 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
20:51:43.442 -> configsip: 0, SPIWP:0xee
20:51:43.442 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:51:43.442 -> mode:DIO, clock div:1
20:51:43.442 -> load:0x3fff0030,len:4916
20:51:43.442 -> load:0x40078000,len:16492
20:51:43.442 -> load:0x40080400,len:4
20:51:43.442 -> load:0x40080404,len:3524
20:51:43.442 -> entry 0x400805b8
20:51:44.047 -> [     1][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400e8140
20:51:44.047 -> [    12][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400e8110
[... 122 lines cut ...]
20:51:46.280 -> INTERNAL Memory Info:
20:51:46.280 -> ------------------------------------------
20:51:46.280 ->   Total Size        :   368180 B ( 359.6 KB)
20:51:46.280 ->   Free Bytes        :   285332 B ( 278.6 KB)
20:51:46.280 ->   Allocated Bytes   :    71808 B (  70.1 KB)
20:51:46.280 ->   Minimum Free Bytes:   284728 B ( 278.1 KB)
20:51:46.280 ->   Largest Free Block:   110580 B ( 108.0 KB)
20:51:46.280 -> ------------------------------------------
20:51:46.280 -> GPIO Info:
20:51:46.280 -> ------------------------------------------
20:51:46.280 ->   GPIO : BUS_TYPE[bus/unit][chan]
20:51:46.280 ->   --------------------------------------  
20:51:46.280 ->      0 : ETHERNET_CLK
20:51:46.280 ->      1 : UART_TX[0]
20:51:46.280 ->      3 : UART_RX[  2566][V][NetworkInterface.cpp:78] _onIpEvent(): eth0 Got New IP: 192.168.1.19 MASK: 255.255.255.0 GW: 192.168.1.1
20:51:46.280 -> [0][  2576][V][NetworkEvents.cpp:117] _checkForEvent(): Network Event: 5 - ETH_GOT_IP
```

### Other Steps to Reproduce

If only step down back to 3.0.7 without changing the code the events works fine.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-03-27:
Use `Network.onEvent(...)`

sergshadrap, 2025-03-27:
Thanks.....   Could you be so kind a bit clear ? I see the class : class NetworkEvents and the 3 functions for registering callback inside.  And I made the instance in my header like NetworkEvents networkEvents;  I don't see any Network class this library..

me-no-dev, 2025-03-27:
`Network` is the instance on `NetworkManager` which implements `NetworkEvents`, so `NetworkEvents` does not need/have it's own instance.

sergshadrap, 2025-03-27:
Thanks. That is clear.