espressif/arduino-esp32 #11189: Zigbee Pressure EP crashes on report()

DarkShortie, 2025-03-27:
### Board

ESP32-C6-WROOM-1

### Device Description

ESP32-C6-DevKitC-1

### Hardware Configuration

i2c sensors planned but removed for testing

### Version

other

### IDE Name

VSCode

### Operating System

Arch Linux

### Flash frequency

40 Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

I'm trying to create a Zigbee ED with different sensors and endpoints. Temperature and humidity worked fine, but when I tried to add a new endpoint for Pressure it started to crash on zbPressureSensor.report().

The provided sketch is stripped down to the bare minimum to reproduce the issue. So all sensors are removed from the DevKit and also from the code and a static value is used.

### Sketch

```cpp
#ifndef ZIGBEE_MODE_ED
#error "Zigbee end device mode is not selected in platformio.ini"
#endif

#include <Arduino.h>
#include <Zigbee.h>

#define FACTORY_RESET_PIN BOOT_PIN

// Zigbee endpoints
#define PRESSURE_SENSOR_ENDPOINT_NUMBER 10

ZigbeePressureSensor zbPressureSensor = ZigbeePressureSensor(PRESSURE_SENSOR_ENDPOINT_NUMBER);

static int16_t zb_float_to_s16(float temp) {
  return (int16_t)(temp * 100);
}

void measureAndSleep() {
  int measure_start = millis();

  zbPressureSensor.setPressure(1014);
  zbPressureSensor.report();

  Serial.println(String("Measurement took ") + (millis() - measure_start));
[... 226 lines cut ...]
4081f360: 0x00000168 0xabba1234 0x0000015c 0x4081f0a0 0x000001fd 0x40814784 0x40814784 0x4081f36c
4081f380: 0x4081477c 0x00000018 0x40817a28 0x40817a28 0x4081f36c 0x00000000 0x00000001 0x4081d35c
4081f3a0: 0x706f6f6c 0x6b736154 0x00000000 0x00000000 0x4081f350 0x00000005 0x00000000 0x00000001
4081f3c0: 0x00000001 0x00000000 0x00000000 0x00018e37 0x00000000 0x40817464 0x408174cc 0x40817534
4081f3e0: 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x4081f874 0x4204949c 0x00000000
4081f400: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081f420: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081f440: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081f460: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081f480: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081f4a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081f4c0: 0x00000000 0x00000000 0xbaad5678 0x00000068 0xabba1234 0x0000005c 0x00000000 0x4081f4d8
4081f4e0: 0x00000000 0x00000000 0x00000000 0x4081f4f0 0xffffffff 0x4081f4f0 0x4081f4f0 0x00000000
4081f500: 0x4081f504 0xffffffff 0x4081f504 0x4081f504 0x00000001 0x00000001 0x00000000 0x6600ffff
4081f520: 0x00000000 0x74657320 0x206f7401 0x65707974 0x52415520 0xbaad5678 0x00000168 0xabba1234
4081f540: 0x0000015c 0x4081f5a0 0x4081f5a0 0x4081f6a0 0x4081f69f 0x00000000 0x4081f55c 0xffffffff

ELF file SHA256: 8bccbf631

Rebooting...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2025-03-27:
@DarkShortie What version of Arduino ESP32 Core are you using? You choose "other" in the issue description.

P-R-O-C-H-Y, 2025-03-27:
I did test the example you provided in the description and I am not able to reproduce the crash. Can you update to latest dev version 3.2.0-rc2 or wait a bit for 3.2.0 stable release. This issue is already fixed.

closed 2025-03-27 by P-R-O-C-H-Y