espressif/arduino-esp32 #11200: RMT: RMTLoopback example hangs and gives core error

fanfanlatulipe26, 2025-03-28:
### Board

ESP32 Dev Kit v1

### Device Description

ESP32 Dev Kit v1

### Hardware Configuration

Jumper between pin 18 and 21

### Version

v3.1.2

### IDE Name

Arduino IDE 1.8.19

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

In fact I use the very last version of ESP32: 3.2.0  (release not yet available in the dropdown list here above)

The standard RMTLoopback  example fails with an error and hangs
`E (1190) rmt: rmt_receive(386): partial receive not supported
`   
If the example is compiled with debug level None we just get:  
`real tick set to: 100ns   Please connect GPIO 18 to GPIO 21, now.  E (11) rmt: rmt_receive(386): partial receive not supported
`
If we compile with debug level Error, we get more output from the testcase as can be seen in the "Debug Message" section

As can be seen in the output of the test, the results are not exactly the one expexted, but not really garbage. Some shift ??

More: the frequency used in the example is 10 000 000
`#define RMT_FREQ               10000000  // tick time is 100ns
`
In fact the generated waveform shows that it is not 100ns, and the wave form is not exactly the one given by the data array.
I found little documentation and the tick time is not at all always as described in the comment in esp32-hal-rmt.h for rmtInit

`New Parameters in Arduino Core 3: RMT tick is set in the rmtInit() function by the
    frequency of the RMT channel. Example: 100ns tick => 10MHz, thus frequency will be 10,000,000 Hz
`
Seems OK for 1Mhz, 2Mhz,8Mhz, but bad for 5Mhz, 10Mhz.

### Sketch

```cpp
The example given in for ESP32 RMT: RMTLoopback
No change.
```

### Debug Message

[... 126 lines cut ...]
000a8004=8004000b 00098005=8005000a 00088006=80060009 00078007=80070008 
00068008=80080007 00058009=80090006 0004800a=800a0005 0003800b=800b0004 
0002800c=800c0003 000e800d=800d0002 000d8001=8001000e 000c8002=8002000d 
000b8003=8003000c 00008004=8004000b 00000000=8005000a 00000000=80060009 
00000000=80070008 00000000=80080007 00000000=80090006 00000000=800a0005 
00000000=800b0004 00000000=800c0003 00000000=800d0002 00000000=8001000e 
00000000=8002000d 00000000=8003000c 00000000=8004000b 00000000=8005000a 
00000000=80060009 00000000=80070008 00000000=80080007 00000000=80090006 
00000000=800a0005 00000000=800b0004 00000000=800c0003 00000000=800d0002 
00000000=8001000e 00000000=8002000d 00000000=8003000c 00000000=8004000b 
00000000=8005000a 00000000=80060009 00000000=80070008 00000000=80080007 

[  1570][V][esp32-hal-rmt.c:377] _rmtRead(): GPIO: 21 - Request: 30 RMT Symbols - Non-Blocking - Timeout: 0
E (1403) rmt: rmt_receive(386): partial receive not supported
[  1580][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 18 - Request: 30 RMT Symbols - Blocking - Timeout: -1
[  1594][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 18 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
```

### Other Steps to Reproduce

I tried with other GPIOs.  Same results

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2025-03-30:
IDF 5.4 has introduced a new flag into Receive Config Struct, causing it to fail when set to 1 when using ESP32.

To make sure that such kind of failure won't happen in any other place, a PR (#11203) to clear all RMT config structs before using them has been posted. This shall fix this issue. 

@fanfanlatulipe26 - If possible, please test the PR. Thanks!

fanfanlatulipe26, 2025-03-31:
I ran the example with the new esp32-hal-rmt.c. 
No more error messages, but I got the same strange results as the ones I posted above.
I would expect send and receive data to be identical. 
```
real tick set to: 100ns

Please connect GPIO 18 to GPIO 21, now.
Got 30 RMT symbols
000d8001=8001000e 000c8002=8002000d 000b8003=8003000c 000a8004=8004000b 
00098005=8005000a 00088006=80060009 00078007=80070008 00068008=80080007 
00058009=80090006 0004800a=800a0005 0003800b=800b0004 0002800c=800c0003 
000e800d=800d0002 000d8001=8001000e 000c8002=8002000d 000b8003=8003000c 
000a8004=8004000b 00098005=8005000a 00088006=80060009 00078007=80070008 
00068008=80080007 00058009=80090006 0004800a=800a0005 0003800b=800b0004 
0002800c=800c0003 000e800d=800d0002 000d8001=8001000e 000c8002=8002000d 
000b8003=8003000c 00008004=8004000b 00000000=8005000a 00000000=80060009 
00000000=80070008 00000000=80080007 00000000=80090006 00000000=800a0005 
00000000=800b0004 00000000=800c0003 00000000=800d0002 00000000=8001000e 
00000000=8002000d 00000000=8003000c 00000000=8004000b 00000000=8005000a 
00000000=80060009 00000000=80070008 00000000=80080007 00000000=80090006 
00000000=800a0005 00000000=800b0004 00000000=800c0003 00000000=800d0002 
00000000=8001000e 00000000=8002000d 00000000=8003000c 00000000=8004000b 
00000000=8005000a 00000000=80060009 00000000=80070008 00000000=80080007 

Got 30 RMT symbols
000d8001=8001000e 000c8002=8002000d 000b8003=8003000c 000a8004=8004000b 
00098005=8005000a 00088006=80060009 00078007=80070008 00068008=80080007 
00058009=80090006 0004800a=800a0005 0003800b=800b0004 0002800c=800c0003 
000e800d=800d0002 000d8001=8001000e 000c8002=8002000d 000b8003=8003000c 
000a8004=8004000b 00098005=8005000a 00088006=80060009 00078007=80070008 
00068008=80080007 00058009=80090006 0004800a=800a0005 0003800b=800b0004 
0002800c=800c0003 000e800d=800d0002 000d8001=8001000e 000c8002=8002000d 
000b8003=8003000c 00008004=8004000b 00000000=8005000a 00000000=80060009 
00000000=80070008 00000000=80080007 00000000=80090006 00000000=800a0005 
00000000=800b0004 00000000=800c0003 00000000=800d0002 00000000=8001000e 
00000000=8002000d 00000000=8003000c 00000000=8004000b 00000000=8005000a 
00000000=80060009 00000000=80070008 00000000=80080007 00000000=80090006 
00000000=800a0005 00000000=800b0004 00000000=800c0003 00000000=800d0002 
00000000=8001000e 00000000=8002000d 00000000=8003000c 00000000=8004000b 
00000000=8005000a 00000000=80060009 00000000=80070008 00000000=80080007 

Got 30 RMT symbols
000d8001=8001000e 000c8002=8002000d 000b8003=8003000c 000a8004=8004000b 
00098005=8005000a 00088006=80060009 00078007=80070008 00068008=80080007 

```

SuGlider, 2025-03-31:
>I would expect send and receive data to be identical.

I'll take the first data as example `000d8001=8001000e`:
`000d8001` can be broken into 2 data blocks: `000d` and `8001` those are read as `8001` and `000e`.
there is an extra +1 on `000d` = `000e`. This is a timming detection shift of the end of the transmission of 1 tick.
If more data is sent at once, this ' tick difference will be seen only in the very last bit.

fanfanlatulipe26, 2025-03-31:
Word on the left of "=" is the received RMT symbol, and on the left the sent one.
The received symbol seems to have the 2 16bit words swapped ???
I have a look with a logic analyzer to the generated waveform and discover that the low part of the pulse correspond in fact to the low part of the next symbol .
I used a low frequency because 10Mhz is too much and the tick is not 100ns, and also bigger pulses.
Here are the changes I made:
```
//#define RMT_FREQ               10000000  // tick time is 100ns
#define RMT_FREQ               1000000  // tick time is 1us
```
and
```
// rmtSetRxMaxThreshold(RMT_RX_PIN, 20);
    rmtSetRxMaxThreshold(RMT_RX_PIN, 30000);
```
and 
```
 for (i = 0; i < 255; i++) {
 //  data[i].val = 0x80010001 + ((i % 13) << 16) + 13 - (i % 13);
   data[i].val = 0x80010001 + (((i*4) % 16) << 16) + 16 - ((i*4) % 16);
    my_data[i].val = 0;
  }
```
And the result
```
Got 30 RMT symbols
000d8001=80010011 00098005=8005000d 00058009=80090009 0011800d=800d0005 
000d8001=80010011 00098005=8005000d 00058009=80090009 0011800d=800d0005 
...

First pulse in the burst should be 1us H / 17usL,  second 5usH / 13usL, third 9usH / 9usL

![Image](https://github.com/user-attachments/assets/53e30bad-b556-486f-8ec0-9095274a1a2e)

fanfanlatulipe26, 2025-03-31:
Sorry, I didn't manage to past the waveform ...
We can see:
first pulse  1usH/13L                 in fact 13usL is the low part of next RMT symbol for low part
second 5usH/9usL
third 9usH/5usL
forth 13H/17L

fanfanlatulipe26, 2025-03-31:
[image]

SuGlider, 2025-03-31:
>000d8001=80010011 00098005=8005000d 00058009=80090009 0011800d=800d0005 
>000d8001=80010011 00098005=8005000d 00058009=80090009 0011800d=800d0005

Sent pulses:
8001 + 0011 ==> both are High, first for 0x4000 ticks (15bits = 16384 ticks) and the second for 1 tick high
8005 + 000d ==> both are High, first for 0x4002 ticks and the second for 6 ticks high
8009 + 0009 ==> both are High, first for 0x4003 ticks and the second for 4 ticks high
800d + 0005 ==> both are high
all are High.  I don't see any RMT Symbol even (last bit 0), which would be low level.

fanfanlatulipe26, 2025-03-31:
80010011   8001  0011  1 tick high, 17 ticks low
8005000d                       5 ticks high, 13 low                            and it is what we see (more or less for the low part ...) on the waveform
The bit and byte numbering and order is very confusing in the documentation ...

SuGlider, 2025-03-31:
> 80010011 8001 0011 1 tick high, 17 ticks low 8005000d 5 ticks high, 13 low and it is what we see (more or less for the low part ...) on the waveform The bit and byte numbering and order is very confusing in the documentation ...

[image]

TRM shows the bits in the reverse order, from LSB to MSB, but the bits are not reversed inside each field.

a 32 bits has 2 RMT information. For example:

0x8001 = Level 1 |  0x1 = High level for 1 tick
0x0011 = Level 0 |  0x11 = Low level for 3 ticks.

Level Bit 15 during Bits 0..14 number of Ticks   +  **is sent first**.
Then, signal in the signal described in bits 16 to 31 is sent after that one.

### Therefore myData = 0x8001000e => First sends LOW for 14 ticks and then sends HIGH for 1 tick.

In order to avoid problems, it would be better to set the transmission using this structure:

``` cpp
typedef union {
  struct {
    uint32_t duration0 : 15;
    uint32_t level0    : 1;
    uint32_t duration1 : 15;
    uint32_t level1    : 1;
  };
  uint32_t val;
} rmt_data_t;
```

The equivalent code would be:

``` cpp
void loop() {
  // Init data
  int i;
  for (i = 0; i < 255; i++) {
    //data[i].val = 0x80010001 + ((i % 13) << 16) + 13 - (i % 13);
    data[i].level0 = 0; // LOW
    data[i].duration0 = 1 + 13 - (i % 13); // number of Tick on Low
    data[i].level1 = 1; // HIGH
    data[i].duration1 = 1 + (i % 13); // number of Ticks on HIGH
    // set the received data to zero
    my_data[i].val = 0;
  }
  data[255].val = 0;
  Serial.println("=======\r\nSent Data:\r\n=======");
  // Printout the received data plus the original values
  for (i = 0; i < 60; i++) {
    Serial.printf("%08lx[%c0x%03x|%c0x%03x] ", data[i].val,
                  data[i].level0 ? 'H' : 'L', data[i].duration0,
                  data[i].level1 ? 'H' : 'L', data[i].duration1);

    if (!((i + 1) % 4)) {
      Serial.println("");
    }
  }
  Serial.println("\n");

  // Start an async data read
  size_t rx_num_symbols = RMT_NUM_EXCHANGED_DATA;
  rmtReadAsync(RMT_RX_PIN, my_data, &rx_num_symbols);

  // Write blocking the data to the loopback
  rmtWrite(RMT_TX_PIN, data, RMT_NUM_EXCHANGED_DATA, RMT_WAIT_FOR_EVER);

  // Wait until data is read
  while (!rmtReceiveCompleted(RMT_RX_PIN));

  // Once data is available, the number of RMT Symbols is stored in rx_num_symbols
  // and the received data is copied to my_data
  Serial.printf("Got %d RMT symbols\n", rx_num_symbols);

  // Printout the received data plus the original values
  for (i = 0; i < 60; i++) {
    Serial.printf("%08lx=%08lx ", my_data[i].val, data[i].val);

    if (!((i + 1) % 4)) {
      Serial.println("");
    }
  }
  Serial.println("\n");

  delay(500);
}
```

SuGlider, 2025-03-31:
>real tick set to: 100ns
>
>Please connect GPIO 18 to GPIO 21, now.
>Got 30 RMT symbols
>000d8001=8001000e 000c8002=8002000d 000b8003=8003000c 000a8004=8004000b 
>00098005=8005000a 00088006=80060009 00078007=80070008 00068008=80080007 

Indeed, received data has the 16bits swapped. When applying the `rmt_data_t` to the received data, [level0][duration0] should be [level1][duration1] instead.

SuGlider, 2025-03-31:
This code ilustrates waveforms and how to encode it using RTM (transmission only):
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/RMT/RMTWrite_RGB_LED/RMTWrite_RGB_LED.ino

fanfanlatulipe26, 2025-03-31:
I fully agree with you that the best is to use the various field of the data structure of rmt_data_t., mainly in a teaching example.
Your example is realy better for a beginner.
In fact the deference between the words we received in my_data[] and the reference in data[] comes from the fact that the GPIO TX pin is set to 0 by default a startup and also after each  rmtWrite. 
The first symbol start with a Low pulse (8001000e[L0x00e|H0x001] ) and so there is no transition on the GPIO and nothing detected by the rmtReadAsync.
If we force the idle state of GPIO to HIGH we receive exactly the same words.
I added at the end of startup
```
  if (!rmtSetEOT(RMT_TX_PIN, HIGH)) {
     Serial.println("rmtSetEOT failed\n");
  }
```
Now, as expected only the first burst is bad but the following are perfect:
( a strange error appears : E rmt: user buffer too small, received symbols truncated               ???????????????????????????????????????????)

```
=======
Sent Data:
=======
8001000e[L0x00e|H0x001] 8002000d[L0x00d|H0x002] 8003000c[L0x00c|H0x003] 8004000b[L0x00b|H0x004] 
8005000a[L0x00a|H0x005] 80060009[L0x009|H0x006] 80070008[L0x008|H0x007] 80080007[L0x007|H0x008] 
80090006[L0x006|H0x009] 800a0005[L0x005|H0x00a] 800b0004[L0x004|H0x00b] 800c0003[L0x003|H0x00c] 
......

E rmt: user buffer too small, received symbols truncated               ???????????????????????????????????????????
Got 30 RMT symbols
000e8003=8001000e 000d8001=8002000d 000c8002=8003000c 000b8003=8004000b 
000a8004=8005000a 00098005=80060009 00088006=80070008 00078007=80080007 
00068008=80090006 00058009=800a0005 0004800a=800b0004 0003800b=800c0003 
............     bad, has expected .
..........

=======
Sent Data:
=======
8001000e[L0x00e|H0x001] 8002000d[L0x00d|H0x002] 8003000c[L0x00c|H0x003] 8004000b[L0x00b|H0x004] 
8005000a[L0x00a|H0x005] 80060009[L0x009|H0x006] 80070008[L0x008|H0x007] 80080007[L0x007|H0x008] 
......

Got 30 RMT symbols
8001000e=8001000e 8002000d=8002000d 8003000c=8003000c 8004000b=8004000b 
8005000a=8005000a 80060009=80060009 80070008=80070008 80080007=80080007 
80090006=80090006 800a0005=800a0005 800b0004=800b0004 800c0003=800c0003 
.....
=======
Sent Data:
=======
8001000e[L0x00e|H0x001] 8002000d[L0x00d|H0x002] 8003000c[L0x00c|H0x003] 8004000b[L0x00b|H0x004]

```

SuGlider, 2025-03-31:
>E rmt: user buffer too small, received symbols truncated

This may be related to this `define RMT_MEM_RX RMT_MEM_NUM_BLOCKS_3` --> for the ESP32 it can go up to 8 BLOCKS.
It may also be related to the resolution (`RMT_FREQ`)  and `rmtSetRxMaxThreshold()` value.

Using the same parameters from the example I don't see this error message.

fanfanlatulipe26, 2025-03-31:
I only have this message when I add
```
  if (!rmtSetEOT(RMT_TX_PIN, HIGH)) {
     Serial.println("rmtSetEOT failed\n");
  }
```

SuGlider, 2025-03-31:
> I only have once this message when I add
> 
> ```
>   if (!rmtSetEOT(RMT_TX_PIN, HIGH)) {
>      Serial.println("rmtSetEOT failed\n");
>   }
> ```

Correct, I've gotten the error message now.
The fix for this example is to do not use the EOT as HIGH and just start the samples with High level, instead of Low level.

SuGlider, 2025-04-01:
Yes, I'll update the example.

closed 2025-04-01 by SuGlider

SuGlider, 2025-04-01:
> I only have once this message when I add
> 
> ```
>   if (!rmtSetEOT(RMT_TX_PIN, HIGH)) {
>      Serial.println("rmtSetEOT failed\n");
>   }
> ```

This will make the RMT Write to set the output pin into HIGH at the end of the transmission.
Given that there is a wire creating a TX->RX, RMT RX will read all 30 transitions and then the line will go to HIGH, starting a long endless read (tick on high), which never goes to Low.
This will trigger the RX Buffer Overflow, because it would be read many H:0x7fff symbols in sequence.

fanfanlatulipe26, 2025-04-01:
Yes the very first ransmission was a bit special, starting with a LOW level and ending with a HIGH one and the reveiver decode 31 symbols . After that it only decode 30 symbols as expected.
Thank  you for your help  (-;
I am learning RMT trying. Espressif API is powerfull but a bit tricky.  Arduino library is far simple.