espressif/arduino-esp32 #11201: Missing include for esp32s3usbotg

BostonBlack, 2025-03-29:
### Board

ESP32-S3-USB-OTG

### Device Description

N/A not compiling

### Hardware Configuration

N/A not compiling

### Version

latest master (checkout manually)

### IDE Name

ArduinoIDE

### Operating System

Windows 11

### Flash frequency

N/A not compiling

### PSRAM enabled

yes

### Upload speed

N/A not compiling

### Description

When trying to compile code compilation fails due to a missing #include <stdbool.h> in the pins_arduino.h file

expected behavior is code compiles.

### Sketch

```cpp
#include <Adafruit_GFX.h>    // Core graphics library
#include <Adafruit_ST7789.h> // Hardware-specific library for ST7789
#include <SPI.h>

#define DEBUG_LED_1 45
#define DEBUG_LED_2 46

#define TFT_CS   10
#define TFT_RST  9 // Or set to -1 and connect to Arduino RESET pin
#define TFT_DC   14
#define TFT_BL   21

Adafruit_ST7789 tft = Adafruit_ST7789(TFT_CS, TFT_DC, TFT_RST);

float p = 3.1415926;

void setup(void) {
  Serial.begin(19200);
  Serial.print(F("Hello! ST77xx TFT Test"));

  pinMode(DEBUG_LED_1, OUTPUT);
  pinMode(DEBUG_LED_2, OUTPUT);
  digitalWrite(DEBUG_LED_1, HIGH);
  analogWrite(TFT_BL, 128);

[... 264 lines cut ...]
   93 | void usbHostEnable(bool enable);
      |                    ^~~~
C:\Users\bosto\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.3\variants\esp32s3usbotg/pins_arduino.h:5:1: note: 'bool' is defined in header '<stdbool.h>'; did you forget to '#include <stdbool.h>'?
    4 | #include <stdint.h>
  +++ |+#include <stdbool.h>
    5 | 
"C:\\Users\\bosto\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp-x32\\2405/bin/xtensa-esp32s3-elf-gcc" -MMD -c "@C:\\Users\\bosto\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.3-489d7a2b-v1\\esp32s3/flags/c_flags" -w -Os -Werror=return-type -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ESP32_S3_USB_OTG -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_S3_USB_OTG\"" "-DARDUINO_VARIANT=\"esp32s3usbotg\"" -DARDUINO_PARTITION_default "-DARDUINO_HOST_OS=\"windows\"" "-DARDUINO_FQBN=\"esp32:esp32:esp32s3usbotg:USBMode=default,UploadMode=default,PartitionScheme=default,DebugLevel=none,EraseFlash=none\"" -DESP32=ESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_MODE=0 -DARDUINO_USB_CDC_ON_BOOT=0 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0 "@C:\\Users\\bosto\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.3-489d7a2b-v1\\esp32s3/flags/defines" "-IC:\\Users\\bosto\\OneDrive\\Documents\\A_Home\\Circuits\\ESP32-S3_Board\\Code\\graphicstest_st7789" -iprefix "C:\\Users\\bosto\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.3-489d7a2b-v1\\esp32s3/include/" "@C:\\Users\\bosto\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.3-489d7a2b-v1\\esp32s3/flags/includes" "-IC:\\Users\\bosto\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.3-489d7a2b-v1\\esp32s3/qio_qspi/include" "-IC:\\Users\\bosto\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.1.3\\cores\\esp32" "-IC:\\Users\\bosto\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.1.3\\variants\\esp32s3usbotg" "@C:\\Users\\bosto\\AppData\\Local\\arduino\\sketches\\B5D570C2B9AC99162A61DDF163B8B7CA/build_opt.h" "@C:\\Users\\bosto\\AppData\\Local\\arduino\\sketches\\B5D570C2B9AC99162A61DDF163B8B7CA/file_opts" "C:\\Users\\bosto\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.1.3\\cores\\esp32\\esp32-hal-misc.c" -o "C:\\Users\\bosto\\AppData\\Local\\arduino\\sketches\\B5D570C2B9AC99162A61DDF163B8B7CA\\core\\esp32-hal-misc.c.o"

Using library Adafruit GFX Library at version 1.12.0 in folder: C:\Users\bosto\Documents\Arduino\libraries\Adafruit_GFX_Library 
Using library Adafruit BusIO at version 1.17.0 in folder: C:\Users\bosto\Documents\Arduino\libraries\Adafruit_BusIO 
Using library Wire at version 3.1.3 in folder: C:\Users\bosto\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.3\libraries\Wire 
Using library SPI at version 3.1.3 in folder: C:\Users\bosto\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.3\libraries\SPI 
Using library Adafruit ST7735 and ST7789 Library at version 1.11.0 in folder: C:\Users\bosto\Documents\Arduino\libraries\Adafruit_ST7735_and_ST7789_Library 
exit status 1

Compilation error: exit status 1
```

### Other Steps to Reproduce

Putting the #include in the file fixes the issue

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2025-03-29:
Seems not using latest. Fixed with https://github.com/espressif/arduino-esp32/pull/11155

me-no-dev, 2025-03-29:
Closing as fixed

closed 2025-03-29 by me-no-dev