espressif/arduino-esp32 #11207: RS232 over UART not working with Core V3

Glujaz, 2025-03-31:
### Board

ESP32 WROOM

### Device Description

ESP32 dev board from Az-Delivery. soldered on a dev breadboard with some wires to a MAX3232. Both devices are socketed to the breadboard for easy replacement.
When not connected to the computer (99.9% of the time), power is coming from a USB cable to the 5V pins of the ESP32 (via the dev breadboard)

### Hardware Configuration

MAX3232 connected to the standard Serial 2 pins.

### Version

v3.2.0

### IDE Name

Arduino IDE

### Operating System

Mac OS 15.3.2

### Flash frequency

240

### PSRAM enabled

no

### Upload speed

921600

### Description

The communication from my ESP32 to my MAX3232 for RS232 communication with an audio amp stop working with the last versions of ESP32 Core. It neither receives nor emits

Using Arduino IDE, and after updating ESP32 core from version 2.0.17 to 3.2.0 (and actually tested with other versions as well like 3.0.0 and some in between), it just stoped working. Going back to version 2.0.17 makes the code work again.

### Sketch

```cpp
#define CONSOLE_SERIAL Serial

bool rs232UpdateEnabled = true; // Initially set RS232 updates to ON
bool ampOn = false; // Initially set amplifier to OFF

void setup() {
  CONSOLE_SERIAL.begin(115200); // Set the baud rate for the Arduino IDE's serial console
  Serial2.begin(115200, SERIAL_8N1); // Initialize Serial1 for RS232 communication with the amplifier (8 data bits, no parity, 1 stop bit)
  delay(1000); // Allow time for the serial connections to initialize

  // Display a welcome message in the serial console
  CONSOLE_SERIAL.println("Welcome to Amp Control!");
  CONSOLE_SERIAL.println("Type 'button' to toggle between 'on-amp' and 'off-amp', 'rs232_update_on' to enable RS232 updates, 'rs232_update_off' to disable RS232 updates, 'power?' to get the power status, or 'restart' to restart the ESP32.");
  CONSOLE_SERIAL.print("> "); // Display a prompt for user input
}

void loop() {
  if (CONSOLE_SERIAL.available() > 0) {
    String command = CONSOLE_SERIAL.readStringUntil('\n'); // Read the command from the console
    command.trim(); // Remove leading/trailing whitespaces

    // Check the received command
    if (command == "button") {
      // Toggle between 'on-amp' and 'off-amp' commands based on the button press
      ampOn = !ampOn;
      if (ampOn) {
        sendCommand("power_on!");
        CONSOLE_SERIAL.println("Sent 'power_on!' command to the amp.");
      } else {
        sendCommand("power_off!");
        CONSOLE_SERIAL.println("Sent 'power_off!' command to the amp.");
      }
    } else if (command == "rs232_update_on") {
      rs232UpdateEnabled = true;
      CONSOLE_SERIAL.println("RS232 updates enabled.");
    } else if (command == "rs232_update_off") {
      rs232UpdateEnabled = false;
      CONSOLE_SERIAL.println("RS232 updates disabled.");
    } else if (command == "power?") {
      sendCommand("power?");
      delay(100); // Delay to allow time for the response from the amplifier
      if (Serial2.available() > 0) {
        String response = Serial2.readStringUntil('$'); // Read the RS232 response until the terminating "$" character
        if (response.length() > 0) {
          CONSOLE_SERIAL.println("Power status response: " + response);
        }
      } else {
        CONSOLE_SERIAL.println("No response received from the amplifier.");
      }
    } else if (command == "restart") {
      CONSOLE_SERIAL.println("Restarting ESP32...");
      delay(1000); // Optional delay for user feedback
      ESP.restart(); // Restart the ESP32
    } else {
      CONSOLE_SERIAL.println("Invalid command.");
    }

    // Display a prompt for the next user input
    CONSOLE_SERIAL.print("> ");
  }
  
  // Check for RS232 updates if enabled
  if (rs232UpdateEnabled && Serial2.available() > 0) {
    String response = Serial2.readStringUntil('$'); // Read the RS232 response until the terminating "$" character
    if (response.length() > 0) {
      CONSOLE_SERIAL.println("Received RS232 update: " + response);
    }
  }
}

void sendCommand(const char* command) {
  // Send the specified command to the amplifier via Serial2 (RS232)
  Serial2.print(command); // Send the command without a newline
  Serial2.print("!");    // Send the terminating "!" character
  delay(100); // Adjust delay as needed to ensure the command is sent
}
```

### Debug Message

```plain
When amp sends text, nothing happens.

When sending a command:
"No response received from the amplifier."

(as expected if no answers)
```

### Other Steps to Reproduce

Tried on another ESP32 Core version 2.1.7, and it works

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2025-03-31:
Please note that default Serial2 pins have changed from Arduino Core 2.0.x [16,17] to Core 3.0.x [4,25].

This may be the cause of this issue.

Please use this:

``` cpp
Serial2.begin(115200, SERIAL_8N1, 16, 17);
```

SuGlider, 2025-04-04:
Due to no replies, I would say that this issue has been solved.
I'll close it. Feel free to open it agian if necessary.

closed 2025-04-04 by SuGlider

Glujaz, 2025-09-01:
It is wise to a late reply that I confirm here that this is the working solution. They simply changed the pins.