espressif/arduino-esp32 #11211: Tolerance and Min/Max settings for Zigbee Temperature and Humidity cluster settings don't work as expected

def1149, 2025-04-01:
### Board

ESP32-cC6

### Device Description

Adafruit Feather ESP32-c6

### Hardware Configuration

N/A

### Version

v3.2.0

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

For the  Temperature and Humidity endpoint. Other numerically valued endpoints were not tested 

1. If either the Temperature value or the Humidity value fall outside the upper and lower limits set when configuring 
    the endpoint, garbage values are reported to Home Assistant ZHA.  The set limit value should be reported, or 
    some indication like ">>" or "<<" should be reported to indicate the limits were exceeded. I'm don't know if the 
    Zigbee standard calls out a result for this scenario 

2. Setting the tolerance to 0.01 for Temperature or Humidity doesn't work. Tolerance is limited to 0.1  

### Sketch

```cpp
N/A
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2025-04-02:
Hi @def1149, thank you for the report. I will take a look.

P-R-O-C-H-Y, 2025-05-06:
Hi @def1149, I can point you to the Zigbee cluster specification:
If you use MIN/MAX measured values, it means that you cannot set Measured values outside this area. If you do, the value is invalid.

[image]

Tolerance should be possible to set to 0,01 °C. Do you have any error messages or logs where you see it does not work?

def1149, 2025-05-06:
Tolerance worked at 0.01, but I had to set it in Home Assistant because I changed the setting in code after **initial** enrollment 

Min and Max settings work, but if your reported values exceed those settings, you get random garbage values in HA.  I think you should get no value or some other Zigbee-defined indication that the range setting was violated, instead of a garbage value

P-R-O-C-H-Y, 2025-05-06:
Then its something maybe worth reporting on HA side.
But as written in the cluster specification, if you set value outside the MIN/MAX range its invalid value. So you are the one controlling the value, so you should set in your code the min/max according to your temp sensor and filter out the values outside the range :)

def1149, 2025-05-06:
Yes, I set it very wide now as it's an unnecessary feature

def1149, 2025-05-06:
Closing this as not a bug

closed 2025-05-12 by def1149