espressif/arduino-esp32 #11219: Only Four of Six Temperature Sensors Appear as Bound Devices

dosportsglobal, 2025-04-05:
### Board

ESP32-C6-WROOM 1

### Device Description

DevKitC

### Hardware Configuration

No.

### Version

latest stable Release (if not listed below)

### IDE Name

Arduino Ide

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I am attempting to connect six temperature sensors to a thermostat using an ESP32-C6 with Zigbee configuration. However, when I print the list of bound devices, only the first four sensors appear. Could you please advise why the remaining two sensors are not being recognized?

### Sketch

```cpp
#ifndef ZIGBEE_MODE_ZCZR
#error "Zigbee coordinator mode is not selected in Tools->Zigbee mode"
#endif

#include <set>
#include "Zigbee.h"
#include <Adafruit_NeoPixel.h>

constexpr uint8_t LED_PIN = 8;
constexpr uint8_t NUM_LEDS = 1;
Adafruit_NeoPixel rgbLed(NUM_LEDS, LED_PIN, NEO_GRB + NEO_KHZ800);

struct RGB {
  uint8_t r, g, b;
};

constexpr RGB COLOR_OFF = { 0, 0, 0 };
constexpr RGB CUSTOM_COLOR = { 0, 255, 255 };
constexpr RGB CUSTOM_COLOR1 = { 255, 0, 0 };
constexpr RGB CUSTOM_COLOR2 = { 100, 100, 50 };

bool light = false;

/* Zigbee thermostat configuration */
#define THERMOSTAT_ENDPOINT_NUMBER 5
#define SWITCH_ENDPOINT_NUMBER 6
uint8_t button = BOOT_PIN;

ZigbeeThermostat zbThermostat = ZigbeeThermostat(THERMOSTAT_ENDPOINT_NUMBER);
ZigbeeColorDimmerSwitch zbSwitch = ZigbeeColorDimmerSwitch(SWITCH_ENDPOINT_NUMBER);
int rem = 40;
bool isOperational = true;
bool isBroadcasting = false;
bool recCon = false;
std::set<uint64_t> connectedDevices;
unsigned long currentmillis;
unsigned long previousmillis;
unsigned long previousmillis1 = 0;
unsigned long extramillis = 0;
const unsigned long period1 = 1000;
const unsigned long period2 = 10000;
const unsigned long period3 = 60000;
const unsigned long period5 = rem * 1000;
const unsigned long period4 = 100;

float manualMinTemp = 0.0;
float manualMaxTemp = 100.0;
float manualTolerance = 0.1;

void setColor(const RGB &color) {
  rgbLed.setPixelColor(0, rgbLed.Color(color.r, color.g, color.b));
  rgbLed.show();
}
/****************** Temperature sensor handling *******************/
void recieveSensorTemp(float temperature) {
  setColor(COLOR_OFF);
  delay(200);
  setColor(CUSTOM_COLOR);
}
/********************* Arduino functions **************************/

void manualConfigCallback(float minTemp, float maxTemp, float tolerance) {
  Serial.print("Setting manual config: ");
  Serial.print(minTemp);
  Serial.print(" - ");
  Serial.print(maxTemp);
  Serial.print(" (Tolerance: ");
  Serial.print(tolerance);
  Serial.println(")");

  // Store values (if needed for later use)
  manualMinTemp = minTemp;
  manualMaxTemp = maxTemp;
  manualTolerance = tolerance;
}

void setup() {
  Serial.begin(115200);
  setColor(CUSTOM_COLOR);

  // Init button switch
  pinMode(button, INPUT_PULLUP);

  // Set callback functions for temperature and configuration receive
  zbThermostat.onTempRecieve(recieveSensorTemp);
  zbThermostat.onConfigRecieve(manualConfigCallback);

  // Manually trigger the config callback

  //Optional: set Zigbee device name and model
  zbThermostat.setManufacturerAndModel("Espressif", "ZigbeeThermostat");
  zbSwitch.setManufacturerAndModel("Espressif", "ZigbeeSwitch");

  zbThermostat.allowMultipleBinding(true);
  zbSwitch.allowMultipleBinding(true);

  //Add endpoint to Zigbee Core
  Zigbee.addEndpoint(&zbThermostat);
  Zigbee.addEndpoint(&zbSwitch);

  // When all EPs are registered, start Zigbee with ZIGBEE_COORDINATOR mode
  if (!Zigbee.begin(ZIGBEE_COORDINATOR)) {
    Serial.println("Zigbee failed to start!");
    Serial.println("Rebooting...");
    ESP.restart();
  }

  if (zbThermostat.bound()) {
    Zigbee.factoryReset();
    delay(1000);
    ESP.restart();
  }

  Serial.println("Network Initialized");
  manualConfigCallback(manualMinTemp, manualMaxTemp, manualTolerance);
}

void loop() {

  if (Serial.available() > 0) {
    String input = Serial.readStringUntil('\n');
    input.trim();
    if (input == "PING") {
      recCon = true;
      Serial.println("PONG");
      delay(100);
    } else if (input.equalsIgnoreCase("change")) {
      rem = 40;
      if (isOperational) {
        isOperational = false;
        isBroadcasting = true;
        Zigbee.openNetwork(rem);
        previousmillis = millis();
        previousmillis1 = millis();
      } else if (isBroadcasting) {
        isOperational = true;
        isBroadcasting = false;
      }
    } else if (input.equalsIgnoreCase("s")) {
      zbSwitch.setLightLevel(2);
    } else if (input.equalsIgnoreCase("b")) {
      if (light) {
        zbSwitch.setLightLevel(3);
        light = false;
      } else {
        zbSwitch.setLightLevel(9);
        light = true;
      }
    } else if (input.equalsIgnoreCase("0")) {
      zbSwitch.setLightLevel(0);
    } else if (input.equalsIgnoreCase("1")) {
      zbSwitch.setLightLevel(1);
    } else if (input.equalsIgnoreCase("t")) {
      zbSwitch.setLightLevel(4);
    } else if (input.equalsIgnoreCase("c")) {
      zbSwitch.setLightLevel(5);
    } else if (input.startsWith("channel")) {
      zbSwitch.printBoundDevices(Serial);
      zbThermostat.printBoundDevices(Serial);
      int inputInt = input.substring(7).toInt();

      if (inputInt >= 1 && inputInt <= 14) {
        // Switch to the operational channel

      } else {
      }
    }
  }

  if (isBroadcasting) {
    currentmillis = millis();
    if (currentmillis - previousmillis >= period5) {
      isOperational = true;
      isBroadcasting = false;
    }
    if (currentmillis - previousmillis1 >= period1) {
      previousmillis1 = currentmillis;
      rem--;
      Serial.println("Time: " + String(rem));
    }
  }

  if (digitalRead(button) == LOW) {  // Push button pressed
    // Key debounce handling
    while (digitalRead(button) == LOW) {
      setColor(CUSTOM_COLOR2);
    }
    // Toggle light
    zbSwitch.lightToggle();
    setColor(CUSTOM_COLOR1);
    zbThermostat.printBoundDevices(Serial);
  }
}
```

### Debug Message

```plain
Nothing like this.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2025-04-07:
Hi @dosportsglobal Thank you for the report. Can you also include some logs? And are the 6 sensors really connected?
Just to clarify, you are using 6 ESPs with temperature sensor example or some manufactured Zigbee temp sensors?

dosportsglobal, 2025-04-11:
Yes the 6 temperature sensors are really connected and the thermostat is being able to receive data from them but since there are only 4 bound device showing both in thermostat as well as switch, the command sent by switch is being sent only to the first 4 devices.

void ZigbeeThermostat::bindCb(esp_zb_zdp_status_t zdo_status, void *user_ctx) {
  if (zdo_status == ESP_ZB_ZDP_STATUS_SUCCESS) {
    if (user_ctx) {
      zb_device_params_t *sensor = (zb_device_params_t *)user_ctx;
      log_i("The temperature sensor originating from address(0x%x) on endpoint(%d)", sensor->short_addr, sensor->endpoint);
      _instance->_bound_devices.push_back(sensor);
    } else {
      log_v("Local binding success");
    }
    _is_bound = true;
  } else {
    log_e("Binding failed!");
  }
}

I had tried some logging where I found that user_ctx is true for the first four devices and giving the details of the bound devices but after that it only gives "Local binding success" and also "Binding Failed!".

P-R-O-C-H-Y, 2025-04-14:
@dosportsglobal Thanks for additional info. I will take a look.

dosportsglobal, 2025-04-18:
> [@dosportsglobal](https://github.com/dosportsglobal) Thanks for additional info. I will take a look.

Any update yet? My project is hanging for this issue only.

P-R-O-C-H-Y, 2025-04-22:
The open network for another 180s does not come from the Zigbee library. I will investigate if this behavior is specified somewhere in Zigbee docs and is expected. The HA does it the same way so I am expecting that. But for any case, I will also add a function `closeNetwork()` which can be called manually.

P-R-O-C-H-Y, 2025-04-22:
This week I will be testing your issue, so I will keep you updated when I have any news.

P-R-O-C-H-Y, 2025-04-25:
@dosportsglobal sending an update:
I am finishing a PR which also includes changes around binding. I did test binding using HA (ZHA) and I was able to bind 6 lights to a switch + the coordinator was also bind so 7 devices with no issue (but the switch was not the coordinator as is in your case). I will do this test that the switch will act as coordinator. If it will be fine, I will open a PR and will be glad if you would be able to test with the changes in your scenario.

P-R-O-C-H-Y, 2025-04-28:
Hi @dosportsglobal, can you please give a try the PR #11316? Maybe it can fix your issues.