espressif/arduino-esp32 #11223: rmtInit() error while booting

fOmey, 2025-04-07:
### Board

ESP32S3 Dev Module

### Device Description

ESP32-S3R8

### Hardware Configuration

PIN 6 being used for NeoPixel RGB

### Version

latest stable Release (if not listed below)

### IDE Name

Arduino

### Operating System

Windows 11

### Flash frequency

80mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

```
E (3892) rmt: rmt_new_tx_channel(269): not able to power down in light sleep
[  3003][E][esp32-hal-rmt.c:548] rmtInit(): GPIO 6 - RMT TX Initialization error.
[  3020][E][esp.c:81] espShow(): Failed to init RMT TX mode on pin 6
```

### Sketch

```cpp
// NeoPixel Ring simple sketch (c) 2013 Shae Erisson
// Released under the GPLv3 license to match the rest of the
// Adafruit NeoPixel library
// Simple example

#include <Adafruit_NeoPixel.h>
#ifdef __AVR__
 #include <avr/power.h> // Required for 16 MHz Adafruit Trinket
#endif

// Which pin on the Arduino is connected to the NeoPixels?
#define PIN        6 // On Trinket or Gemma, suggest changing this to 1

// How many NeoPixels are attached to the Arduino?
#define NUMPIXELS 1 // Popular NeoPixel ring size

// When setting up the NeoPixel library, we tell it how many pixels,
// and which pin to use to send signals. Note that for older NeoPixel
// strips you might need to change the third parameter -- see the
// strandtest example for more information on possible values.
Adafruit_NeoPixel pixels(NUMPIXELS, PIN, NEO_GRB + NEO_KHZ800);

#define DELAYVAL 500 // Time (in milliseconds) to pause between pixels

void setup() {
  // These lines are specifically to support the Adafruit Trinket 5V 16 MHz.
  // Any other board, you can remove this part (but no harm leaving it):
#if defined(__AVR_ATtiny85__) && (F_CPU == 16000000)
  clock_prescale_set(clock_div_1);
#endif
  // END of Trinket-specific code.

  pixels.begin(); // INITIALIZE NeoPixel strip object (REQUIRED)
}

void loop() {
  pixels.clear(); // Set all pixel colors to 'off'

  // The first NeoPixel in a strand is #0, second is 1, all the way up
  // to the count of pixels minus one.
  for(int i=0; i<NUMPIXELS; i++) { // For each pixel...

    // pixels.Color() takes RGB values, from 0,0,0 up to 255,255,255
    // Here we're using a moderately bright green color:
    pixels.setPixelColor(i, pixels.Color(0, 150, 0));

    pixels.show();   // Send the updated pixel colors to the hardware.

    delay(DELAYVAL); // Pause before next pass through loop
  }
}
```

### Debug Message

```plain
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-S3
  Package           : 0
  Revision          : 0.02
  Cores             : 2
  CPU Frequency     : 240 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000012
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   391672 B ( 382.5 KB)
  Free Bytes        :   355280 B ( 347.0 KB)
  Allocated Bytes   :    31456 B (  30.7 KB)
  Minimum Free Bytes:   350432 B ( 342.2 KB)
  Largest Free Block:   286708 B ( 280.0 KB)
------------------------------------------
SPIRAM Memory Info:
------------------------------------------
  Total Size        :  2097152 B (2048.0 KB)
  Free Bytes        :  2095104 B (2046.0 KB)
  Allocated Bytes   :        0 B (   0.0 KB)
  Minimum Free Bytes:  2095104 B (2046.0 KB)
  Largest Free Block:  2064372 B (2016.0 KB)
  Bus Mode          : QSPI
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         :  8388608 B (8 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1920.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x001F0000, size:  1920.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x003D0000, size:   128.0 KB, type: DATA, subtype: SPIFFS
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
Software Info:
------------------------------------------
  Compile Date/Time : Apr  4 2025 14:59:42
  Compile Host OS   : windows
  ESP-IDF Version   : v5.4.1-1-g2f7dcd862a-dirty
  Arduino Version   : 3.2.0
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32S3_DEV
  Arduino Variant   : esp32s3
  Arduino FQBN      : esp32:esp32:esp32s3:UploadSpeed=921600,USBMode=hwcdc,CDCOnBoot=default,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,CPUFreq=240,FlashMode=qio,FlashSize=8M,PartitionScheme=min_spiffs,DebugLevel=debug,PSRAM=enabled,LoopCore=1,EventsCore=1,EraseFlash=all,JTAGAdapter=default,ZigbeeMode=default
============ Before Setup End ============

=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-S3
  Package           : 0
  Revision          : 0.02
  Cores             : 2
  CPU Frequency     : 240 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000012
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes

> =========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   296232 B ( 289.3 KB)
  Free Bytes        :   239300 B ( 233.7 KB)
  Allocated Bytes   :    51516 B (  50.3 KB)
  Minimum Free Bytes:   187472 B ( 183.1 KB)
  Largest Free Block:   122868 B ( 120.0 KB)
------------------------------------------
SPIRAM Memory Info:
------------------------------------------
  Total Size        :  2097152 B (2048.0 KB)
  Free Bytes        :  2090668 B (2041.7 KB)
  Allocated Bytes   :     3908 B (   3.8 KB)
  Minimum Free Bytes:  2062916 B (2014.6 KB)
  Largest Free Block:  2064372 B (2016.0 KB)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2025-04-07:
@fOmey - I have uploaded the sketch from Adafruit NeoPixel Library called `simple.ino`. Changed it a bit to display some different colors.

Using the devkit ESP32-S3-WROOM-2 that has an RGB LED on pin 38. It works fine. No issues here.

``` cpp
// NeoPixel Ring simple sketch (c) 2013 Shae Erisson
// Released under the GPLv3 license to match the rest of the
// Adafruit NeoPixel library

#include <Adafruit_NeoPixel.h>
#ifdef __AVR__
 #include <avr/power.h> // Required for 16 MHz Adafruit Trinket
#endif

// Which pin on the Arduino is connected to the NeoPixels?
#define PIN        38 // On Trinket or Gemma, suggest changing this to 1

// How many NeoPixels are attached to the Arduino?
#define NUMPIXELS 1 // Popular NeoPixel ring size

// When setting up the NeoPixel library, we tell it how many pixels,
// and which pin to use to send signals. Note that for older NeoPixel
// strips you might need to change the third parameter -- see the
// strandtest example for more information on possible values.
Adafruit_NeoPixel pixels(NUMPIXELS, PIN, NEO_GRB + NEO_KHZ800);

#define DELAYVAL 800 // Time (in milliseconds) to pause between pixels

void setup() {
  // These lines are specifically to support the Adafruit Trinket 5V 16 MHz.
  // Any other board, you can remove this part (but no harm leaving it):
#if defined(__AVR_ATtiny85__) && (F_CPU == 16000000)
  clock_prescale_set(clock_div_1);
#endif
  // END of Trinket-specific code.

  pixels.begin(); // INITIALIZE NeoPixel strip object (REQUIRED)
}

void loop() {
  pixels.clear(); // Set all pixel colors to 'off'

  // The first NeoPixel in a strand is #0, second is 1, all the way up
  // to the count of pixels minus one.
  for(int i=0; i<NUMPIXELS; i++) { // For each pixel...

    // pixels.Color() takes RGB values, from 0,0,0 up to 255,255,255
    // Here we're using a moderately bright green color:
    pixels.setPixelColor(i, pixels.Color(0, 150, 0));
    pixels.show();   // Send the updated pixel colors to the hardware.
    delay(DELAYVAL); // Pause before next pass through loop
    pixels.setPixelColor(i, pixels.Color(150, 0, 0));
    pixels.show();   // Send the updated pixel colors to the hardware.
    delay(DELAYVAL); // Pause before next pass through loop
    pixels.setPixelColor(i, pixels.Color(0, 0, 150));
    pixels.show();   // Send the updated pixel colors to the hardware.
    delay(DELAYVAL); // Pause before next pass through loop
  }
}
```

#### Console Output:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
[... 125 lines cut ...]
[  3211][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 38 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[  3221][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 38 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[  4032][V][esp32-hal-rmt.c:454] rmtInit(): GPIO 38 - TX MODE - MemSize[48] - Freq=10000000Hz
[  4040][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 38 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[  4050][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 38 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[  4861][V][esp32-hal-rmt.c:454] rmtInit(): GPIO 38 - TX MODE - MemSize[48] - Freq=10000000Hz
[  4869][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 38 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[  4879][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 38 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[  5690][V][esp32-hal-rmt.c:454] rmtInit(): GPIO 38 - TX MODE - MemSize[48] - Freq=10000000Hz
[  5698][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 38 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[  5708][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 38 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[  6519][V][esp32-hal-rmt.c:454] rmtInit(): GPIO 38 - TX MODE - MemSize[48] - Freq=10000000Hz
[  6527][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 38 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[  6537][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 38 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[  7348][V][esp32-hal-rmt.c:454] rmtInit(): GPIO 38 - TX MODE - MemSize[48] - Freq=10000000Hz
[  7356][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 38 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[  7366][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 38 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[  8177][V][esp32-hal-rmt.c:454] rmtInit(): GPIO 38 - TX MODE - MemSize[48] - Freq=10000000Hz
[  8185][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 38 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[  8195][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 38 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[  9006][V][esp32-hal-rmt.c:454] rmtInit(): GPIO 38 - TX MODE - MemSize[48] - Freq=10000000Hz
[  9014][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 38 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[  9024][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 38 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[  9835][V][esp32-hal-rmt.c:454] rmtInit(): GPIO 38 - TX MODE - MemSize[48] - Freq=10000000Hz
```

fOmey, 2025-04-07:
@SuGlider - Thanks for the quick reply.

I tested your example and same issue, however I did notice I was using `debug` mode instead of the `verbose` core debug level. Verbose has revealed a little more info:

```
[   651][V][esp32-hal-rmt.c:454] rmtInit(): GPIO 38 - TX MODE - MemSize[48] - Freq=10000000Hz
[   660][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type RMT_TX (10) successfully set to 0x42003294
[   671][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type RMT_RX (11) successfully set to 0x42003294
[   682][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 38 successfully set to type INIT (0) with bus 0x0
E (555) rmt: rmt_new_tx_channel(269): not able to power down in light sleep
[   692][E][esp32-hal-rmt.c:548] rmtInit(): GPIO 38 - RMT TX Initialization error.
[   706][V][esp32-hal-rmt.c:142] _rmtDetachBus(): Detaching RMT GPIO Bus
[   713][E][esp.c:81] espShow(): Failed to init RMT TX mode on pin 38
```

I have imitated your board configuration settings and by disabling `PSRAM`, the `rmtInit()` error has disappeared:

```
[   583][V][esp32-hal-rmt.c:454] rmtInit(): GPIO 38 - TX MODE - MemSize[48] - Freq=10000000Hz
[   592][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type RMT_TX (10) successfully set to 0x42003060
[   603][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type RMT_RX (11) successfully set to 0x42003060
[   614][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 38 successfully set to type INIT (0) with bus 0x0
[   624][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 38 successfully set to type RMT_TX (10) with bus 0x3fc9d5f0
[   636][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 38 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[   646][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 38 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[  1456][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 38 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[  1465][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 38 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[  2276][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 38 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[  2285][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 38 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[  3096][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 38 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[  3105][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 38 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[  3916][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 38 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[  3925][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 38 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[  4736][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 38 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[  4745][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 38 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[  5556][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 38 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[  5565][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 38 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[  6376][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 38 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[  6385][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 38 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[  7196][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 38 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[  7205][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 38 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[  8016][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 38 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[  8025][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 38 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[  8836][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 38 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[  8845][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 38 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[  9656][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 38 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[  9665][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 38 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[ 10476][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 38 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[ 10485][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 38 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[ 11296][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 38 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[ 11305][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 38 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[ 12116][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 38 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[ 12125][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 38 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[ 12936][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 38 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[ 12945][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 38 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[ 13756][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 38 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[... cut ...]
[ 14585][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 38 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[ 15396][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 38 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[ 15405][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 38 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[ 16216][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 38 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[ 16225][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 38 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
```

SuGlider, 2025-04-07:
`E (555) rmt: rmt_new_tx_channel(269): not able to power down in light sleep` is an IDF error message.
This is from esp-idf/components/esp_driver_rmt/src/rmt_tx.c:
``` cpp
#if !SOC_RMT_SUPPORT_SLEEP_RETENTION
    ESP_RETURN_ON_FALSE(config->flags.allow_pd == 0, ESP_ERR_NOT_SUPPORTED, TAG, "not able to power down in light sleep");
#endif // SOC_RMT_SUPPORT_SLEEP_RETENTION
```
It is due to some memory trash in RMT config file.
It  has been fixed with the PR https://github.com/espressif/arduino-esp32/pull/11203

Could you please apply this PR and let me know if this issue is gone?

fOmey, 2025-04-07:
@SuGlider - PR #11203 appears to have fixed the issue for me 👍

closed 2025-04-07 by fOmey