espressif/arduino-esp32 #11225: Error initializing RMT: not able to power down in light sleep

moose4lord, 2025-04-07:
### Board

Waveshare ESP32-S3 Matrix

### Device Description

The Waveshare ESP32-S3 Matrix has 64 WS2812B LEDs built-in, driven by GPIO 14.

### Hardware Configuration

GPIO 14 drives the WS2812B LEDs

### Version

v3.2.0

### IDE Name

Arduino IDE

### Operating System

macOS 15.4

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

460800

### Description

After upgrading to arduino=esp32 v3.2.0, when using the Adafruit Neopixel library's **simple** example sketch to drive WS2812B LEDs, I get this error message and the ESP32-S3 halts:
```
E (4740) rmt: rmt_new_tx_channel(269): not able to power down in light sleep
```

If I set **USB CDC On Boot: Disabled** in the Arduino IDE tools menu, then the problem goes away.

### Sketch

```cpp
#include <Adafruit_NeoPixel.h>

#define PIN       14
#define NUMPIXELS 64

Adafruit_NeoPixel pixels(NUMPIXELS, PIN, NEO_RGB + NEO_KHZ800);

#define DELAYVAL 20

void setup() {
  Serial.begin(115200);

  pixels.begin();
}

void loop() {
  pixels.clear();

  for(int i=0; i<NUMPIXELS; i++) {
    pixels.setPixelColor(i, pixels.Color(0, 8, 0));  // green
    pixels.show();
    delay(DELAYVAL);
  }
}
```

### Debug Message

```plain
Turning on verbose logging shows this:

[  4871][V][esp32-hal-rmt.c:454] rmtInit(): GPIO 14 - TX MODE - MemSize[48] - Freq=10000000Hz
 [  4871][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type RMT_TX (10) successfully set to 0x42003d2c
[  4879][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type RMT_RX (11) successfully set to 0x42003d2c
[  4890][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 14 successfully set to type INIT (0) with bus 0x0
E (4740) rmt: rmt_new_tx_channel(269): not able to power down in light sleep
[  4907][E][esp32-hal-rmt.c:548] rmtInit(): GPIO 14 - RMT TX Initialization error.
[  4914][V][esp32-hal-rmt.c:142] _rmtDetachBus(): Detaching RMT GPIO Bus
[  4921][E][esp.c:81] espShow(): Failed to init RMT TX mode on pin 14
```

### Other Steps to Reproduce

ESP-IDF v5.4.1 introduced  a new **rmt_tx_channel_config_t::allow_pd** parameter which seems to be related to this error, since the "not able to power down in light sleep" message is mentioned in the v5.4.1 RMT API documentation here:
[https://docs.espressif.com/projects/esp-idf/en/v5.4.1/esp32s3/api-reference/peripherals/rmt.html](https://docs.espressif.com/projects/esp-idf/en/v5.4.1/esp32s3/api-reference/peripherals/rmt.html)

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

guooscar, 2025-04-07:
I also encountered the same issue, but when I set **build_type = debug**, the functionality works correctly.

SuGlider, 2025-04-07:
@guooscar @moose4lord - this is the same issue from https://github.com/espressif/arduino-esp32/issues/11223
Please apply the PR #11203 to fix this issue.

SuGlider, 2025-04-10:
Closing as solved.