espressif/arduino-esp32 #11230: Glitch during ESP32-C6 Startup

def1149, 2025-04-08:
### Board

Adafruit ESP32-C6 Feather

### Device Description

Nothing attached

### Hardware Configuration

Nothing attached

### Version

v3.2.0

### IDE Name

Arduino 3.3.5

### Operating System

Windows 11

### Flash frequency

40 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Occasionally, expected Zigbee data transmission is missing in Home Assistant

To debug: 
Using the Espressif Sleepy Temperature sensor example with modifications to log events of note and running in ESP32 debug mode 

I created a test temperature generator that monotonically increases the temperature by 0.5C every wake-up cycle and I plot the values in Home Assistant. I expect to see a consistent stair-step pattern on the graph. Occasionally the pattern demonstrates missing values 

The ESP32-C6 seems to experience an error on startup during initialization after awakening from sleep. 

[image]

This is the annotated (####) corresponding log output.
```
12:37:07.236 -> Going to sleep now                               #### end of previous connection cycle
12:37:17.981 -> ========== Before Setup Start ===========        #### wakeup for this connection and start of good cycle 
12:37:17.981 -> Chip Info:
12:37:17.981 -> ------------------------------------------
12:37:17.981 ->   Model             : ESP32-C6
12:37:17.981 ->   Package           : 1
12:37:17.981 ->   Revision          : 0.01
12:37:17.981 ->   Cores             : 1
12:37:17.981 ->   CPU Frequency     : 80 MHz
12:37:17.981 ->   XTAL Frequency    : 40
12:37:17.981 ->   Features Bitfield : 0x00000052
12:37:17.981 ->   Embedded Flash    : No
12:37:17.981 ->   Embedded PSRAM    : No
12:37:17.981 ->   2.4GHz WiFi       : Yes
12:37:17.981 ->   Classic BT        : No
12:37:17.981 ->   BT Low Energy     : Yes
12:37:17.981 ->   IEEE 802.15.4     : Yes
12:37:17.981 -> ------------------------------------------
12:37:17.981 -> INTERNAL Memory Info:
[... 350 lines cut ...]
        esp_deep_sleep_start();
      }
    }
  }

  // Call the function to measure temperature and put the device to sleep
  meausureAndSleep();
}
```

### Debug Message

```plain
See above
```

### Other Steps to Reproduce

Program an ESP32-C6 with the script and enroll it as a Zigbee ZHA device on Home Assistant 2025.4.1
Use a history card to graph the temperature values represented by the device
 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2025-04-09:
Hi @def1149, How often this glitch happens? If its reproducible, can you try to set Core Debug Level to "Error" only or disable, to see if its related to the printing to UART/USB or if its something related to Zigbee.

def1149, 2025-04-09:
@P-R-O-C-H-Y  I have noticed other unexplainable "flat" spots in the HA temperature plots of other sensors I've created to monitor freezers where temperatures naturally cycle higher and lower, using both C6 and H2 processors. When I couldn't find a cause in my code, I modified the example code to create a situation to more easily identify any missing data points, and logged while running the code. I've seen the glitch in the log previously and repeated the testing before opening the issue. 

It's not clear that this is a  Zigbee problem because the glitch occurs during startup before any application code executes. I don't know what executes during board startup. 

I will repeat this test on a different board with error level logging

def1149, 2025-04-09:
@lbernstone 

> Do you get the issue if you remove all the button logic?
> 
> Please post code that matches the output. Why are you forcing millis and wakeup_reason into incorrect types? (this is definitely not the problem, but is going to bite you in the long run) Use serial.flush to wait for output to finish rather than delays

I haven't tried removing button logic, but I'm not activating that while testing so I doubt that it's an issue.  

"It's not clear that this is a Zigbee problem because the glitch occurs during startup before any application code executes. I don't know what executes during board startup."

The  posted code was used to create the log 

I'm not using delays for serial output, but to ensure the Zigbee stack has time to complete the transactions as there doesn't seem to be a function that returns something like transaction complete that I can wait on. I know that without a delay after reporting that data will sometimes not be transmitted even though zbTempSensor.report() returns true 

I'm using humidity and batteryPercentage to send some status values to help with debugging when I run with all logging and serial output disabled. I'm often running multiple test devices distributed at various distances from the controller to asses performance at low signal levels and can't hook up a PC to log output all of them. There doesn't seem to be an easy way to add additional "debug data clusters" to an endpoint to capture and transfer additional debug data to Home Assistant during testing.  

I'll be adding support for this device to my testing to capture and log data to a microSD card
 
https://www.sparkfun.com/sparkfun-qwiic-openlog.html 

Some of these tests have to run for hours or days to expose intermittent issues

def1149, 2025-04-09:
Additional failure recorded as directed with Error level logging enabled, see ### annotations
`
14:07:33.952 -> Cycle: 814 Restart: 5
14:07:33.999 -> Starting Zigbee
14:07:34.276 -> Started in 323 mS
14:07:34.276 -> Connecting to network
14:07:34.276 -> 
14:07:34.311 -> Successfully connected to Zigbee network in 0 mS
14:07:35.291 -> Check for Button
14:07:35.291 -> Measure 
14:07:35.291 -> Report 
14:07:35.291 -> Reported 1, temperature: 45.00°C, Humidity: 0.00%
14:07:35.510 -> Going to sleep nowCycle: 815 Restart: 5
14:07:50.532 -> Starting Zigbee                                          ####
14:07:52.323 -> [0;31mE (7724) ESP_ZIGBEE_CORE: In BDB commissioning, an error occurred (for example: the device has already been running)[0m 
14:07:52.371 -> Started in 1836 mS
14:07:52.371 -> Connecting to network
14:07:52.371 -> 
14:07:52.371 -> Successfully connected to Zigbee network in 0 mS
14:07:53.333 -> Check for Button
14:07:53.333 -> Measure 
14:07:53.333 -> Report 
14:07:53.333 -> Reported 1, temperature: 45.50°C, Humidity: 0.00%
14:07:53.549 -> Going to sleep now
14:08:08.525 -> Cycle: 816 Restart: 5
14:08:08.525 -> Starting Zigbee
14:08:08.883 -> Started in 320 mS
14:08:08.883 -> Connecting to network
14:08:08.883 -> 
14:08:08.883 -> Successfully connected to Zigbee network in 0 mS
14:08:09.850 -> Check for Button
14:08:09.850 -> Measure 
14:08:09.888 -> Report 
14:08:09.888 -> Reported 1, temperature: 46.00°C, Humidity: 0.00%
14:08:10.062 -> Going to sleep now
14:08:25.085 -> Cycle: 817 Restart: 5
14:08:25.085 -> Starting Zigbee
14:08:26.123 -> Started in 1057 mS
14:08:26.123 -> Connecting to network
14:08:26.123 -> 
14:08:26.123 -> Successfully connected to Zigbee network in 0 mS
14:08:27.129 -> Check for Button
14:08:27.129 -> Measure 
14:08:27.129 -> Report 
14:08:27.129 -> Reported 1, temperature: 46.50°C, Humidity: 0.00%
14:08:27.358 -> Going to sleep now
14:08:42.352 -> Cycle: 818 Restart: 5
14:08:42.352 -> Starting Zigbee
14:08:47.395 -> [ 10047][E][ZigbeeCore.cpp:51] begin(): ZigbeeCore begin failed or timeout    ####
14:08:47.395 -> Zigbee failed to start!
14:08:47.395 -> Going to sleep now for 30 seconds
14:09:22.279 -> Cycle: 819 Restart: 5
14:09:22.279 -> Starting Zigbee
14:09:27.336 -> [ 10047][E][ZigbeeCore.cpp:51] begin(): ZigbeeCore begin failed or timeout    ####
14:09:27.336 -> Zigbee failed to start!
14:09:27.336 -> Going to sleep now for 30 seconds
14:10:02.203 -> Cycle: 820 Restart: 5
14:10:02.203 -> Starting Zigbee
14:10:04.717 -> Started in 2518 mS
14:10:04.717 -> Connecting to network
14:10:04.717 -> 
14:10:04.717 -> Successfully connected to Zigbee network in 0 mS
14:10:05.703 -> Check for Button
14:10:05.703 -> Measure 
14:10:05.703 -> Report 
14:10:05.703 -> Reported 1, temperature: 47.00°C, Humidity: 0.00%
14:10:05.925 -> Going to sleep now
14:10:20.885 -> Cycle: 821 Restart: 5
14:10:20.928 -> Starting Zigbee
14:10:21.224 -> Started in 313 mS
14:10:21.224 -> Connecting to network
14:10:21.224 -> 
14:10:21.224 -> Successfully connected to Zigbee network in 0 mS
14:10:22.245 -> Check for Button
14:10:22.245 -> Measure 
14:10:22.245 -> Report 
14:10:22.245 -> Reported 1, temperature: 47.50°C, Humidity: 0.00%
14:10:22.439 -> Going to sleep now
``
A similar-looking graphical view of data loss

[image]

P-R-O-C-H-Y, 2025-04-10:
I cannot see the reason of failure which is described in the issue header. But in this last log you send, it seems that the Zigbee does not want to start. Are you using Zigbee debug libs for EndDevice or normal once? If not, can you please use the ones with Debug, so maybe we can have more logs.

But overall, 2 crashes during 3 hours of reporting every 30s, its approx 0,5% mistake. Which to be honest is not bad.

def1149, 2025-04-10:
There may be multiple causes for a failure to report. My strategy will shift from trying to identify and correct the cause to detecting and recovering as best I can. It's likely that any H/W, library, or my code issues will be sorted out over time. Thanks

def1149, 2025-04-10:
@P-R-O-C-H-Y After running more tests and spending time digging through the ZHA debug log I have three instances where no errors were produced in the EP application but three datapoints in a row were missed in Home Assistant. The missing values are not in the ZHA log. 

The test app ( based on the sleepy temperature sensor example) is sending temperature values about every 16 seconds. 
These values are artificially created starting at -30 and continuing up to 80 by adding 0,5 for each cycle. I do this because Home Assistant will not record duplicates received sequentially. A temperature sensor chip could produce duplicate values. And I chart the received in HA so missing values are easy to spot. See previous entries in this case. The ZHA logs are very detailed and I don't understand most of it, but did find data from my device Nwk ID x0544. 

I've attached an annotated ZHA debug log using >>>> to mark the start of key entries 

At about 15:23:05 in the log is an entry that starts a series of records  for temperature value 59.5  
The next reported temperature value occurs starting at about 15:24:08 with a value 61.5.  
Reports for values 60.0,  60.5, and 61.0 are missing 

I expect that wireless protocols will experience data loss, it but as far as I understand it  Zigbee uses a protocol that uses ACKS to confirm receipt of data, so I would expect an error to be signaled on the EP if it doesn't get an ACK from the coordinator.

[m4152305.log](https://github.com/user-attachments/files/19695210/m4152305.log)

def1149, 2025-04-13:
After a lot of tweaking of EP code for my ESP32-C6 Zigbee temperature sensor to detect and capture problems at key points (to minimize any logging impact), and sending log data via serial to separate  ESP32 with an SD card. I've found three categories of data missing in HA. 

1. HA does not record successive duplicate values. When this happens, I see the last seen date/time update.
2. Occasionally the EP fails to connect. I added intelligent retries with retry interval backoff to not flood the air with messages to prevent message storms.
3. The EP thinks it delivered the data, but it's missing in HA and there's no trace of the transaction in the ZHA debug logs, or there's signs of some kind error but the EP did not see an error. Not much I can do to investigate this with my minimal Zigbee stack and ZHA knowledge.

The occasional glitching  when the C6 wakes up from deep sleep  appears to be due to something that happens in the startup code when Verbose logging is enabled.  I didn't see it when log level was reduced to Error.

At this point, I can create reliable enough Zigbee/ZHA sensors using the ESP32- C6/H2 so I'm replacing my power hungry Wi-Fi sensors  and MQTT.

closed 2025-04-13 by def1149