espressif/arduino-esp32 #11241: ESP32-C6 Actual I2C frequency not change, always 100KHz

bhcuong2008, 2025-04-12:
### Board

ESP32-C6-DevKitC-1

### Device Description

ESP32-C6-DevKitC-1

### Hardware Configuration

GPIO 23 for SDA, 22 for SCL.

### Version

latest stable Release (if not listed below)

### IDE Name

Arduino IDE

### Operating System

Ubuntu 22

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

512000

### Description

Using Wire Scan example. I set frequency 10Khz but actual waveform is still 100Khz. Change to 400Khz, it still keeps 100Khz. Using Wire.getClock() shows the same freq as setting.

### Sketch

```cpp
#include "Wire.h"

void setup() {
  Serial.begin(115200);
  Wire.begin(23, 22, 10000);
}

void loop() {
  byte error, address;
  int nDevices = 0;

  delay(5000);
  Serial.printf("Wire freq: %u\n", Wire.getClock());
  Serial.println("Scanning for I2C devices ...");
  for (address = 0x01; address < 0x7f; address++) {
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
    if (error == 0) {
      Serial.printf("I2C device found at address 0x%02X\n", address);
      nDevices++;
    } else if (error != 2) {
      Serial.printf("Error %d at address 0x%02X\n", error, address);
    }
  }
  if (nDevices == 0) {
    Serial.println("No I2C devices found");
  }
}
```

### Debug Message

```plain
Not show any errors/warnings.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

bhcuong2008, 2025-04-12:
But with Wire Master example, its frequency could be changed. It's strange.

lbernstone, 2025-04-12:
Please verify which version of arduino-esp32 you are using, as this driver is currently going through changes.

bhcuong2008, 2025-04-13:
I use esp-arduino 3.2.

FrankS71, 2025-04-22:
Have you tried Wire.setClock(); ? This is working fine for me.

TD-er, 2025-04-22:
Just looking at the code for ESP32-classic and ESP32-C6.

ESP32-classic:

https://github.com/espressif/esp-idf/blob/186f2a8929d2a2ecbc773fe62fe20dc936241f01/components/hal/esp32/include/hal/i2c_ll.h#L86-L96

ESP32-C6:
https://github.com/espressif/esp-idf/blob/186f2a8929d2a2ecbc773fe62fe20dc936241f01/components/hal/esp32c6/include/hal/i2c_ll.h#L90-L126

The code for ESP32-classic is very basic and easy to understand.
However the code for the -C6 is really odd and I still have no clue what it should do.

Take for example this line:
```c++
    uint32_t clkm_div = source_clk / (bus_freq * 1024) + 1;
```
Since nearly all I2C clock frequencies are >= 100 kHz, this `clkm_div` will likely always be 1? (unless the `source_clk` > 102.4 MHz)
In test code, this seems to be the [xtal_freq](https://github.com/espressif/esp-idf/blob/186f2a8929d2a2ecbc773fe62fe20dc936241f01/components/hal/test_apps/hal_i2c/components/hal_i2c/hal_i2c.c#L137-L143)
This is always < 80 MHz.

This [same line](https://github.com/search?q=repo%3Aespressif%2Fesp-idf+i2c_ll_master_cal_bus_clk&type=code) is present for:
- ESP32-C2
- ESP32-C5
- ESP32-C3
- ESP32-C61
- ESP32-H2
- ESP32-H21
- ESP32-P4
- ESP32-S3

Only the ESP32-classic and the ESP32-S2 seem to do it differently.

TD-er, 2025-04-22:
OK, just hooked up an ESP32-C6 to the logic analyzer....

400 kHz

[image]

200 kHz

[image]

100 kHz

[image]

bhcuong2008, 2025-04-23:
@TD-er , you run my test code above? If I run normal code of I2C, such as writing, reading then I could change I2C freq. Only scan code above is fixed at 100KHz.

bhcuong2008, 2025-04-23:
> Have you tried Wire.setClock(); ? This is working fine for me.

I've already tried. But it is still fixed at 100KHz for my scan code above. For other readings, writings, it could be changed to any frequency I set.

TD-er, 2025-04-23:
> [@TD-er](https://github.com/TD-er) , you run my test code above? If I run normal code of I2C, such as writing, reading then I could change I2C freq at any value within its possible range. Only scan code above is fixed at 100KHz.

Nope, I ran my own project code.
Just to make sure there isn't something seriously wrong with the Arduino/IDF code.

I'll have a look at your code to see if I see something obviously different compared to how I initialize Wire

TD-er, 2025-04-23:
> > Have you tried Wire.setClock(); ? This is working fine for me.
> 
> I've already tried. But it is still fixed at 100KHz for my scan code above. For other readings, writings, it could be changed to any frequency I set.

Hmm scan code running at 100 kHz is some really good feature, so you can be sure to also find those older devices not capable of running at > 100 kHz.
So this might also be a specific feature to limit the clock to 100 kHz on scanning.

TD-er, 2025-04-23:
Just checked my code...
This is what I use to init Wire...
```c++
    Wire.setPins(sda, scl);
    Wire.begin();
    Wire.setClock(clockFreq);
```
This differs from what you're doing, so maybe you can try it using this sequence?

me-no-dev, 2025-04-23:
Here is the answer: https://github.com/espressif/esp-idf/blob/master/components/esp_driver_i2c/i2c_master.c#L1295-L1300

The new IDF driver always does scan at 100kHz. Closing the issue :)

closed 2025-04-23 by me-no-dev

bhcuong2008, 2025-04-24:
@me-no-dev thank you for your clarification.