espressif/arduino-esp32 #11243: ESP NOW compilation issue of

yonagy82, 2025-04-13:
### Board

XIAO ESP32-C3

### Device Description

XIAO ESP32-C3 stand alone (wifi communication)

### Hardware Configuration

XIAO ESP32-C3 stand alone (wifi communication)

### Version

v3.1.3

### IDE Name

Arduino IDE Version 2.3.5

### Operating System

Windows 10

### Flash frequency

N/A

### PSRAM enabled

no

### Upload speed

9600

### Description

Hello, friends,
I know that the issue was discussed multiple times. I read the solutions, but none had worked for me. So, regretfully' I have to place the question ove again.

I'm using Aarduino IDE Version 2.3.5, The library is ESP-NOW version 3.1.3. The ESP_NOW sub library is Unicast transmission by Lucas Saavedra Vaz - 2024, downloaded from https://github.com/espressif/arduino-esp32/. It is stored on my computer at c:\Users\USER\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-2f7dcd86-v1\esp32c3/include/esp_wifi/include/esp_now.h

I tried several code variants and combunations of them:
```
void OnDataRecv(const uint8_t *mac_addr, const uint8_t *incomingData, int len) 
void OnDataRecv(const esp_now_recv_info_t *info, const uint8_t *incomingData, int len) 
OnDataRecv(const esp_now_recv_info_t *info, const uint8_t *incomingData, int len)
```
and registrations:
```
esp_now_register_recv_cb(OnDataRecv);
esp_now_register_recv_cb(esp_now_recv_cb_t(OnDataRecv));
```
The error message is consistent with small variants:
` invalid use of non-static member function 'void (* espNowServerClass::OnDataRecv(const uint8_t*, const uint8_t*, int))(const esp_now_recv_info_t*, const uint8_t*, int),` with or without the suffix` (did you forget the '()'?)`

When using the non-void version of the method `OnDataRecv` I've noticed another issue: any macro returned by the method invoked two notices: 
1. When using the value ESP_ERR_ESPNOW_NOT_INIT - warning from esp_now.h: `pointer to arithmetic expression` 
2. When using the value LOW or 1- error from esp_now.h: ` invalid conversion from 'int' to 'esp_now_recv_cb_t' {aka 'void (*)(const esp_now_recv_info*, const unsigned char*, int)'} [-fpermissive]`
In both cases the was a note from the .ino - `in expansion of macro '...'`.

I'm already on this for quite some time; should I downgrade the library? to which revision? What are the functions or nugs that will be lost? How to do it?
Best regards
Yona

### Sketch

```cpp
N/A - the module is stand alone
```

### Debug Message

```plain
See the description
```

### Other Steps to Reproduce

N/A - before using the ESP32 module

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-04-14:
Please provide minimal (full) example that we can try to compile and debug. Maybe also have a look at our own ESP-NOW library that is included in the core

yonagy82, 2025-04-14:
Hello'
Thank you very much for your reply.

I am attaching my complete code.
1. The file espNowServer.h is a reduced H file. Compiling this sketch shows
that the only error is about this line.
2. Renaming espNowServer.h.txt by removing the .txt will yield the full
code code. (in this case the file espNowCommon.h completes the declarations
with those common to both sides..
   2.1  The only error is for line 131 in the file espNowServer.h.txt.
   2.2  Commenting line 131 there will yield a no-error compilation.

Basically, I started from the application "
https://randomnerdtutorials.com/esp-now-auto-pairing-esp32-esp8266/", but
rewrothe the pairing mechanism so it is done only once, during the
non-server stations powering. This is because my application maintains a
timely keep-alive data messaging.

There are other parts that are not completed yet, but this issue  is
holding my project.

Best regards

#ifndef espNowServer_h
#define espNowServer_h

#include "espNowCommon.h"

class espNowServerClass {
private:
  #define loopbackFlag
  #define ackInterval 100
  #define ESP_NOW_VERBOSE

  int channel = 1;
  unsigned int previousMessageNum;
  bool acknowledgeReceived;

  /*********************************************************************************/
  bool addPeer(const uint8_t *incomingData) {  // Server side
    structPairing peer;
    memcpy(&peer, incomingData, sizeof(peer));
    if ((peer.peerId > maxClientsCount) || (peer.peerId == 0)) {
#ifdef ESP_NOW_VERBOSE
      Serial.print("Corrupted ID: " + String(peer.peerId));
#endif
      return LOW;
    }
    // Send pairing acknowledge
    peer.msgType = PAIR_ACK;
    peer.channel = channel;
    for (uint8_t i = 0; i < macAddrLen; i++)
      peer.macAddr[i] = serverAddr[i];
    bool sucs = (esp_now_send(peer.macAddr, (uint8_t *)&peer, sizeof(peer)) == ESP_NOW_SEND_SUCCESS);
    uint8_t id = peer.peerId;
    if (sucs) {  // Register the paired slave in the peersList array.
      memcpy(&peersList[id].macAddr, peer.macAddr, macAddrLen);
      peersList[id].missesCount = 0;
      peersList[id].pairingStatus = PAIR_PAIRED;
    } else
      peersList[id].pairingStatus = NOT_PAIRED;
#ifdef ESP_NOW_VERBOSE
    Serial.print("A message of type PAIRING from station ");
    Serial.print(id);
    Serial.print(" at mac address ");
    for (uint8_t i = 0; i < macAddrLen; i++) {
      Serial.print(peer.macAddr[i], HEX);
      Serial.print(":");
    }
    Serial.print("; Pairing status: ");
[... 250 lines cut ...]
#endif  // espNewCommon_h

/**************************  Server spare code ************************************

  WiFi.mode(WIFI_AP_STA);      // Set the device as a Station and Soft Access Point simultaneously
  WiFi.begin(ssid, password);  // Set device as a Wi-Fi Station
  chan = WiFi.channel();

  WiFi.mode(WIFI_STA);
  WiFi.STA.begin();
  Serial.print("Waiting for wifi connection: ");
  while (WiFi.status() != WL_CONNECTED){
    delay(1000);
    Serial.print(".");
  }
  Serial.println("\nconnected to network");

Serial.print("Station IP Address: ");
Serial.println(WiFi.localIP());
Serial.print("Wi-Fi Channel: ");
Serial.println(WiFi.channel());
Serial.print("Server ESP NOW init:  ");
Serial.println(WiFi.softAPmacAddress());
Serial.print("Server SOFT AP MAC Address:  ");
*/

Jason2866, 2025-04-14:
The used examples looks like are not adopted to the changed implementation in Arduino Core 3.x and are conflicting. You can try to remove the provided ESP NOW libs provided in Arduino Core and use only the code you posted / linked. The API calls and `esp_now.h` from IDF is there in Arduino Core and working. Using ESP NOW in project Tasmota this way and it is verified working.

yonagy82, 2025-04-14:
OK, got it. I isolated the following objects from esp now libraries, and
I'll imitate them: 'esp_now_init';'esp_now_send'; 'ESP_NOW_SEND_SUCCESS';
'esp_wifi_get_mac'; 'esp_wifi_set_channel';

The most urgent, and, actually, those I'm not familiar with, for me are the
call back registration
methods 'esp_now_register_recv_cb' and 'esp_now_recv_cb_t' .
Will you be so kind as to send an implementation example of these methods?

Best regards
Yona

Jason2866, 2025-04-14:
This is github issues, not a place to discuss personal coding questions. Anyways here is an example of an esp now driver used in Tasmota ahttps://github.com/arendst/Tasmota/blob/development/tasmota/tasmota_xdrv_driver/xdrv_57_9_tasmesh.ino

yonagy82, 2025-04-16:
Dear friend,

Thank you for the reference. However, I still think that this is a package issue. 

 

As you can see in the attached file, I continued the experiments. The code was shrunk to almost the bare minimum. Lines 34 to 37 contain variety of cb registrations. The errors reported in lines 53 to 69 of the same file, show that the issue is only in lines 34 and 36, both are registration variants of onDataReceived, and none to the other registrations in lines 35 and 37. I assume there is some vulnerability of this registration, which causes so many troubles to so many developers.  

 

Unfortunately, I didn’t find online explanation of the lower level of the events handling, and I don’t know what is behind the syntax of these lines, so I can’t further analyze them. I’d appreciate your help here.

 

Regards

Yona

lbernstone, 2025-04-16:
https://github.com/espressif/esp-now/tree/master/examples.  The source code is there as well.
https://docs.espressif.com/projects/esp-idf/en/v5.3.3/esp32c3/api-reference/network/esp_now.html

yonagy82, 2025-04-17:
Thank you. Do you have examples for Arduino IDE?

Regards
Yona

yonagy82, 2025-05-05:
Hello,
Thank you very much for your support so far.

I’m still trying to struggle with the ESP NOW. I solved the previous issue (it appears that the callback methods can’t be embedded in a class). Now I get an error message from the send callback method.  Lack of on-depth understanding of the ESP NOW methods makes the debugging more difficult because the behavior is unpredictable. Is there a document that explain the ESP NOW internals?

Best regards
Yona

lbernstone, 2025-05-05:
[Arduino-esp32 doco](https://docs.espressif.com/projects/arduino-esp32/en/latest/api/espnow.html)
[ESP-IDF doco](https://docs.espressif.com/projects/esp-idf/en/v5.3.3/esp32c3/api-reference/network/esp_now.html)
The [code](https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP_NOW/src/ESP32_NOW.h) is generally the best reference

yonagy82, 2025-05-05:
Thank you very much for the link. I have had this, but I need a more
in-depth understanding of the interdependencies between the data structures
and relationships between the methods. Especially regarding the callback
mechanism.

Regards
Yona

closed 2025-10-02 by Jason2866