espressif/arduino-esp32 #11244: BT device not connecting via A2DP

Admsher, 2025-04-13:
### Board

ESP32 One

### Device Description

Using BLE A2DP with Wifi on ESP32 Chip ( As mentioned in ESP IDF)

### Hardware Configuration

SD card  reader and OV2460 Camera connected.

### Version

latest stable Release (if not listed below)

### IDE Name

ESP IDF v5.3

### Operating System

WIndows 11

### Flash frequency

40 Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I am trying to connect an A2DP speaker to my ESP32 one board using the arduino-audio-tools lbrary, i tried to connect with Arduino IDE At first and i worked fine, but now for some reason the device doesnt connect even when there are no errors in build or runtime itself and flashes properly. I have tried resetting the speaker itself but no luck there.

### Sketch

```cpp
#define HELIX_LOGGING_ACTIVE false
#include "AudioTools.h"
#include "AudioTools/AudioLibs/A2DPStream.h"
#include "AudioTools/Disk/AudioSourceSDFAT.h"
#include "AudioTools/AudioCodecs/CodecMP3Helix.h"

const char *startFilePath="/music/";
const char* ext="mp3";
AudioSourceSDFAT source(startFilePath, ext, 15);
A2DPStream out;
MP3DecoderHelix decoder;
AudioPlayer player(source, out, decoder);

void setup() {
  Serial.begin(115200);
  AudioToolsLogger.begin(Serial, AudioToolsLogLevel::Warning);
  // setup player
  // Setting up SPI if necessary with the right SD pins by calling 
  SPI.begin(14, 12, 13, 15);
  player.setVolume(0.8);
  player.begin();

  // setup output - We send the test signal via A2DP - so we conect to a Bluetooth Speaker
  auto cfg = out.defaultConfig(TX_MODE);
  cfg.silence_on_nodata = true; // prevent disconnect when there is no audio data
  cfg.name = "CLIP 5";  // set the device here. Otherwise the first available device is used for output
  //cfg.auto_reconnect = true;  // if this is use we just quickly connect to the last device ignoring cfg.name
  out.begin(cfg);
}

void loop() {
  player.copy();
}
```

### Debug Message

```plain
/home/admsher/Pictures/Screenshots/Screenshot from 2025-04-12 22-22-32.png
```

### Other Steps to Reproduce

You can go through the configuration for the flags in this link :https://github.com/pschatzmann/ESP32-A2DP/wiki/Espressif-IDF-as-a-Component
One thing i would also tell you to go through this issue as well for configuring the HAL files: https://github.com/espressif/arduino-esp32/pull/11167

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

closed 2025-04-14 by me-no-dev

Admsher, 2025-04-14:
I am actually using esp32-arduino as a component so that I could directly import the Arduino ide code here , so thought it would be a relevant issue

Jason2866, 2025-04-14:
@Admsher To solve you can use pioarduino to compile Arduino as an component of IDF.
I made an example how to setup pschatzmann Audio tools here https://github.com/Jason2866/audio-tools-test/tree/issue11181

Admsher, 2025-04-14:
So just to be on the same track, should I use piarduino as a component for my esp-idf project?

Jason2866, 2025-04-14:
Yes, the only working way to compile Arduino BT projects as a component of IDF is using pioarduino.