espressif/arduino-esp32 #11245: Simultaneous USB CDC and MSC

PaulaScharf, 2025-04-14:
### Board

sensebox_mcu_esp32s2

### Device Description

[schematics](https://github.com/sensebox/senseBox-MCU-S2-ESP32S2/blob/main/hardware/2.1/SenseBox-MCU-2-1.pdf) in case thats relevant

### Hardware Configuration

no

### Version

latest stable Release (if not listed below)

### IDE Name

Arduino IDE 2.3.6

### Operating System

Ubuntu

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

In version 2.0.17 of arduino-esp32 we used to be able to use both MSC and CDC. Since version 3.0.0 or newer we have issues with that.

If only either MSC or CDC are enabled they each work fine.

However, when both CDC and MSC are enabled:​

- We are able to receive Serial output through the serial monitor​
- When trying to copy bin-files on it the device appears as read-only

Do we need to change anything in our boards.txt definition perhaps? Or could it be an issue with our TinyUF2?

```
##############################################################
# senseBox MCU ESP32-S2

sensebox_mcu_esp32s2.name=senseBox MCU-S2 ESP32-S2
sensebox_mcu_esp32s2.vid.0=0x303A
sensebox_mcu_esp32s2.pid.0=0x81B8
sensebox_mcu_esp32s2.vid.1=0x303A
sensebox_mcu_esp32s2.pid.1=0x81B9
sensebox_mcu_esp32s2.vid.2=0x303A
sensebox_mcu_esp32s2.pid.2=0x81BA

sensebox_mcu_esp32s2.bootloader.tool=esptool_py
sensebox_mcu_esp32s2.bootloader.tool.default=esptool_py

sensebox_mcu_esp32s2.upload.tool=esptool_py
sensebox_mcu_esp32s2.upload.tool.default=esptool_py
sensebox_mcu_esp32s2.upload.tool.network=esp_ota

sensebox_mcu_esp32s2.upload.maximum_size=1310720
sensebox_mcu_esp32s2.upload.maximum_data_size=327680
[... 152 lines cut ...]
Any insight appreciated. Thank you for this nice repo.

### Sketch

```cpp
void setup() {
  Serial.begin(115200);
}

void loop() {
  Serial.println("hello");
  delay(500);
}
```

### Debug Message

```plain
> Error while copying to 'disk'. 
The destination is read-only
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Thiemann96, 2025-05-09:
Commenting the said lines worked for me ! Thanks for the headaches 🙃

Jason2866, 2025-05-13:
Hi Paula, great investigation and finding. Seems weak defines are lottery game.
This [one](https://github.com/espressif/arduino-esp32/issues/11181) is another example for.

PR #11353 (2025-06-10): remove weak function declaration of tud_msc_is_writable_cb to fix simultaneous CDC and MSC
## Description of Change
Solves issue #11245 through the [workaround suggested by nitz](https://github.com/espressif/arduino-esp32/issues/11245#issuecomment-2842952196).

## Tests scenarios
I have tested the proposed changes with a [senseBox MCU ESP32S2](https://github.com/espressif/arduino-esp32/tree/master/variants/sensebox_mcu_esp32s2) and another microcontroller with an ESP32S3 with the following script:
```cpp
void setup() {
  Serial.begin(115200);
}

void loop() {
  Serial.println("hello");
  delay [... cut]