espressif/arduino-esp32 #11247: A quick question about Arduino IDE

tvixen, 2025-04-14:
### Board

ESP32C6

### Device Description

I have 2 ESP32S3 and 2 ESP32C6, 

### Hardware Configuration

uint8_t led = RGB_BUILTIN;
uint8_t button = BOOT_PIN;

### Version

v3.2.0

### IDE Name

Arduino IDE 1.8.19

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I use the the Arduino compiler IDE version 1.8.19, and is working very fine with ESP32S3 and all of my code using I2C, SPI, Wifi etc.
But when I use the ESP32C6 in Zigbee mode as end device, it seem to struggle a bit. It compiles fine with the examples in the library and uploads fine til the board. But it dosent seem to connect to my Zigbee gateway.

Any one who can clarify this for me? 

### Sketch

```cpp
// Copyright 2024 Espressif Systems (Shanghai) PTE LTD
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at

//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

/**
 * @brief This example demonstrates Zigbee Color Dimmable light bulb.
 *
 * The example demonstrates how to use Zigbee library to create an end device with
 * color dimmable light end point.
 * The light bulb is a Zigbee end device, which is controlled by a Zigbee coordinator.
 *
 * Proper Zigbee mode must be selected in Tools->Zigbee mode
 * and also the correct partition scheme must be selected in Tools->Partition Scheme.
 *
 * Please check the README.md for instructions and more detailed description.
 *
 * Created by Jan Procházka (https://github.com/P-R-O-C-H-Y/)
 */

#ifndef ZIGBEE_MODE_ED
#error "Zigbee end device mode is not selected in Tools->Zigbee mode"
#endif

#include "Zigbee.h"

/* Zigbee color dimmable light configuration */
#define ZIGBEE_RGB_LIGHT_ENDPOINT 10
uint8_t led = RGB_BUILTIN;
uint8_t button = BOOT_PIN;

ZigbeeColorDimmableLight zbColorLight = ZigbeeColorDimmableLight(ZIGBEE_RGB_LIGHT_ENDPOINT);

/********************* RGB LED functions **************************/
void setRGBLight(bool state, uint8_t red, uint8_t green, uint8_t blue, uint8_t level) {
  if (!state) {
    rgbLedWrite(led, 0, 0, 0);
    return;
  }
  float brightness = (float)level / 255;
  rgbLedWrite(led, red * brightness, green * brightness, blue * brightness);
}

// Create a task on identify call to handle the identify function
void identify(uint16_t time) {
  static uint8_t blink = 1;
  log_d("Identify called for %d seconds", time);
  if (time == 0) {
    // If identify time is 0, stop blinking and restore light as it was used for identify
    zbColorLight.restoreLight();
    return;
  }
  rgbLedWrite(led, 255 * blink, 255 * blink, 255 * blink);
  blink = !blink;
}

/********************* Arduino functions **************************/
void setup() {
  Serial.begin(115200);

  // Init RMT and leave light OFF
  rgbLedWrite(led, 0, 0, 0);

  // Init button for factory reset
  pinMode(button, INPUT_PULLUP);

  // Set callback function for light change
  zbColorLight.onLightChange(setRGBLight);

  // Optional: Set callback function for device identify
  zbColorLight.onIdentify(identify);

  // Optional: Set Zigbee device name and model
  zbColorLight.setManufacturerAndModel("Espressif", "ZBColorLightBulb");

  // Add endpoint to Zigbee Core
  Serial.println("Adding ZigbeeLight endpoint to Zigbee Core");
  Zigbee.addEndpoint(&zbColorLight);

  // When all EPs are registered, start Zigbee in End Device mode
  if (!Zigbee.begin()) {
    Serial.println("Zigbee failed to start!");
    Serial.println("Rebooting...");
    ESP.restart();
  }
  Serial.println("Connecting to network");
  while (!Zigbee.connected()) {
    Serial.print(".");
    delay(100);
  }
  Serial.println();
}

void loop() {
  // Checking button for factory reset
  if (digitalRead(button) == LOW) {  // Push button pressed
    // Key debounce handling
    delay(100);
    int startTime = millis();
    while (digitalRead(button) == LOW) {
      delay(50);
      if ((millis() - startTime) > 3000) {
        // If key pressed for more than 3secs, factory reset Zigbee and reboot
        Serial.println("Resetting Zigbee to factory and rebooting in 1s.");
        delay(1000);
        Zigbee.factoryReset();
      }
    }
    // Increase blightness by 50 every time the button is pressed
    zbColorLight.setLightLevel(zbColorLight.getLightLevel() + 50);
  }
  delay(100);
}
```

### Debug Message

```plain
No debug info
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

tvixen, 2025-04-15:
I can certainly do that, but I actually just wanted to know if I could continue using Arduino IDE 1.8.19 or if I should switch to IDE 2.0 before it will work with Zigbee?

As I worte, IDE 1.8.19 is working fine with ESP32S3.

P-R-O-C-H-Y, 2025-04-15:
> I can certainly do that, but I actually just wanted to know if I could continue using Arduino IDE 1.8.19 or if I should switch to IDE 2.0 before it will work with Zigbee?
> 
> As I worte, IDE 1.8.19 is working fine with ESP32S3.

The IDE 1.8.19 is working. I am using that actually.

tvixen, 2025-04-15:
ok thanks for that. I will provide you the requested info asap.

P-R-O-C-H-Y, 2025-04-15:
I see in the logs that after many tries it has successfully joined the network:
```
4:27:24.703 -> ......................................[207626][I][ZigbeeCore.cpp:308] esp_zb_app_signal_handler(): Joined network successfully (Extended PAN ID: 4b:a0:61:84:42:21:a4:e3, PAN ID: 0x04a6, Channel:20, Short Address: 0x1a00)
```

tvixen, 2025-04-15:
Yes, but my Tuya app says "no new devices found"

Im a supposed to delete the old device in the Tuya app before adding a new function to the Zigbee device?

P-R-O-C-H-Y, 2025-04-15:
For sure, when flashing new firmware to ESP, remove the device first , flash and add again.
Also set Erase Flash before upload to Enabled in the IDE Tools menu. 

But TBH I am not sure if the ESP is supposed to work under Tuya app. I did not see anyone trying that before.