espressif/arduino-esp32 #11253: Upload Error: ESP32-C6 UART Upload Error w/ Arduino Update

mitch-gres, 2025-04-15:
### Board

ESP32-C6

### Device Description

Device: ESP32C6 Development Module
Accessories: Board is attached to a Custom-PCB but with none powered. eg. It's all passive elements. 
 

### Hardware Configuration

GPIO9 was previously connected to a  switch that is now disconnected which leaves it to float. 
All other GPIO Pin are connected with passive elements.

I can show parts of the design upon request but this is a company project so I can't just upload all the files for the design. 

### Version

v3.1.1

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40 MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Unable to upload via the UART Port on the ESP32C6. Previously had thought it was an issue with the boot mode but after attempts to put it into the right boot mode by either holding down the BOOT button or holding the BOOT button then pressing RESET and then releasing the BOOT button I've been unable to get Arduino IDE to upload anything to the board. 

The results I get are...
```
esptool.py v4.8.1
Serial port COM22
Connecting......................................
```
I've tried with other modules that I've had around and have had no issue uploading via the UART on the device. When I looked at the Serial Output from the device that's having the issue I got the following...
```
E (639) rmt: rmt_new_tx_channel(298): register channel failed

[   327][E][esp32-hal-rmt.c:548] rmtInit(): GPIO 9 - RMT TX Initialization error.

[   556][E][esp32-hal-rmt.c:128] _rmtGetBus(): ==>_rmtWrite():GPIO 9 is not attached to an RMT channel.

[   674][E][esp32-hal-rmt.c:128] _rmtGetBus(): ==>_rmtWrite():GPIO 9 is not attached to an RMT channel.

=========== After Setup Start ============

INTERNAL Memory Info:

------------------------------------------

  Total Size        :   397740 B ( 388.4 KB)

  Free Bytes        :   359572 B ( 351.1 KB)

  Allocated Bytes   :    31056 B (  30.3 KB)

  Minimum Free Bytes:   358536 B ( 350.1 KB)

  Largest Free Block:   327668 B ( 320.0 KB)

------------------------------------------

GPIO Info:

------------------------------------------

  GPIO : BUS_TYPE[bus/unit][chan]

  --------------------------------------  

    10 : RMT_TX

    16 : UART_TX[0]

    17 : UART_RX[0]

    18 : RMT_TX

============ After Setup End =============

[  2484][E][esp32-hal-rmt.c:128] _rmtGetBus(): ==>_rmtWrite():GPIO 9 is not attached to an RMT channel.
```
As you will see the code that I'm trying to put onto the device has nothing to do with allocating the GPIO9 pin to do anything and I'm confused why GPIO10 is even referenced. The only change made before this was updating to a new firmware version as prompted by Arduino. One confusion I have is whether the update that I installed on this board was meant for the C6 as when I opened Arduino it seemed like it was trying to do a firmware update on a C3 when I had the C6 connected. 

Steps I've taken....
* I've successfully uploaded to other identical boards with UART. 
* I've disconnected GPIO9 from any peripheral on the board by taking off the switch is was previously connected with. 
* I've successfully uploaded via the USB Port but nothing with regards to Serial output - even when I try to Serial.print(...) on UART. 

Thanks!

### Sketch

```cpp
//
//    FILE: TCA9555_digitalWrite.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: test TCA9555 library
//     URL: https://github.com/RobTillaart/TCA9555

#include "TCA9555.h"

TCA9555 TCA(0x27);

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("TCA9555_LIB_VERSION: ");
  Serial.println(TCA9555_LIB_VERSION);
  Serial.println();

  Wire.begin();
  TCA.begin();

  Wire.setClock(50000);

  Serial.println("Set pinMode1 OUTPUT");
  for (int i = 0; i < 16; i++)
  {
    TCA.pinMode1(i, OUTPUT);
    TCA.write1(i, LOW);
  }

  Serial.println("TEST write1(0)");
  for (int i = 0; i < 16; i++)
  {
    TCA.write1(0, i % 2);  //  alternating HIGH/LOW
    Serial.print(i % 2);
    Serial.print('\t');
    delay(250);
  }
  Serial.println();
  Serial.println();

  Serial.println("TEST write1(pin)");
  for (int pin = 0; pin < 16; pin++)
  {
    TCA.write1(pin, 1 - pin % 2);  //  alternating HIGH/LOW
    Serial.print(1 - pin % 2);
    Serial.print('\t');
  }
  Serial.println();
  Serial.println();

  Serial.println("TEST read back");
  for (int pin = 0; pin < 16; pin++)
  {
    int val = TCA.read1(pin);
    Serial.print(val);
    Serial.print('\t');
  }
  Serial.println();
  Serial.println("\ndone...");
}

void loop()
{
}

//  -- END OF FILE --
```

### Debug Message

```plain
E (639) rmt: rmt_new_tx_channel(298): register channel failed

[   327][E][esp32-hal-rmt.c:548] rmtInit(): GPIO 9 - RMT TX Initialization error.

[   556][E][esp32-hal-rmt.c:128] _rmtGetBus(): ==>_rmtWrite():GPIO 9 is not attached to an RMT channel.

[   674][E][esp32-hal-rmt.c:128] _rmtGetBus(): ==>_rmtWrite():GPIO 9 is not attached to an RMT channel.

=========== After Setup Start ============

INTERNAL Memory Info:

------------------------------------------

  Total Size        :   397740 B ( 388.4 KB)

  Free Bytes        :   359572 B ( 351.1 KB)

  Allocated Bytes   :    31056 B (  30.3 KB)

  Minimum Free Bytes:   358536 B ( 350.1 KB)

  Largest Free Block:   327668 B ( 320.0 KB)

------------------------------------------

GPIO Info:

------------------------------------------

  GPIO : BUS_TYPE[bus/unit][chan]

  --------------------------------------  

    10 : RMT_TX

    16 : UART_TX[0]

    17 : UART_RX[0]

    18 : RMT_TX

============ After Setup End =============

[  2484][E][esp32-hal-rmt.c:128] _rmtGetBus(): ==>_rmtWrite():GPIO 9 is not attached to an RMT channel.
```

### Other Steps to Reproduce

I've been unable to reproduce on other DevKits.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2025-04-15:
Hold the boot pin low and reset.  Then erase the flash using `esptool.exe erase_flash`

mitch-gres, 2025-04-15:
@lbernstone Thanks for the help. 

I've tried what you've said on both the UART and USB Ports. On the USB Port the flash erase was successful but the UART still is having issues with uploading. The following is the error that I'm seeing. 

```
C:\windows\system32>esptool --chip esp32c6 --port COM22 erase_flash
esptool.py v4.8.1
Serial port COM22
Connecting......................................
```

Yet the flash erase was successful on the USB. 
```
esptool.py v4.8.1
Serial port COM21
Connecting...
Chip is ESP32-C6 (QFN40) (revision v0.1)
Features: WiFi 6, BT 5, IEEE802.15.4
Crystal is 40MHz
MAC: f0:f5:bd:ff:fe:1a:a2:9c
BASE MAC: f0:f5:bd:1a:a2:9c
MAC_EXT: ff:fe
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 1.3s
Hard resetting via RTS pin...
```

I don't think it's a RMT Channel issue anymore since I was able to remove a library that I think was causing the issue meaning that it's no longer throwing that issue and in this case successfully printing out "Hello World" but still unable to upload via UART regardless.

lbernstone, 2025-04-16:
Sounds like the UART RX pin (gpio17) is not connected.

SuGlider, 2025-04-20:
@mitch-gres - please update the Arduino Core to 3.2.0 and try again. Let me know. Thanks.