espressif/arduino-esp32 #11256: class WiFiMulti' has no member named 'setAllowOpenAP'

rudiratlos, 2025-04-17:
### Board

ESP32-S3

### Device Description

m5pagerS3

### Hardware Configuration

no

### Version

latest stable Release (if not listed below)

### IDE Name

Arduino IDS

### Operating System

macos

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

115200

### Description

/EPDWifi.h:121:11: error: 'class WiFiMulti' has no member named 'setAllowOpenAP'
 WiFimulti.setAllowOpenAP(WiFi_useOpenAP);
           ^~~~~~~~~~~~~~

### Sketch

```cpp
#include <WiFi.h>
#include <WiFiMulti.h>
...
void WiFi_enable() {
  WiFi.mode(WIFI_STA); // client mode
  WiFi.disconnect();
  WiFi.setHostname(c_ProgName.c_str());
  WiFimulti.setAllowOpenAP(true);
  WiFi_multAdd();
}
```

### Debug Message

```plain
x
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2025-04-17:
Works for me; likely an environmental problem.  Make sure the compiler is using the WiFiMulti.h inside the esp32 directory, not an Arduino version.  If you installed manually, make sure you have run get.exe.

rudiratlos, 2025-04-17:
which library is that. I have used Arduino_MultiWiFi. after deleting it, I've used WiFiMulti_Generic, but also no success.
Where can I find this library, which offers setAllowOpenAP. Can you supply github reference pls.

lbernstone, 2025-04-18:
[WiFiMulti](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiMulti.h) is part of the arduino-esp32 base.  You don't need to install any additional library, and actually need to uninstall any other WiFiMulti.h.

rudiratlos, 2025-04-18:
no success
I get this output:
`Dependency Graph
|-- M5Unified @ 0.2.5
|-- I2C BM8563 RTC @ 1.0.4
|-- M5GFX @ 0.2.6+sha.d7fea53
|-- Time @ 1.6.1
|-- SolarCalculator @ 2.0.1
|-- minIniFS @ 1.0.1
|-- ArduinoJson @ 7.4.1
|-- SD @ 2.0.0
|-- WiFi @ 2.0.0
|-- HTTPClient @ 2.0.0
Building in release mode`

Error:
`Compiling .pio/build/esp32-s3-devkitm-1/src/main.cpp.o
In file included from src/main.cpp:44:
src/EPDWifi.h: In function 'void WiFi_enable()':
src/EPDWifi.h:121:11: error: 'class WiFiMulti' has no member named 'setAllowOpenAP'
 WiFimulti.setAllowOpenAP(WiFi_useOpenAP);
           ^~~~~~~~~~~~~~
In file included from src/main.cpp:52:`

my platformio.ini within vcode:

`[env:esp32-s3-devkitm-1]
;platform = espressif32
platform = platformio/espressif32
board = esp32-s3-devkitm-1
framework = arduino
board_build.partitions = default_16MB.csv
board_upload.flash_size = 16MB
board_upload.maximum_size = 16777216
board_build.arduino.memory_type = qio_opi
monitor_speed = 115200
build_flags =
    -DBOARD_HAS_PSRAM
    -DARDUINO_USB_CDC_ON_BOOT=1
	-DARDUINO_USB_MODE=1
lib_deps =
    m5stack/M5Unified
    tanakamasayuki/I2C BM8563 RTC  
    guerratron/minIniFS
    paulstoffregen/Time
    jpb10/SolarCalculator
    guerratron/minIniFS
    bblanchon/ArduinoJson
    M5GFX=https://github.com/m5stack/M5GFX.git#develop
`
include of my sketch:
`#include <M5Unified.h>
#undef USE_M5_FONT_CREATOR
#include "Data.h"
#include "WiFi.h"
#include "WiFiMulti.h"
#include "time.h" 
#include "Utils.h"
#include "NMEA.h"`

rudiratlos, 2025-04-19:
If I click on go to definition, I get the following info:
It's code for esp8266. Function setAllowOpenAP is not included. How to setup platformio, to get the right S3 library?

 * @file WiFiMulti.cpp
 * @date 16.05.2015
 * @author Markus Sattler
 *
 * Copyright (c) 2015 Markus Sattler. All rights reserved.
 * This file is part of the esp8266 core for Arduino environment.

lbernstone, 2025-04-19:
Remove anything from your lib directory that is included in the [platform libraries](https://github.com/espressif/arduino-esp32/tree/master/libraries)- WiFi, WiFiMulti and HTTPClient. Your environment seems damaged, and I can't know what you did to get there.  You may need to wipe out the .pio directory and start over. The WiFiMulti in esp32 still has the old info from esp8266 (@SuGlider )

rudiratlos, 2025-04-20:
I have removed platformlib and all in .pio. But it installs the old esp8266 version.

in my plattformio.ini:
platform = espressif32
board = esp32-s3-devkitm-1
framework = arduino

lbernstone, 2025-04-20:
If you had properly filled out the template information you would have saved us both a lot of time.  Add this to your platformio.ini:
`platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip`