espressif/arduino-esp32 #11259: Absturz verschiedener ESP32, seit Version 3.2.0 + Fehlerhafte Ansteuerung von Neopixel

schneeer, 2025-04-18:
### Board

ESP32-D0WD-V3 (revision v3.1) = ESP32-S, Aber auch ESP32-WROOM...

### Device Description

ESP32-S von AZ-Delivery,
montiert auf einer Lochstreifenplatine,
Über Pegelwandler ist ein Atmel328P angebunden, dieser empfängt Infrarot-Ferbedienungs-Befehle.
Zusätzlich gibt es einen Helligkeitssensor und ein Potentiometer.
Weiterhin sind mehrere Neopixel-LEDs angeschlossen.
Der ESP32 mit der restlichen Hardware und die Neopixel werden getrennt mit Spannung versorgt.
Alle Lötartbeiten wurden sorgfältig durchgeführt.

Tests habe ich auch mit einem ESP32 Dev Module ohne angeschlossene Hardware durchgeführt.

### Hardware Configuration

(BuildinLED an 2)
Potentiometer an 36
Lichtsensor (Licht-Widerstand) an 39
Atmega328P am Hardware-I2C (als Empfänger für IR-Signale (Fernbedienung)) ESP32 ist Master
Atmega328P-Reset-Pin an 32 (als Hardware-Reset zu Atmega328P)
20x Neopixel an 16

### Version

v3.1.3

### IDE Name

Arduino IDE 2.3.5 und 2.3.6 (" verschiedene Notebooks)

### Operating System

Windows 11 Pro

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921600

### Description

Mein ESP lief mehrere Monate fehlerfrei.
Ich habe verschiedene Sensoren angeschlossen und steuere aber hauptsächlich Neopixel an.
Zur Veränderung der Neopixel macht der ESP32 einen AP auf und es ist eine Webseite hinterlegt.
Bis Version 3.1.3 hatte ich keine Probleme.
1. Der ESP32 steuert die Neopixel nicht mehr an, läuft aber sonst.
2. Er läuft aber nur solange, wie nicht versucht wird sich per WLAN zum AP zu verbinden; beim Verbindungsversuch stürzt der ESP sofort ab.
ein Zurückwechseln von 3.2.0 auf 3.1.3 beseitigt sofort alle Probleme.

### Sketch

```cpp
// Benötigte Bibliotheken
#include <Arduino.h>
#include <EEPROM.h> 
// I2C Bibliothek zur Kommunikation zwischen den Boards
#include <Wire.h>
//Wifi
#include <WiFi.h>
#include <WiFiClient.h>
[... 1779 lines cut ...]
14:50:26.858 -> assert failed: vTaskPriorityDisinheritAfterTimeout tasks.c:5267 (pxTCB != pxCurrentTCBs[ xPortGetCoreID() ])
14:50:26.858 -> 
14:50:26.858 -> 
14:50:26.858 -> Backtrace: 0x4008259c:0x3ffb2010 0x4008d011:0x3ffb2030 0x40093261:0x3ffb2050 0x4008ff72:0x3ffb2190 0x4008d887:0x3ffb21b0 0x400e1ab2:0x3ffb21f0 0x400e19af:0x3ffb2210 0x400d93ad:0x3ffb2230 0x400d9dbf:0x3ffb2250 0x400e6984:0x3ffb2270 0x4008dc96:0x3ffb2290
14:50:26.903 -> 
14:50:26.903 -> 
14:50:26.903 -> 
14:50:26.903 -> 
14:50:26.903 -> ELF file SHA256: ed09114b4
14:50:26.903 -> 

Die letzte Meldung wiederholt sich dann immer wieder.
```

### Other Steps to Reproduce

Ich habe mehrere verschiedene ESP32 probiert.
Ich habe mehrere Rechner zum kompilieren probiert.
Ich habe verschiedene Versionen der Arduino IDE probiert.
Alles ohne Erfolg.
Erfolg hate ich nur beim Downgrade der Arduino ESP32 Boarderweiterung zurück auf 3.1.3.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2025-04-19:
Maybe same as #11166

schneeer, 2025-04-19:
> Maybe same as [#11166](https://github.com/espressif/arduino-esp32/issues/11166)

Ok und danke, ich schaue dort vorbei.

schneeer, 2025-04-19:
> > Vielleicht dasselbe wie [#11166](https://github.com/espressif/arduino-esp32/issues/11166)
> 
> Ok und danke, ich schaue dort vorbei.

Ich habe, wie dort angegeben, die Datei "build_opt.h" erstellt.
Das hilft mir leider nicht. Der  ESP32 stürzt immer noch ab, wenn versucht wird eine Verbindung zum AP aufzubauen.

me-no-dev, 2025-04-19:
Closing this because it is not in English. Feel free to start a new issue that is written and communicated in English

closed 2025-04-19 by me-no-dev

schneeer, 2025-04-19:
Super Gleichberechtigung hier. Das ist mal wirklich gelebte Diversität.
Jeder vernünftige Browser übersetzt das in Echtzeit.
Dann schließt eben andersprachige Menschen hier aus.
Schämt euch!

schneeer, 2025-04-19:
Ich habe das wegen Diskriminierung Github gemeldet.
Wenn das hier häufiger vorkommt und gemeldet wird, nehmen die ganz nett Kontrakt zu euch auf.