espressif/arduino-esp32 #11262: Version 3.2.0 breaks old libraries (Arduino PS4Controller)

mnemocron, 2025-04-19:
### Board

AI Thinker ESP32-CAM

### Device Description

ESP32-S

### Hardware Configuration

-

### Version

v3.2.0

### IDE Name

Arduino 3.2.6

### Operating System

Windows 10

### Flash frequency

-

### PSRAM enabled

yes

### Upload speed

-

### Description

Arduino Sketches using the [PS4_Controller ](https://github.com/pablomarquez76/PS4_Controller_Host) library compile ok with ESP32 version 3.1.3. When using 3.2.0 the compilation process generates errors while compiling the library.

### Sketch

```cpp
// based on Examples > PS4Controller > PS4ReceiveData.ino
#include <PS4Controller.h>

void setup() {
  Serial.begin(115200);
  PS4.begin("11:22:33:AA:BB:CC");  // <-- explicitly setting a custom MAC
  Serial.println("Ready.");
}

void loop() {
  // Below has all accessible outputs from the controller
  if (PS4.isConnected()) {
    if (PS4.Right()) Serial.println("Right Button");
    // etc...
  }
}
```

### Debug Message

```plain
c:\Users\<user>\Documents\Arduino\libraries\PS4Controller\src\ps4.c: In function 'ps4Deinit':
c:\Users\<user>\Documents\Arduino\libraries\PS4Controller\src\ps4.c:59:5: error: implicit declaration of function 'spp_deinit'; did you mean 'sppInit'? [-Wimplicit-function-declaration]
59 | spp_deinit();
| ^~~~~~~~~~
| sppInit
c:\Users\<user>\Documents\Arduino\libraries\PS4Controller\src\ps4.c: In function 'ps4SetBluetoothMacAddress':
c:\Users\<user>\Documents\Arduino\libraries\PS4Controller\src\ps4.c:244:3: error: implicit declaration of function 'esp_base_mac_addr_set' [-Wimplicit-function-declaration]
244 | esp_base_mac_addr_set(baseMac);
| ^~~~~~~~~~~~~~~~~~~~~
exit status 1

Compilation error: exit status 1
```

### Other Steps to Reproduce

- compiles with version 3.1.3
- fails with version 3.2.0

see also discussion here: https://forum.arduino.cc/t/compilar-library-ps4-controler/1373660

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

schneeer, 2025-04-19:
Willkommen im Club.

https://github.com/espressif/arduino-esp32/issues/11259

lbernstone, 2025-04-19:
The upstream APIs can change in major/minor releases.  You will need to have the library author update their code to match.

me-no-dev, 2025-04-19:
This is not connected to Arduino, but rather changes in APIs in ESP-IDF v5.4. External libraries that use ESP-IDF APIs would need to be updated in order to continue to function.

closed 2025-04-19 by me-no-dev