espressif/arduino-esp32 #11281: RMT standard example  RMT_LED_Blink crashes

fanfanlatulipe26, 2025-04-23:
### Board

ESP32 Dev Kit v1

### Device Description

nothing

### Hardware Configuration

nothing
Using the builtin LED on pin 2

### Version

v3.2.0

### IDE Name

Arduino IDE 1.8.19

### Operating System

Windows

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

In fact I use esp32 3.2.0 but with a patched file  core/esp32/esp32-hal-rmt.c (PR #11203 )
The test craches after some blinks.

### Sketch

```cpp
The exact example provided with ESP32 Arduino distribution. Not  line changed.
```

### Debug Message

```plain
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4888
load:0x40078000,len:16516
load:0x40080400,len:4
load:0x40080404,len:3476
entry 0x400805b4
[    17][D][esp32-hal-cpu.c:263] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32
  Package           : D0WD-Q6
  Revision          : 1.00
  Cores             : 2
  CPU Frequency     : 240 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000032
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : Yes
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   374904 B ( 366.1 KB)
  Free Bytes        :   334240 B ( 326.4 KB)
  Allocated Bytes   :    33560 B (  32.8 KB)
  Minimum Free Bytes:   328456 B ( 320.8 KB)
  Largest Free Block:   110580 B ( 108.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         :  4194304 B (4 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00290000, size:  1408.0 KB, type: DATA, subtype: SPIFFS
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Apr 23 2025 13:49:50
  Compile Host OS   : windows
  ESP-IDF Version   : v5.4.1-1-g2f7dcd862a-dirty
  Arduino Version   : 3.2.0
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32_DEV
  Arduino Variant   : esp32
  Arduino FQBN      : esp32:esp32:esp32:JTAGAdapter=default,PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=debug,EraseFlash=none,ZigbeeMode=default
============ Before Setup End ============
Starting Blink testing...
===> rmtInit OK! Tick = 2.5us - OK for testing

======================================
All set. Starting RMT testing Routine.
======================================

===> rmtWriteLooping() to Blink the LED.
Blinking at 1s on + 1s off :: 3 blinks
===> rmtWrite() (Blocking Mode) to Blink the LED.
Blinking at 500ms on + 500ms off :: 4 blinks
===> rmtWriteAsync() (Non-Blocking Mode) to Blink the LED.
Blinking at 250ms on + 250ms off :: 5 blinks
Blinking OFF for 1 seconds
End of Mixed Calls testing

===============================
Starting a Blinking sequence...
===============================

=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   374904 B ( 366.1 KB)
  Free Bytes        :   330144 B ( 322.4 KB)
  Allocated Bytes   :    37096 B (  36.2 KB)
  Minimum Free Bytes:   324360 B ( 316.8 KB)
  Largest Free Block:   110580 B ( 108.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
     1 : UART_TX[0]
     2 : RMT_TX
     3 : UART_RX[0]
============ After Setup End =============
Using RMT Async Writing and its Completion to blink an LED.
Blinking at 1s on + 1s off :: 5 blinks
Blinking at 500ms on + 500ms off :: 5 blinks
Blinking at 250ms on + 250ms off :: 5 blinks
Blinking OFF for 1 seconds
Using RMT Writing loop to blink an LED.
Blinking at 1s on + 1s off :: 3 blinks
Blinking at 500ms on + 500ms off :: 5 blinks
Blinking at 250ms on + 250ms off :: 10 blinks
Blinking OFF for 2 seconds
Using RMT Writing and its Completion to blink an LED.
Blinking at 1s on + 1s off :: 2 blinks
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40084582  PS      : 0x00060f30  A0      : 0x80084640  A1      : 0x3ffb2100  
A2      : 0x3ffb8e9c  A3      : 0x3ffb8e9c  A4      : 0x61a861a8  A5      : 0x0000000f  
A6      : 0x3ffb92b8  A7      : 0x00000000  A8      : 0x00000010  A9      : 0x00000000  
A10     : 0x3ff5683c  A11     : 0x3ff56800  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000040  A15     : 0x3ffb9388  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000018  LBEG    : 0x40086480  LEND    : 0x40086496  LCOUNT  : 0xffffffff  

Backtrace: 0x4008457f:0x3ffb2100 0x4008463d:0x3ffb2120 0x40084756:0x3ffb2150 0x400e0b79:0x3ffb2180 0x400d22a2:0x3ffb21d0 0x400d2332:0x3ffb2210 0x400d1b11:0x3ffb2230 0x400d1c59:0x3ffb2250 0x400d3290:0x3ffb2270 0x40089895:0x3ffb2290

ELF file SHA256: 9ac58ff70

Rebooting...
ets Jun  8 2016 00:22:57
```

### Other Steps to Reproduce

Fail in the main loop.
It seems that each function allone  is OK ( RMT_Write_Aync_Non_Blocking_Blink(); / RMT_Loop_Write_Blink(); /RMT_Single_Write_Blocking_Blink();)
But when executed one after each other in a row there is a crash.
Trace:
```

PC: 0x40084582
EXCVADDR: 0x00000018

Decoding stack results
0x4008457f: rmt_tx_mark_eof at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_driver_rmt/src/rmt_tx.c line 636
0x4008463d: rmt_encode_check_result at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_driver_rmt/src/rmt_tx.c line 674
0x40084756: rmt_tx_do_transaction at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_driver_rmt/src/rmt_tx.c line 753
0x400e0b79: rmt_transmit at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_driver_rmt/src/rmt_tx.c line 594
0x400d22a2: _rmtWrite at C:\Users\Francis\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\cores\esp32\esp32-hal-rmt.c line 347
0x400d2332: rmtWrite at C:\Users\Francis\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\cores\esp32\esp32-hal-rmt.c line 405
0x400d1b11: RMT_Single_Write_Blocking_Blink() at C:\Users\Francis\AppData\Local\Temp\arduino_modified_sketch_770913/RMT_LED_Blink.ino line 281
0x400d1c59: loop() at C:\Users\Francis\AppData\Local\Temp\arduino_modified_sketch_770913/RMT_LED_Blink.ino line 358
0x400d3290: loopTask(void*) at C:\Users\Francis\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\cores\esp32\main.cpp line 74
0x40089895: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c line 139
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2025-04-23:
@fanfanlatulipe26 - issue confirmed. Analysis in progress.

SuGlider, 2025-04-24:
The issue was found. The problem was related to RMT Loop Mode when the sketch stops this infinite loop mode and tries to use a different mode. It only happens with the ESP32. The original example works correctly for other SoC because those support `SOC_RMT_SUPPORT_TX_LOOP_AUTO_STOP`, which is not supported by the ESP32.

The PR #11292 fixes the example to work as supposed using the ESP32.

Please try the new example version and let me know.

fanfanlatulipe26, 2025-04-24:
Very fast fix ;-)
I made the test with the new example only on a ESP32. Perfect. No crash
(and also good timing )
Thanks

PR #11292 (2025-04-28): feat(rmt): fixes example to run correctly within IDF 5.x
## Description of Change
This example crashes when running with the ESP32 because it doesn't support `SOC_RMT_SUPPORT_TX_LOOP_AUTO_STOP` feature. All other SoC were running the example correctly.

The timming was also incorrect becuase it was sending one RMT Symbol less.

This PR fixes the example to run with no crashes for all the SoC supported by Arduino Core 3.2.x+
It also fixes the blink timming as verified by a logic analizer. 

## Tests scenarios
Tested using the example with ESP32, S2, S3, C3 and C6.

## Related links
Closes #1128 [... cut]
closed 2025-04-28 by me-no-dev