espressif/arduino-esp32 #11282: 3.2.0 with IDF 5.4.1 build fails

kungfupizza, 2025-04-23:
### Board

ESP32S3

### Device Description

Build fails when usingESP-IDF v5.4.1 with arduino-esp32 3.2.0

Error

```
/managed_components/espressif__arduino-esp32/libraries/WiFi/src/WiFiType.h:28:10: fatal error:
 esp_wifi_types.h: No such file or directory
   28 | #include "esp_wifi_types.h"
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
```

### Hardware Configuration

ESP32S3

### Version

v3.2.0

### IDE Name

VSCode

### Operating System

Windows 11

### Flash frequency

80M

### PSRAM enabled

yes

### Upload speed

115200

### Description

Error in building
```

/managed_components/espressif__arduino-esp32/libraries/WiFi/src/WiFiType.h:28:10: fatal error:
 esp_wifi_types.h: No such file or directory
   28 | #include "esp_wifi_types.h"
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
```

### Sketch

```cpp
void setup()
{

}

void loop
{

}
```

### Debug Message

```plain
/managed_components/espressif__arduino-esp32/libraries/WiFi/src/WiFiType.h:28:10: fatal error:
 esp_wifi_types.h: No such file or directory
   28 | #include "esp_wifi_types.h"
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
```

### Other Steps to Reproduce

Just build

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

kungfupizza, 2025-04-23:
Similar build error on

```
D:/Work/Spashta/spashta-speaks/managed_components/espressif__arduino-esp32/libraries/ESP_SR/src/ESP_SR.h:11:10: fatal error: ESP_I2S.h: No such file or directory
   11 | #include "ESP_I2S.h"

```
This is when I don't have ESP-SR disabled.

Jason2866, 2025-04-23:
Just compiled Arduino branch master as an component of IDF 5.4.1 successfully

kungfupizza, 2025-04-24:
I was using this [option](https://github.com/espressif/arduino-esp32/blob/master/Kconfig.projbuild#L255) earlier. Builds fine as soon as I deselect it.