espressif/arduino-esp32 #11285: WiFi.macAddress() returns 00:00:00:00:00:00 before WiFi.begin()

sans-ltd, 2025-04-23:
### Board

ESP32-Mini 1U

### Device Description

-

### Hardware Configuration

-

### Version

v3.1.1

### IDE Name

VSCode

### Operating System

Linux

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

- When calling `WiFi.macAddress()` before `WiFi.begin()`, we cannot get the MAC address. 
- With Arduino 2, it worked because it used `esp_read_mac()` as fallback, if WiFi was not initialized yet. 
- If we call `WiFi.begin()`, our station does not work correctly, we cannot see the WiFi with another device.
- Reading the MAC address with `esp_read_mac(mac, ESP_MAC_WIFI_STA)` works fine though, without initializing the WiFi first.

### Sketch

```cpp
setup()
{
  auto mac = WiFi.macAddress();
  // mac is 00:00:00:00:00:00
  // do some magic to generate an SSID
  WiFi.softAP(name.c_str(), passwd.c_str());
  ...
}
```

### Debug Message

```plain
-
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-04-23:
The network layer has changed. You can call the following APIs to get the STA MAC without connecting to anything
```cpp
WiFi.STA.begin();
auto mac = WiFi.STA.macAddress();
```

me-no-dev, 2025-04-23:
To get the base MAC address, you can also do directly:
```cpp
auto mac = Network.macAddress();
```

sans-ltd, 2025-04-23:
Yes OK, thanks for the hints. But it is quite unintuitive and breaks with Arduino 2.0 in the behavior.

me-no-dev, 2025-04-23:
v3 is a major version update that does break some APIs. It is for the best though. Especially in the Network layer :)

> And it is a real use case to create the SSID from the MAC address when setting up a station, where you normally would not call begin.

Best to use `Network.macAddress()` in that case to get the ESP's unique base MAC, from which all other MACs are derived

closed 2025-04-23 by me-no-dev