espressif/arduino-esp32 #11287: ESP32-HS is not entering normal execution mode after power on

Mqxx, 2025-04-23:
### Board

ESP32-H2

### Device Description

Currently I use this Board without any attachments: https://de.aliexpress.com/item/1005007941405877.html

### Hardware Configuration

None

### Version

latest stable Release (if not listed below)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

64MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

During ZigBee development, I encountered an issue where the ESP does not enter normal execution mode after power on on its own. Previously the ESP showed up as an "Unknown Device" on the Device Manager on Windows, but now when I plug it in it just instantly goes into flash mode, where I can program it. Normally before that I needed to hold down the boot button and then power it on to enter flash mode.

How can I reverse this behavior? Is there a way I can completely wipe everything on the ESP and reset it to "the original state"?

### Sketch

```cpp
#include "ZigbeeCore.h"
#include "ep/ZigbeeWindowCovering.h"

#define ZIGBEE_COVERING_ENDPOINT 10
#define BUTTON_PIN 9  // ESP32-C6/H2 Boot button
#define UP_PIN 10
#define STOP_PIN 11
#define DOWN_PIN 12

ZigbeeWindowCovering zbCovering = ZigbeeWindowCovering(ZIGBEE_COVERING_ENDPOINT);

void setup() {
  Serial.begin(115200);

  // Init button for factory reset
  pinMode(BUTTON_PIN, INPUT_PULLUP);
  pinMode(UP_PIN, OUTPUT);
  pinMode(STOP_PIN, OUTPUT);
  pinMode(DOWN_PIN, OUTPUT);

  // Optional: set Zigbee device name and model
  zbCovering.setManufacturerAndModel("ESMART", "Professional MIMOTO");

  // Set proper covering type
  zbCovering.setCoveringType(PROJECTOR_SCREEN);

  // Set configuration (keine Positionssteuerung)
  zbCovering.setConfigStatus(
    true,  // operational
    true,  // online
    false, // not command reversed
    false, // lift closed loop
    false, // tilt closed loop
    false, // lift encoder controlled
    false  // tilt encoder controlled
  );

  // Set mode: keine Positionssteuerung
  zbCovering.setMode(false, false, false, false);

  // Deaktiviere Lift- und Tilt-Positionslimits
  zbCovering.setLimits(0, 100, 0, 0); // Keine Limits für Positionen

  // Set callback functions for open, close, and stop
  zbCovering.onOpen(fullUp);
  zbCovering.onClose(fullDown);
  zbCovering.onStop(stopMotor);

  // Entferne Callbacks für Positionssteuerung
  // zbCovering.onGoToLiftPercentage(nullptr); // Deaktiviere Positionssteuerung
  // zbCovering.onGoToTiltPercentage(nullptr); // Deaktiviere Tilt-Steuerung

  zbCovering.setLiftPercentage(0);
  zbCovering.setTiltPercentage(0);

  // Add endpoint to Zigbee Core
  Serial.println("Adding ZigbeeWindowCovering endpoint to Zigbee Core");
  Zigbee.addEndpoint(&zbCovering);

  // Start Zigbee
  Serial.println("Calling Zigbee.begin()");
  if (!Zigbee.begin()) {
    Serial.println("Zigbee failed to start!");
    Serial.println("Rebooting...");
    ESP.restart();
  }
  Serial.println("Connecting to network");
  while (!Zigbee.connected()) {
    Serial.print(".");
    delay(100);
  }
  Serial.println();
}

void loop() {
  // Factory reset handling
  if (digitalRead(BUTTON_PIN) == LOW) {
    delay(100);
    int startTime = millis();
    while (digitalRead(BUTTON_PIN) == LOW) {
      delay(50);
      if ((millis() - startTime) > 3000) {
        Serial.printf("Resetting Zigbee to factory settings, reboot.\n");
        Zigbee.factoryReset();
        delay(30000);
      }
    }
  }
  delay(500);
}

void triggerPin(uint8_t pin) {
  digitalWrite(pin, HIGH);
  delay(10); // Kurzer Impuls für Motorsteuerung
  digitalWrite(pin, LOW);
}

void fullUp() {
  triggerPin(UP_PIN);
}

void fullDown() {
  triggerPin(DOWN_PIN);
}

void stopMotor() {
  triggerPin(STOP_PIN);
}
```

### Debug Message

```plain
ESP-ROM:esp32h2-20221101
Build:Nov  1 2022
rst:0xc (SW_CPU),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400031b6
SPIWP:0xee
mode:DIO, clock div:1
load:0x408460e0,len:0x1108
load:0x4083cad0,len:0xca4
load:0x4083efd0,len:0x2db4
entry 0x4083cad0
[0;31mE (322) ZB OSIF: Zigbee lock is not ready![0m
Zigbee stack assertion failed zcl/zcl_general_commands.c:608

abort() was called at PC 0x420121b5 on core 0
Core  0 register dump:
MEPC    : 0x408015f6  RA      : 0x40806b4a  SP      : 0x4081d6b0  GP      : 0x4080e3c4  
TP      : 0x4081d830  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x4081d6dc  S1      : 0x4081d6dc  A0      : 0x4081d6e8  A1      : 0x4081d6ca  
A2      : 0x00000000  A3      : 0x4081d715  A4      : 0x00000001  A5      : 0x40815000  
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x00000000  S3      : 0x0000000a  
S4      : 0x00000102  S5      : 0x00000001  S6      : 0x00000003  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000002  MTVAL   : 0x00000000  
MHARTID : 0x00000000 

4081d710: 0x2065726f 0x00000030 0x00000000 0xf2f2d849 0x0000002f 0x0000ffff 0x42072859 0x420121b8
4081d730: 0x00000001 0x4081d79c 0x42073810 0x00000260 0x00000000 0x00000000 0x00000003 0x42024b52
4081d750: 0x00000000 0x00000000 0x00000001 0x00000102 0x00000000 0x4080fe4e 0x4080fdf0 0x4202307a
4081d770: 0x00000000 0x00001800 0x00000002 0x00000000 0x40850000 0x00000000 0x00000000 0x00000000
4081d790: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x40810000 0x4080fdf0 0x42004f22
4081d7b0: 0x00000000 0x40810000 0x40810000 0x420015de 0x00000000 0x408160c0 0x4080fdf0 0x42001512
4081d7d0: 0x00000000 0x40810000 0x40810000 0x42000130 0x00000000 0x00000000 0x00000000 0x42004050
4081d7f0: 0x00000000 0x00000000 0x00000000 0x40807cba 0x00000000 0x00000000 0x00000000 0x00000000
4081d810: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4081d830: 0xa5a5a5a5 0xbaad5678 0x00000168 0xabba1234 0x0000015c 0x4081d270 0x00000000 0x40812b6c
4081d850: 0x40812b6c 0x4081d844 0x40812b64 0x00000018 0x00000000 0x00000000 0x4081d844 0x00000000
4081d870: 0x00000001 0x4081b834 0x706f6f6c 0x6b736154 0x00000000 0x00000000 0x4081d830 0x00000005
4081d890: 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x0000030c 0x00000000 0x408157b4
4081d8b0: 0x4081581c 0x40815884 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000
4081d8d0: 0x40011484 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081d8f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081d910: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081d930: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081d950: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081d970: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081d990: 0x00000000 0x00000000 0x00000000 0x00000000 0xbaad5678 0x00000068 0xabba1234 0x0000005c
4081d9b0: 0x00000000 0x4081d9b0 0x00000000 0x00000000 0x00000000 0x4081d9c8 0xffffffff 0x4081d9c8
4081d9d0: 0x4081d9c8 0x00000000 0x4081d9dc 0xffffffff 0x4081d9dc 0x4081d9dc 0x00000001 0x00000001
4081d9f0: 0x00000000 0x9f00ffff 0x00000000 0x0a887658 0x0985d401 0x0b3f6dff 0xaff9ecee 0xbaad5678
4081da10: 0x00000088 0xabba1234 0x0000007c 0x00000000 0x00000014 0x4081e274 0x00000000 0x00000000
4081da30: 0x00000000 0x00000000 0x00000000 0x4081daa8 0x00000000 0x00000101 0x00000000 0x00000000
4081da50: 0x0000000a 0x4081e1d0 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081da70: 0x00000000 0x00000000 0x4081dcb8 0x4081de24 0x00000000 0x4081e020 0x4081dfb4 0x4081e164
4081da90: 0x4081e0f8 0x4081e08c 0xbaad5678 0x0000020c 0xabba1234 0x00000200 0x00000000 0x00000000

ELF file SHA256: 154b5ced9

Rebooting...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2025-04-23:
erase flash with [esptool.py](https://docs.espressif.com/projects/esptool/en/latest/esp32/)

Mqxx, 2025-05-04:
Got it working now. For some reason the behavior did not change after erasing the flash but at least I can now "reprogram" the ESP 👍

closed 2025-05-04 by Mqxx