espressif/arduino-esp32 #11290: Enable USB host callback

rjjrbatarao, 2025-04-24:
### Related area

Usb ethernet esp32 s3

### Hardware specification

ESP32 S3 Devkitc

### Is your feature request related to a problem?

im trying to enable a feature to use usb ethernet on esp32 s3 and CONFIG_USB_HOST_ENABLE_ENUM_FILTER_CALLBACK is a requirement to be enabled, maybe it can be enabled in the next release? thank you

### Describe the solution you'd like

edit sdkconfig
CONFIG_USB_HOST_ENABLE_ENUM_FILTER_CALLBACK=y

### Describe alternatives you've considered

using esp-idf 

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2025-04-29:
The only way to change a skdconfig parameter is by building an Arduino project as an IDF Component.
In that case, it is possible to change any IDF configuration using `idf.py menuconfig`.

https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html
https://www.youtube.com/watch?v=8w3MmsZBa9o
Examples at https://github.com/SuGlider/devcon2024

Regarding changing `CONFIG_USB_HOST_ENABLE_ENUM_FILTER_CALLBACK`, it is necessary to investigate the possible implications to the current USB Client driver of the ESP32 Arduino Framework, which is based on TinyUSB code.
I'd suggest sending this same question to the https://github.com/hathach/tinyusb/issues which would be the right place for it.

rjjrbatarao, 2025-07-24:
I did what @SuGlider suggested and I think its the only way unless the maintainers enable this by default. ESP32 S3 has storage options of 8, 16 and even 32Mb flash an I see no reason they dont enable this by default only on esp32 s3. Im using it now for usb ethernet and many more usb device could be made available if they enable this.

jetpax, 2025-07-25:
Unfortunately 'my' Arduino project is Tasmota that I am adding 4G modem support to. Its a fairly  complex build...