espressif/arduino-esp32 #11291: ADC vs WiFi

smartes, 2025-04-24:
### Board

ESP32-s3-wroom1

### Device Description

ESP32-s3-wroom1

### Hardware Configuration

GPIO5 ADC 

### Version

v3.2.0

### IDE Name

Arduino IDE 2.3.4

### Operating System

Ubuntu 24

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Seems in version 3.2 something has been changed with adc continuous mode. I have ADC on GPIO5 and I want to have WiFi, which is activated by pressing GPIO0. Now when I try to set up wifi I get "Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)." 
In v3.1.0 this sketch works with no issues.

### Sketch

```cpp
#include <WiFi.h>
#include <ESPmDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

const char* ssid = "SSID";
const char* password = "Password";
IPAddress ip(192, 168, 4, 1);
IPAddress netmask(255, 255, 255, 0);
const int port = 8080;

adc_continuous_data_t *adcresult = NULL;
uint8_t adc_pins[] = { 6 };  
volatile bool adc_coversion_done = false;
void ARDUINO_ISR_ATTR adcComplete() {
  adc_coversion_done = true;
}
bool OTA=0;
void setup()
      {analogContinuousSetWidth(12);
      analogContinuousSetAtten(ADC_0db);
  
      analogContinuous(adc_pins, 1, 5, 40000, &adcComplete);
  
      analogContinuousStart();
     
      }
void loop()
      {if(digitalRead(0)==0)
          {OTA_init();

          }
      if(OTA)
          ArduinoOTA.handle();

        
      if (adc_coversion_done == true)
        {adc_coversion_done = false;
        if (analogContinuousRead(&adcresult, 0)) 
          {
          analogContinuousStop();       
          uint32_t C_V = adcresult[0].avg_read_mvolts;  ///CPU_BAT          
          analogContinuousStart();
          }
        }
      
      }
void OTA_init() {
  WiFi.softAPConfig(ip, ip, netmask);
  WiFi.softAP(ssid, password);
  WiFi.mode(WIFI_AP);
  ArduinoOTA.setHostname(ssid);
  ArduinoOTA.begin();
  OTA=1;
}
```

### Debug Message

```plain
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed). 
Backtrace: 0x42002b9f:0x3fc99e00 0x403791ae:0x3fc99e20 0x40379233:0x3fc99e70 0x4037b7e8:0x3fc99e90 0x403796d5:0x3fc99ec0 0x40376394:0x3fca4440 0x403765cf:0x3fca4460 0x4037e8be:0x3fca4490

Decoding stack results
0x403791ae: adc_dma_intr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_adc/adc_continuous.c line 90
0x40379233: adc_dma_in_suc_eof_callback at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_adc/gdma/adc_dma.c line 15
0x4037b7e8: gdma_default_rx_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/dma/gdma.c line 829
0x40376394: spi_flash_op_block_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_utils.c line 134
0x403765cf: ipc_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_ipc.c line 65
0x4037e8be: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c line 139
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

TD-er, 2025-04-25:
There is also a remark about this in the docs for ESP-IDF:
- One Shot mode: 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/adc_oneshot.html#iram-safe
- Continuous mode: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/adc_continuous.html#iram-safe

Parsaabasi, 2026-10-14:
Hello, 

Closing this issue because it refers to an older version. Please update to the latest version and check if the issue persists. If you still need help, feel free to reopen this issue or create a new one. Thanks!

closed 2026-10-14 by Parsaabasi