espressif/arduino-esp32 #11293: Problems with IDE 2.3.6  and ESP32 version  3.2.0

roliwa, 2025-04-24:
### Board

ESP32 Wroom        using  ESP32-WROOM-DA Modul  Board

### Device Description

I use my sketch  with  IDE  2.3.4  and   ESp32  version 3.1.1  without Problems.  I was informed about new  IDE
3.2.0 and installed it  as well as  related Boards ....  
I had some Issues  with USB  flash at the beginning.  Even more issue 👍 

d:\SynologyDrive\Development\Arduino-ESP8622\Sample Sketches\libraries\OneWire\OneWire.cpp:599:22: warning: extra tokens at end of #undef directive
  599 | #  undef noInterrupts() {portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;portENTER_CRITICAL(&mux)
      |                      ^
d:\SynologyDrive\Development\Arduino-ESP8622\Sample Sketches\libraries\OneWire\OneWire.cpp:600:20: warning: extra tokens at end of #undef directive
  600 | #  undef interrupts() portEXIT_CRITICAL(&mux);}

Still Sketch compile continue and was flashed .   But the  WIFI connection did not work in my sketch anymore  

When I flashed it with the prev version of IDE/ESP32   --- everything worked fine ..  what was messed up ?

### Hardware Configuration

Only LCD  

### Version

v3.2.0

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80 Mhz

### PSRAM enabled

yes

### Upload speed

11520

### Description

WIFI networking not working   -- see above

### Sketch

```cpp
#include <WiFi.h>
#include <WebServer.h>

void WiFi_Connect() {

  
  int   count         = 0 ; 

  MySerialprint("WLAN Connecting to: %s \n", ssid);

  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);

  while ((WiFi.status() != WL_CONNECTED) && ( count < 5 )) {
    WiFi.begin(ssid, password);
    delay(500);
    count = count + 1;
    MySerialprint(".");
  }

 if (WiFi.status() == WL_CONNECTED) {

  WiFi_verbunden = true;
  
  // get  signal strength:
  WiFi_dBm  = WiFi_getQuality();

#ifdef OTA

  // Port defaults to 3232
  // Wegen Fehler in Arduino IDE muss ich dies setzten
  ArduinoOTA.setPort(3232);

  // Hostname defaults to esp3232-[MAC]
  ArduinoOTA.setHostname(Hostname);

  // No authentication by default
  // ArduinoOTA.setPassword("admin");

  // Password can be set with it's md5 value as well
  // MD5(admin) = 21232f297a57a5a743894a0e4a801fc3
  // ArduinoOTA.setPasswordHash("21232f297a57a5a743894a0e4a801fc3");

  ArduinoOTA
    .onStart([]() {
      String type;
      if (ArduinoOTA.getCommand() == U_FLASH) {
        type = "sketch";
      } else {  // U_SPIFFS
        type = "filesystem";
      }

      // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
      Serial.println("Start updating " + type);
    })
    .onEnd([]() {
      Serial.println("\nEnd");
    })
    .onProgress([](unsigned int progress, unsigned int total) {
      Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
    })
    .onError([](ota_error_t error) {
      Serial.printf("Error[%u]: ", error);
      if (error == OTA_AUTH_ERROR) {
        Serial.println("Auth Failed");
      } else if (error == OTA_BEGIN_ERROR) {
        Serial.println("Begin Failed");
      } else if (error == OTA_CONNECT_ERROR) {
        Serial.println("Connect Failed");
      } else if (error == OTA_RECEIVE_ERROR) {
        Serial.println("Receive Failed");
      } else if (error == OTA_END_ERROR) {
        Serial.println("End Failed");
      }
    });

  ArduinoOTA.begin();
#endif   //  OTA

  MySerialprint("\nWiFi connected - IP address: %s \n", WiFi.localIP().toString());

  //   Attach Server Handler Functions & Start WebServer
  server.on("/", WiFi_handleGet);  // ip/cmd   - request handler
  server.begin();

  MySerialprint("HTTP Server gestartet \n");
  delay(500);

 }
 else
 {
     WiFi_verbunden = false; 
     MySerialprint("\nWiFi NOT connected , will try later - IP address: %s \n", WiFi.localIP().toString());
 }

}
```

### Debug Message

```plain
not needed I guess
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-04-24:
Move `WiFi.begin(ssid, password);` before the `while` loop. Calling it in a loop is a bad idea

lbernstone, 2025-04-24:
tasmota (@Jason2866) maintains a version of onewire that works with esp32.
https://github.com/arendst/Tasmota/tree/master/lib/lib_basic/TasmotaOneWire-2.3.3

roliwa, 2025-04-25:
Does not Point to error which was introduced. I do not know the root cause, but have found out, that with the new IDE setup the  Sample sketched  like "DS18x20_Temperature do not compile without the error described anymore !! 
I did minimize it to 

 
#include <OneWire.h>

// OneWire DS18S20, DS18B20, DS1822 Temperature Example
//
// http://www.pjrc.com/teensy/td_libs_OneWire.html
//
// The DallasTemperature library can do all this work for you!
// https://github.com/milesburton/Arduino-Temperature-Control-Library
 
void setup(void) {
}
void loop(void) {
 
} 

-->  same error !!   By adding  on the top      #include <Arduino.h>      it will be compiling without error

SO what has now changed   --  Sample sketches wrong     or  IDE   messed something up ?

roliwa, 2025-04-25:
> Move `WiFi.begin(ssid, password);` before the `while` loop. Calling it in a loop is a bad idea

Not sure  if I understand this ?   I have checked as well the   Documentation    like 

// attempt to connect to Wifi network:
  while ( status != WL_CONNECTED) {
    Serial.print("Attempting to connect to WEP network, SSID: ");
    Serial.println(ssid);
    status = WiFi.begin(ssid, keyIndex, key);

    // wait 10 seconds for connection:
    delay(10000);
  }

from :      https://reference.arduino.cc/reference/en/libraries/wifi/wifi.status/

If you considre it as wrong.  How should it look like instead.  I try 5 times to get connection.    And this maybe all 10 minutes  it I do not get a connetion.

me-no-dev, 2025-04-25:
You are waiting only half a second in the example code above. That is rarely enough to connect to the WiFi

closed 2025-10-02 by Jason2866