espressif/arduino-esp32 #11294: WiFiClient.read in 3.2.0 is very slow

kungfupizza, 2025-04-24:
### Board

ESP32S3

### Device Description

ESP32S3 board wifi connected

### Hardware Configuration

Nothing connected.

### Version

v3.2.0

### IDE Name

VSCode

### Operating System

Windows 11

### Flash frequency

80M

### PSRAM enabled

yes

### Upload speed

115200

### Description

When using `WiFiClient*           _client;` to read from a server, `_client->read(InBuff.getWritePtr(), availableBytes);` I see difference in speed in 3.0.7 vs 3.2.0

in 3.0.7(IDF version 5.1.4 I can see 
```
I (403056) content-length: 2335192
I (403264) stream ready, buffer filled in 193 ms
```
in 3.2.0
``` stream ready, buffer filled in 4054 ms``` for same length of content.

### Sketch

```cpp
WiFiClient _client;
void setup()
{
 // setup the client
 // call a GET
 // read from server
 // check time taken to fill the entire buffer
}

void loop()
{
 // do nothing
}
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2025-04-24:
Do you see the same issue with the [example](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClientBasic/WiFiClientBasic.ino)?

kungfupizza, 2025-04-24:
Similar results @lbernstone

TD-er, 2025-04-25:
Can you also try this with a fully wiped ESP?
I did notice the RF-calibration between ESP-IDF 5.3 and ESP-IDF 5.4 is incompatible.

When switching to a build based on IDF5.4 (or 5.5) from a build based on an older IDF I had to wipe the RF calibration to get a stable WiFi connection.
Also the other way around, when moving from a build using IDF 5.4 to one based on older IDF.

N.B. this is not required on all nodes, so I guess it is only affecting those boards which already needed some extra antenna matching options (as in, boards with already poor antenna design)

kungfupizza, 2025-04-26:
`esp_phy_load_cal_and_init` does not return. Raised a [bug](https://github.com/espressif/esp-idf/issues/15862) on ESP-IDF

closed 2025-10-02 by Jason2866