espressif/arduino-esp32 #11301: v3.2.0 Internal error tinyusb

RegularSource, 2025-04-25:
### Board

ESP32S3 develop board

### Device Description

ESP32S3board

### Hardware Configuration

No usb was used Including program and compilation configuration.But this part reports an error

### Version

v3.2.0

### IDE Name

Arduino IDE

### Operating System

windows11

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

No usb was used Including program and compilation configuration.But this part reports an error

### Sketch

```cpp
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
C:\Users\admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\cores\esp32\esp32-hal-tinyusb.c: In function 'tinyusb_driver_install':
C:\Users\admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\cores\esp32\esp32-hal-tinyusb.c:176:3: error: unknown type name 'tusb_rhport_init_t'
  176 |   tusb_rhport_init_t tinit;
      |   ^~~~~~~~~~~~~~~~~~
C:\Users\admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\cores\esp32\esp32-hal-tinyusb.c:177:28: error: 'tusb_rhport_init_t' undeclared (first use in this function)
  177 |   memset(&tinit, 0, sizeof(tusb_rhport_init_t));
      |                            ^~~~~~~~~~~~~~~~~~
C:\Users\admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\cores\esp32\esp32-hal-tinyusb.c:177:28: note: each undeclared identifier is reported only once for each function it appears in
C:\Users\admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\cores\esp32\esp32-hal-tinyusb.c:178:8: error: request for member 'role' in something not a structure or union
  178 |   tinit.role = TUSB_ROLE_DEVICE;
      |        ^
C:\Users\admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\cores\esp32\esp32-hal-tinyusb.c:178:16: error: 'TUSB_ROLE_DEVICE' undeclared (first use in this function); did you mean 'TUSB_DESC_DEVICE'?
  178 |   tinit.role = TUSB_ROLE_DEVICE;
      |                ^~~~~~~~~~~~~~~~
      |                TUSB_DESC_DEVICE
C:\Users\admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\cores\esp32\esp32-hal-tinyusb.c:183:8: error: request for member 'speed' in something not a structure or union
  183 |   tinit.speed = TUSB_SPEED_FULL;
      |        ^
C:\Users\admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\cores\esp32\esp32-hal-tinyusb.c:184:8: error: too many arguments to function 'tusb_init'
  184 |   if (!tusb_init(0, &tinit)) {
      |        ^~~~~~~~~
In file included from C:\Users\admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\cores\esp32\esp32-hal-tinyusb.h:27,
                 from C:\Users\admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\cores\esp32\esp32-hal-tinyusb.c:40:
C:\Users\admin\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-466a392a\esp32s3/include/arduino_tinyusb/tinyusb/src/tusb.h:136:6: note: declared here
  136 | bool tusb_init(void);
      |      ^~~~~~~~~
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2025-04-25:
Looks like your ArduinoIDE setup is broken. Reinstall.

RegularSource, 2025-04-25:
> Looks like your ArduinoIDE setup is broken. Reinstall.

It has been solved. It was caused by the tools not being updated