espressif/arduino-esp32 #11306: mdns_networking: Cannot allocate memory

JetForMe, 2025-04-27:
### Board

UM Feather S3

### Device Description

I'm using a UM Feather S3 on a custom PCB with a few additional components to drive an LED matrix panel.

### Hardware Configuration

I'm using a UM Feather S3 on a custom PCB with a few additional components to drive an LED matrix panel.

### Version

v3.2.0

### IDE Name

Arduino IDE 2

### Operating System

macOS

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921600

### Description

I have been working on this project that I just added mDNS to. I use it to advertise the built-in web server, nothing more. After just a few minutes running, I get a ton of logging like this:

```
E (244434) mdns: Cannot allocate memory (line: 1767, free heap: 8337220 bytes)
E (244435) mdns_networking: Cannot allocate memory (line: 149, free heap: 8332324 bytes)
) mdns_networking: Cannot allocate memory (line: 149, free heap: 8330060 bytes)
mdns_networking: Cannot allocate memory (line: 149, free heap: 8327612 bytes)
1498327612 bytes)
E (244438) mdns_networking: Cannot allocate memory (line: 149, free heap: 8327612 bytes)
E (244438) mdns_networking: Cannot allocate memory (line: 149, free heap: 8330060 bytes)
244439) mdns_networking: Cannot allocate memory (li, free heap: 8327612 bytes)
E (244439) mdns_networking: Cannot allocate memory (line: 149, free heap: 8330060 bytes)
mdns_networking8327612 bytes)
E (244440) mdns_networking: Cannot allo149, free heap: 8330060 bytes)
E (244440orking: Cannot allocate memory (line: 149, free heap: 8327612 bytes)
E (244441) mdns_networking, free heap: 8325164 bytes)
E (244442) mdns_networking: Cannot allocate m, free heap: 8322716 bytes)
) mdns_networking: Cannot allocate memory (line: 149, free heap: 8320268 bytes)
E (244443) mdns_networking: Cannot allocate memory (line: 149, free heap: 8317820 bytes)
mdns_networking: Cannot allocate memory (line: 149, free heap: 8315372 bytes)
ree heap: 8315372 bytes)
mdns_networking: Cannot allocate mem, free heap: 8317820 bytes)
, free heap: 8320268 bytes)
```

Eventually I get a Core 1 panic'ed (corrupted stack) and the app restarts.

### Sketch

```cpp
//	Advertise mDNS…
	
#if true	
	Serial.println("Advertising mDNS…");
	if (!MDNS.begin("clock"))
	{
		Serial.println("mDNS init failed");
	}
	else
	{
		Serial.println("mDNS responder started");
		MDNS.addService("http", "tcp", 80);
	}
#endif
```

### Debug Message

```plain
See above
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2025-04-29:
@JetForMe - 

Memory allocation can happen within many different memory spaces. 
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/mem_alloc.html

When `malloc()` is used within ESP32 Arduino, it can allocate from PSRAM, whenever the block is bigger than 4KB.
`E (244441) mdns_networking, free heap: 8325164 bytes)` indicates the total available memory, including PSRAM, which shall be most of it.

It is very possible that the DRAM available space is short and `mdns` is trying to allocate from it using something like `heap_caps_malloc(MALLOC_CAP_8BIT)`, which will not try to allocate space from PSRAM, only from SRAM.

It is possible to check what is the DRAM remaining space by calling `heap_caps_get_free_size(MALLOC_CAP_8BIT)`
Check how the application is allocating memory in order to analyse what could be changed to make more room in DRAM.

SuGlider, 2025-04-29:
@JetForMe - I have executed the example from
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESPmDNS/examples/mDNS_Web_Server/mDNS_Web_Server.ino
This does the same as in the fragment code that you have posted in this issue.

Using Arduino Core 3.2.0.
I can browse `http://esp32.local` from my computer and I got a plain web page with `Hello from ESP32 at 192.168.15.35`.
It works fine. No error messages.

JetForMe, 2025-04-29:
> I can browse `http://esp32.local` from my computer and I got a plain web page with `Hello from ESP32 at 192.168.15.35`.
> It works fine. No error messages.

Yeah, it works for quite a while, until it doesn't. It first reports the errors, then eventually reboots.

SuGlider, 2025-05-03:
If you are using Windows with a Web Browser to access `clock.local` please note that Windows has many issues with mDNS.

https://learn.microsoft.com/en-us/answers/questions/2195980/windows-10-does-not-receive-mdns-response

SuGlider, 2025-05-03:
I see that you may be using macOS.
It should work fine with macOS, right @me-no-dev?
The current implementation has been certified.

In order to allow us to help you, we need a sketch and detailed issue replication information.
After that we shall be able to consistently replicate the issue and investigated it.
Once the issue is confirmed and investigated, we can issue a PR to fix it.

JetForMe, 2025-05-03:
It will be a challenge to give you the full sketch, but let me see what I can do.

I'm using macOS to access it. But the node doing the lookup should have no bearing on whether or not the Arduino device crashes after advertising for a while.

JetForMe, 2025-10-01:
Oh, perhaps someday, but I've put that project aside for the time being. I moved away from the Arduino build environment, and gave up on mDNS for the time being.

Parsaabasi, 2025-10-01:
Thank you for your response, 
I will close this issue for now then. In case you plan to work on it again, please feel free to reopen or create a new issue. 
Kind regards,