espressif/arduino-esp32 #11312: The data transfer causes the USB to fail to communicate

guess-name, 2025-04-28:
### Board

ESP32S3

### Device Description

ESP32-S3-WROOM-1U-N16R8

Single-module testing

[image]

### Hardware Configuration

Minimal system

GPIO18 and GPIO19 No other peripherals

### Version

v3.1.0

### IDE Name

VSCODE

### Operating System

win11

### Flash frequency

The module comes with it

### PSRAM enabled

yes

### Upload speed

115200

### Description

Sending a string of data every 50ms will cause the USB to fail (unacceptable and unable to send)。
Sending the same data every 100ms will not cause the above problems

Send data：{"cmd":1,"data":100}

### Sketch

```cpp
void setup()
{
    Serial.begin(115200);
    Serial.setRxBufferSize(4096);
    Serial.setTxBufferSize(4096);
    delay(1000); // 等待串口初始化
}

// 线程函数
void PcComTask(void *arg)
{
    while (adcUartObj._initFinish == false)
    {
        delay(100);
    }
    pcComObj.init(&Serial);
    uint8_t lent = 0;
    while (1)
    {
        if (lent == pcComObj._serial->available() && lent > 0)
        {
            pcComObj.handleData();
            lent = 0;
        }
        else
        {
            lent = pcComObj._serial->available();
        }
        delay(5);
    }
}
```

### Debug Message

```plain
No errors
```

### Other Steps to Reproduce

If you send within 100 ms, you will not have this exception, but if the transmission interval is less than 80 ms, an error will occur

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

guess-name, 2025-04-28:
[image]

guess-name, 2025-04-28:
[image]

SuGlider, 2025-04-29:
Which USB CDC mode is used? 
ESP32-S3 has 2 modes: Hardware JTAG/CDCD and OTG USB CDC.

guess-name, 2025-05-06:
I'm not sure which one to use.I'm not sure which one to use.I defined the following properties directly in the cmakelists.txt

[image]

But I disabled the USB-OTA feature in eFuse