espressif/arduino-esp32 #11319: Preferences getString fails even when provided default value on one particular board?!

ddundovic, 2025-04-28:
### Board

GOOUUU-ESP32

### Device Description

It is development board bought from Aliexpress with ESP32 "classic" MCU on it.
Name of the board printed on the PCB is "GOOUUU-ESP32" (38-pin).
Nothing special on the board, no LCD, no card slot, nothing.

### Hardware Configuration

Nothing is attached

### Version

latest stable Release (if not listed below)

### IDE Name

VSCode + PlatformIO

### Operating System

Fedora Linux 41

### Flash frequency

not sure, default for the board

### PSRAM enabled

yes

### Upload speed

921600

### Description

Same code to access values stored in NVS memory works on numerous different development boards and MCUs (ESP8266, ESP32 classic, ESP32C3, etc.) but fails on this one particular development board (also based on ESP32 classic).
Name of the board printed on the PCB is "GOOUUU-ESP32" (38-pin).

At first run (or until some values are created in NVS), they, obviously, don't exist.
So to read the string value from NVS I used preferences.getString function with the default value of String("").
I would then check if string I got back was "" and if it was I would provide some default value or whatever.

On multiple development boards that worked just fine but on this one I get an error:
"[    49][E][Preferences.cpp:483] getString(): nvs_get_str len fail: deviceName NOT_FOUND".
It is true that the key "deviceName" doesn't exist yet, but why function does not simply return the default value of String("")?

### Sketch

```cpp
#include <Arduino.h>
#include <Preferences.h>
#include <WiFi.h>
const char *flashNamespace = "data";         // flash memory namespace
char MAC[20] = {0}; // device's MAC address
void setup()
{
  String s;
  strncpy(MAC, WiFi.macAddress().c_str(), 20);
  preferences.begin(flashNamespace, false);
  // lines below work
  if (!preferences.isKey("deviceLocation")) preferences.putString("deviceLocation", "");  
  s = preferences.getString("deviceLocation");
  if (s == "") s = String(MAC); 
  strncpy(location, s.c_str(), maxDevLocLength-1);

  // lines below don't work  
  s = preferences.getString("deviceName", String(""));
  if (s == "") s = String(MAC);
  strncpy(device, s.c_str(), maxDevLocLength-1);
  preferences.end();

void loop() {}
```

### Debug Message

```plain
[    49][E][Preferences.cpp:483] getString(): nvs_get_str len fail: deviceName NOT_FOUND
```

### Other Steps to Reproduce

Getting string value (or the default one) from NVS that doesn't yet exist by calling Preferences.getString with default value works on these boards, to name the few:

- esp32doit-devkit-v1 (ESP32 classic)
- lolin_c3_mini (ESP32C3)
- d1_mini_lite (ESP8266)
- esp01_1m (ESP8266)  

Everything else, the code, IDE, tool chain is the same.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2025-04-29:
Maybe it can't read the flash. If you set logging to verbose (-DCORE_DEBUG_LEVEL=5) you should get more info.
Please find esptool (.py) in a terminal and run `./esptool.py chip_id`
Post the chip info.  It may only work as DIO if it is ancient and cheap.

ddundovic, 2025-04-29:
It can read flash. For instance I have boot counter that is stored in NVS, when it boot it reads it, adds 1, and stores it back. That works, the number is increasing.

Debug info reads:
`
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[     7][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[    43][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[    52][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
[    63][V][esp32-hal-uart.c:467] uartBegin(): UART0 initialization done.

CPU0 reset reason: (0x0C) Software reset CPU
CPU1 reset reason: (0x0C) Software reset CPU
[    92][E][Preferences.cpp:483] getString(): nvs_get_str len fail: deviceName NOT_FOUND
`

When uploading esptool prints:
`
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-D0WDQ6 (revision v1.0)
Features: WiFi, BT, Dual Core, Coding Scheme None
Crystal is 40MHz
MAC: 30:ae:a4:44:7f:68
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x000e5fff...
`

chip_id prints:
`
esptool.py v4.8.1
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision v1.0)
Features: WiFi, BT, Dual Core, Coding Scheme None
Crystal is 40MHz
MAC: 30:ae:a4:44:7f:68
Uploading stub...
Running stub...
Stub running...
Warning: ESP32 has no Chip ID. Reading MAC instead.
MAC: 30:ae:a4:44:7f:68
Hard resetting via RTS pin...
`

ddundovic, 2025-04-29:
My bad ...

Function works.

I had some other problem with the board and the last message I saw in the terminal was from preferences library, so I assumed it was causing problems.

My apologies.

closed 2025-04-29 by ddundovic