espressif/arduino-esp32 #11325: WiFiClientSecure, Object destroys itself even after the server separates the connection

schreibfaul1, 2025-04-29:
### Board

ESP32 Dev Module

### Device Description

ESP32 and ESP32-S3 boards

### Hardware Configuration

The board has no external circuit

### Version

v3.2.0

### IDE Name

Arduino IDE

### Operating System

Linux Mint

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I create an object `WiFiClientSecure client`, Connect to a server, get the HTTP Response header and the Payoad. As far as everything is expected.
As soon as the server closes the connection, the client no longer exists.

The query `client.connected() `should return false and the object should continue to exist.
but I get:
`fail on 0, errno: 9, "Bad file number"` and "client is null"

I wonder if that should be so

### Sketch

```cpp
#include "Arduino.h"
#include "WiFi.h"
#include "WiFiClientSecure.h"

const char *ssid = "*****";
const char *password = "*******";
const char *server = "arduino.tips";  // Server URL
int av;

WiFiClientSecure client;

void setup() {
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(1000);
  }
  log_i("WiFi connected");
  client.setInsecure();
  bool res = client.connect(server, 443);
  if (res) {
    client.print("GET /asciilogo.txt HTTP/1.1\r\nHost: ");
    client.print(server);
    client.print("\r\nConnection: close\r\n\r\n");
  }
}

void loop() {
  vTaskDelay(1);
  if(!client) log_e("client is null");
  if (client.connected()) {
    av = client.available();
    for (int i = 0; i < av; i++) Serial.write(client.read());
  }
}
```

### Debug Message

```plain
;;;   ;;;;;`  ;;;;:  .;;  ;; ,;;;;;, ;;. `;,  ;;;;   
    ;;;   ;;:;;;  ;;;;;; .;;  ;; ,;;;;;: ;;; `;, ;;;:;;  
   ,;:;   ;;  ;;  ;;  ;; .;;  ;;   ,;,   ;;;,`;, ;;  ;;  
   ;; ;:  ;;  ;;  ;;  ;; .;;  ;;   ,;,   ;;;;`;, ;;  ;;. 
   ;: ;;  ;;;;;:  ;;  ;; .;;  ;;   ,;,   ;;`;;;, ;;  ;;` 
  ,;;;;;  ;;`;;   ;;  ;; .;;  ;;   ,;,   ;; ;;;, ;;  ;;  
  ;;  ,;, ;; .;;  ;;;;;:  ;;;;;: ,;;;;;: ;;  ;;, ;;;;;;  
  ;;   ;; ;;  ;;` ;;;;.   `;;;:  ,;;;;;, ;;  ;;,  ;;;;   
[  3648][E][sketch_apr29a.ino:31] loop(): client is null
[  3654][E][NetworkClient.cpp:327] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[  3664][E][NetworkClient.cpp:327] setSocketOption(): fail on 0, errno: 9, "Bad file number"
```

### Other Steps to Reproduce

The sketch is enough to reproduce

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

schreibfaul1, 2025-05-04:
There are hosts, such as "api.openai.com", which do not deliver a `contentLength` after the speech synthesis "/v1/audio/speech".
The audio data is complete when the server closes the connection.

My idea is to use `client.connected()` to recognise when the payload is complete. But this returns `setSocketOption(): fail on 0, errno: 9, "Bad file number"` in case the TCP connection was closed. This should be easy reproducible with the sketch.

kaloprojects, 2025-05-10:
@me-no-dev or @P-R-O-C-H-Y : any thoughts a/o plans ?
Thx!

me-no-dev, 2025-05-13:
```cpp
if(!client) log_e("client is null");
```

Is the same thing as
```cpp
if(!client.connected()) log_e("client is not connected");
```

Client is disconnected and it's underlying socket is gone. The object still exists

me-no-dev, 2025-05-13:
[This PR](https://github.com/espressif/arduino-esp32/pull/11356) should fix the constant errors. Example above print only `client is null` in a loop. Again, client is not `null`, only it's internal socket object is gone. Client can be re-used