espressif/arduino-esp32 #11329: Last libs with release 3.0.4 - Are there any newer for the more recent releases?

hmbacher, 2025-05-01:
### Board

ESP32S3

### Device Description

Not relevant

### Hardware Configuration

Not relevant

### Version

v3.0.4

### IDE Name

VSCode

### Operating System

Windows 11

### Flash frequency

-

### PSRAM enabled

no

### Upload speed

-

### Description

Hello Folks!
Im wondering why for only for some releases libs are available in the assets. The latest release with libs I found is  3.0.4.

As i would like to use arduino-esp32 with platform.io in VSCode I have to manually specify the arduino-esp32 target version and libs:

```ini
[env]
framework = arduino
platform = espressif32 @ 6.10.0
platform_packages=
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.4
  framework-arduinoespressif32-libs @ https://github.com/espressif/arduino-esp32/releases/download/3.0.4/esp32-arduino-libs-3.0.4.zip
```

I would like to upgrade to 3.3.0, but the assets do not contain the corresponding libs.

Could you please explain me, what to do, to obtain the appropriate libs?

And is my understanding right, that if I would like to change settings using _sdkonfig / menuconfig_, that mean manually building a new version of the libs, before continuing to use arduino-esp32 with platform.io build system! Or is that possible directly from within platform.io build system somehow?

Thanky you very much in advance!

### Sketch

```cpp
-
```

### Debug Message

```plain
-
```

### Other Steps to Reproduce

-

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

hmbacher, 2025-05-02:
Oh nice, I will have a depper look to pioarduino!

Generally. I knew, that arduino-esp32 V3 is curretnly not supported by platform.io, but I got it to work following https://github.com/platformio/platform-espressif32/issues/1225#issuecomment-2155602773

Unfortunately you have to link the libs explicitely, what brings me back to my initial question. How are the libs for a specific arduino-esp32 releases intended to be build, if they are not distributed with the release assets regularely?

lbernstone, 2025-05-02:
The built-in libraries are part of the release assets, along with the core and the pre-compiled sdk, which will all be a matching (versioned) set.  That's why you should not have separate lines for platform_packages.