espressif/arduino-esp32 #11333: 308 Redirect when Making POST Request, Despite Curl Working

sr5434, 2025-05-02:
### Board

ESP32-S v1.1

### Device Description

Plain module on breadboard

### Hardware Configuration

I have connected an LED to pin 12 and a soil moisture sensor to pin 36.

### Version

v3.2.0

### IDE Name

Arduino IDE

### Operating System

macOS 15.4

### Flash frequency

I am unsure

### PSRAM enabled

no

### Upload speed

9600

### Description

I am trying to make an IoT Soil Moisture Monitor. When I make a POST request to my server from curl, the request goes through successfully. However, when I make the request through my ESP32, it keeps redirecting to an empty URL. 

### Sketch

```cpp
#include <WiFi.h>
#include <HTTPClient.h>

#define SENSOR_INPUT 36//15
#define LIGHT_PORT 12

const String ssid = "REDACTED";
const String password = "REDACTED";
const String serverName = "https://soil-moisture-monitor-phi.vercel.app/api/updateMeasurement/";
int prevHumidity = 0;

void setup() {
  Serial.begin(9600);
  WiFi.begin(ssid, password);
  while(WiFi.status() != WL_CONNECTED){
    Serial.print("Connecting...");
    delay(100);
  }
  Serial.println("Connected!");
  // delay(1000);
  pinMode(LIGHT_PORT, OUTPUT);
  if(WiFi.status() == WL_CONNECTED){
    WiFiClient client;
    HTTPClient http;
  
    // Your Domain name with URL path or IP address with path
    http.begin(client, serverName);

    // Data to send with HTTP POST
    String payload = "{\"measurement\": "+String(12345) + "}";
    // Authorization
    // Send HTTP POST request
    http.addHeader("Content-Type", "application/json");
    // http.setFollowRedirects(HTTPC_FORCE_FOLLOW_REDIRECTS);
    int httpResponseCode = http.POST(payload);
    Serial.print("HTTP Response code: ");
    Serial.println(httpResponseCode);
    if (httpResponseCode == 308) {
      String location = http.header("Location");  // Get redirect target
      http.end();

      Serial.println("Redirected to: " + location);

      // Try the new location
      http.begin(client, location);
      http.addHeader("Content-Type", "application/json");
      httpResponseCode = http.POST(payload);
    }
    // Free resources
    http.end();
  }
  else {
    Serial.println("WiFi Disconnected");
  }
}
```

### Debug Message

```plain
Connecting...Connecting...Connecting...Connecting...Connecting...Connecting...Connecting...Connecting...Connecting...Connecting...Connecting...Connecting...Connecting...Connecting...Connecting...Connecting...Connecting...Connecting...Connecting...Connecting...Connecting...Connecting...Connected!
17:39:00.373 -> HTTP Response code: 308
17:39:00.405 -> Redirected to:
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jeroen88, 2025-05-03:
You need to set collectHeaders like explaned [here](https://github.com/esp8266/Arduino/issues/2336#issuecomment-236089322).

sr5434, 2025-05-03:
I just tried that, but unfortunately that does not seem to work either. Here is the code I used:

```cpp
void setup() {
  Serial.begin(9600);
  WiFi.begin(ssid, password);
  while(WiFi.status() != WL_CONNECTED){
    Serial.print("Connecting...");
    delay(100);
  }
  Serial.println("Connected!");
  // delay(1000);
  pinMode(LIGHT_PORT, OUTPUT);
  if(WiFi.status() == WL_CONNECTED){
    WiFiClient client;
    HTTPClient http;
  
    // Your Domain name with URL path or IP address with path
    http.begin(client, serverName);
    const char* headerNames[] = { "Location", "Last-Modified" };
    http.collectHeaders(headerNames, sizeof(headerNames)/sizeof(headerNames[0]));
    // Data to send with HTTP POST
    String payload = "{\"measurement\": "+String(12345) + "}";
    // Authorization
    // Send HTTP POST request
    http.addHeader("Content-Type", "application/json");
    // http.setFollowRedirects(HTTPC_FORCE_FOLLOW_REDIRECTS);
    int httpResponseCode = http.POST(payload);
    Serial.print("HTTP Response code: ");
    Serial.println(httpResponseCode);
    if (httpResponseCode == 308) {
      String location = http.header("Location");  // Get redirect target
      http.end();

      Serial.println("Redirected to: " + location);

      // Try the new location
      http.begin(client, location);
      http.addHeader("Content-Type", "application/json");
      httpResponseCode = http.POST(payload);
    }
    // Free resources
    http.end();
  }
  else {
    Serial.println("WiFi Disconnected");
  }
}
```

lbernstone, 2025-05-03:
Is what you really want here to have 308 (HTTP_CODE_PERMANENT_REDIRECT) behave like the other redirects?  Try changing [sendRequest](https://github.com/espressif/arduino-esp32/blob/master/libraries/HTTPClient/src/HTTPClient.cpp#L629-L631) and see if that gives you the behavior you want.

sr5434, 2025-05-04:
Using ```esp_log.h``` and running ```esp_log_level_set("*", ESP_LOG_VERBOSE);``` in ```setup``` does not print anything. Am I doing something wrong?

lbernstone, 2025-05-04:
Tools menu

[image]

sr5434, 2025-05-05:
I noticed that in the logs on my ESP32 it said that it received an HTTP1.0 response but in CuRL it said it was running HTTP 2.0. Could that be the reason why I am getting redirected?