espressif/arduino-esp32 #11336: Add Battery support Matter

ybarigou, 2025-05-03:
### Related area

Matter

### Hardware specification

ESP32

### Is your feature request related to a problem?

No

### Describe the solution you'd like

This feature request is on adding battery info (capacity, voltage, etc) support in Matter for battery-operated devices.

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2025-05-03:
Implement Server PowerSource Cluster 0x002F with  ATTR 0x000B to 0x000D:
- 0x000B :: BatVoltage (uint32_t)
- 0x000C :: BatPercentRemaining (uint8_t)
- 0x000D :: BatTimeRemaining (uint32_t)

References:
https://github.com/home-assistant/core/issues/116090
https://developer.smartthings.com/docs/edge-device-drivers/matter/generated/clusters/PowerSource.html
Matter Specification documentation at https://csa-iot.org/developer-resource/specifications-download-request/

louisvangeldrop, 2026-09-28:
Any progress on adding MatterPowerSource as endpoint?