espressif/arduino-esp32 #11337: core is too fat

crousekimo, 2025-05-03:
### Related area

core library

### Hardware specification

esp32-s3

### Is your feature request related to a problem?

The esp32 core library is too fat.
It install esp32 esp32-C* esp32-S* and esp32-P* on my computer.
Can I only install the core library for esp32-s3?

### Describe the solution you'd like

The esp32 core falling into esp32 core, esp32-c* core and esp32-s* core.

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2025-05-03:
@me-no-dev PTAL.

lbernstone, 2025-05-03:
The CI is separated enough now that separate (assets and) packages could be put together for each variant.

crousekimo, 2025-05-09:
> The CI is separated enough now that separate (assets and) packages could be put together for each variant.

I know that Tools\esp32-arduino-libs\idf-release_v5.4-2f7dcd86-v1\ has separated enough now.
However, are files in Tools\esp-rv32 and esp-x32 for esp32 or esp32-s3?
Can you separated it?
Tools\esp-rv32 has 2.25G now.
Tools\esp-x32 has 1.2G now.
They are too fat.