espressif/arduino-esp32 #11338: Matter sleep mode temperature sensor example

lboue, 2025-05-04:
### Related area

Matter

### Hardware specification

ESP32-C6

### Is your feature request related to a problem?

Could you add Matter sleep mode temperature sensor example?
The aim is to be able to run this example on battery power.

### Describe the solution you'd like

Base on MatterTemperatureSensor example but with LowPower (sleep mode) to save battery.

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

P-R-O-C-H-Y, 2025-05-05:
@SuGlider Can you take a look please?

SuGlider, 2025-05-05:
> Could you add Matter sleep mode temperature sensor example?
> The aim is to be able to run this example on battery power.

Using Matter over Thread or WiFi?

lboue, 2025-05-05:
By using the wifi connection, as it seems to me that Thread is not supported directly with Arduino.

SuGlider, 2025-05-05:
@lboue - Some analysis about it:

### ESP-Matter Battery Operation:
Matter has an operating mode called ICD (Intermittently Connected Devices) which can be used to extend the battery life of power-limited devices. The Matter specification says that it could be used with Wi-Fi / Thread.

Currently, ESP-Matter only supports Thread based ICD. No Wi-Fi ICD Support at this time.

In order to achieve something similar with WiFi, I think that it could be possible to use Wi-Fi 6 TWT (IEEE802.11ax) capability.
Another way is by using Delivery Traffic Information Map (DTIM) to enter into sleep mode and let the WiFi AP know that it has to cache any messages to the Matter WiFi end device.

Both, Wi-Fi 6 TWT and Wi-Fi DTIM could work within Matter ICD, **BUT** it would require manual CHIP library setup, given that it is not natively supported by ESP-Matter library.

TWT is better than DTIM, once it can be negotiated by the Station (Matter Device). DTIM interval is imposed by the Wi-Fi AP and can't be changed. Modem/Light Sleep modes use DTIM information to wake up on the Wi-Fi Beacon right time and keep the Wi-Fi Connection to the AP.

### ESP32 Arduino support:
Currently, ESP32 Arduino doesn't support NimBLE based Arduino BLE Library, which is necessary in order to use NimBLE and support Matter over Thread. This is a current on-going work.

ESP32 Arduino doesn't support FreeRTOS Tickeless idle mode, neither ESP-IDF Power Management mode (based on DFS - Dynamic Frequency Scaling). Those could work with Wi-Fi 6 TWT in order to reduce the power consumption using an ESP32-C6 with Wi-Fi 6 Network. Anyway, it would require a Wi-Fi 6/6E network infrastructure, including a proper Wi-Fi 6 AP Router.

**In other words, ESP32 Arduino is not in a position to support necessary IDF features that could allow a sufficient power consumption reduction and make it work correctly with Matter.**

It may be possible to be achieved using Arduino as IDF Component, **BUT** this would require specific MENUCONFIG settings.

### Some IDF code examples: 
ESP-Matter ICD using Thread (ESP32-C6 and ESP32-H2):
https://github.com/espressif/esp-matter/tree/main/examples/icd_app/
IDF WiFi 6 with ITWT (ESP32-C6):
https://github.com/espressif/esp-idf/blob/master/examples/wifi/itwt/
IDF  WiFi Power Saving using Light Sleep
https://github.com/espressif/esp-idf/blob/master/examples/wifi/power_save/

A video about ESP32-C6 battery operated devices:
https://www.youtube.com/watch?v=FpTwQlGtV0k

lboue, 2025-12-15:
Does ESP32 Arduino support NimBLE now?