espressif/arduino-esp32 #11340: Wifi connection on 3.2.0 often failed to get IP from DHCP

dzungpv, 2025-05-07:
### Board

ESP32 Dev Module

### Device Description

ESP32 Dev Module

### Hardware Configuration

UART on GPIO 19 and 21

### Version

v3.2.0

### IDE Name

VS

### Operating System

MacOS 15.4.1

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Sometime it failed to get IP and disconnect from Wi-Fi router. It not occurs with previous version.
This is some logs:
```
I (15947) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated
I (15960) Modbus: Start modbus test... TXD: 15, RXD: 18
E (15976) udp_handler: Error occurred during sending: errno 12
I (16241) my_app: Connected to WIFI_1
I (16242) my_app: Ready, IP address: 
I (16243) my_app: 169.254.109.119
I (16243) my_app: Starting network connection
E (16244) my_app: Wifi IP address: 169.254.109.119
I (16500) esp_netif_handlers: sta ip: 192.168.1.124, mask: 255.255.255.0, gw: 192.168.1.1
I (16502) my_app: WiFi connected, IP address: 192.168.1.124
I (16940) wifi:<ba-add>idx:1 (ifx:0, 74:da:88:9a:c5:d6), tid:0, ssn:0, winSize:64
```
And

```
I (675) cpu_start: Multicore app
I (683) cpu_start: Pro cpu start user code
I (683) cpu_start: cpu freq: 160000000 Hz
I (683) app_init: Application information:
I (686) app_init: Project name:     myproject
I (692) app_init: App version:      1.0
I (698) app_init: Compile time:     May  7 2025 07:39:19
I (704) app_init: ELF file SHA256:  6f88bd787...
I (709) app_init: ESP-IDF:          v5.4.1
I (714) efuse_init: Min chip rev:     v3.0
I (719) efuse_init: Max chip rev:     v3.99 
I (724) efuse_init: Chip rev:         v3.1
I (729) heap_init: Initializing. RAM available for dynamic allocation:
I (736) heap_init: At 3FFAE6E0 len 0000F480 (61 KiB): DRAM
I (742) heap_init: At 3FFCC1D8 len 00013E28 (79 KiB): DRAM
I (748) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (754) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (761) heap_init: At 4009C1C4 len 00003E3C (15 KiB): IRAM
I (769) spi_flash: detected chip: generic
I (772) spi_flash: flash io: dio
I (786) coexist: coex firmware version: e727207
I (790) main_task: Started on CPU0
I (794) main_task: Calling app_main()
I (837) main_task: Returned from app_main()
I (851) Button: Install IRS service success
I (852) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (864) gpio: GPIO[18]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (866) gpio: GPIO[23]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (889) my_app: Opening Non-Volatile Storage (NVS) ... 
E (895) my_app: Error (ESP_ERR_NVS_NOT_FOUND) opening NVS!

E (6956) esp-tls: [sock=48] connect() error: Host is unreachable
I (6958) my_app: Opening Non-Volatile Storage (NVS) ... 
E (6959) transport_base: Failed to open a new connection: 32772
E (6966) mqtt_client: Error transport connect
E (6981) my_app: Error (ESP_ERR_NVS_NOT_FOUND) opening NVS!
I (6987) my_app: Opening Non-Volatile Storage (NVS) ... 
E (6993) my_app: Error (ESP_ERR_NVS_NOT_FOUND) opening NVS!
I (6999) my_app: Opening Non-Volatile Storage (NVS) ... 
E (7005) my_app: Error (ESP_ERR_NVS_NOT_FOUND) opening NVS!
I (7015) wifi:wifi driver task: 3ffcf96c, prio:23, stack:6144, core=0
I (7045) wifi:wifi firmware version: 79fa3f41ba
I (7045) wifi:wifi certification version: v7.0
I (7045) wifi:config NVS flash: enabled
I (7046) wifi:config nano formatting: enabled
I (7050) wifi:Init data frame dynamic rx buffer num: 32
I (7055) wifi:Init static rx mgmt buffer num: 5
I (7059) wifi:Init management short buffer num: 32
I (7064) wifi:Init dynamic tx buffer num: 32
I (7068) wifi:Init static rx buffer size: 1600
I (7072) wifi:Init static rx buffer num: 4
I (7076) wifi:Init dynamic rx buffer num: 32
I (7081) wifi_init: rx ba win: 6
I (7084) wifi_init: accept mbox: 6
I (7088) wifi_init: tcpip mbox: 32
I (7092) wifi_init: udp mbox: 10
I (7096) wifi_init: tcp mbox: 6
I (7100) wifi_init: tcp tx win: 5744
I (7104) wifi_init: tcp rx win: 5744
I (7108) wifi_init: tcp mss: 1440
I (7112) wifi_init: WiFi IRAM OP enabled
I (7117) wifi_init: WiFi RX IRAM OP enabled
I (7122) wifi_init: LWIP IRAM OP enabled
I (7131) phy_init: phy_version 4860,6b7a6e5,Feb  6 2025,14:47:07
I (7204) wifi:mode : sta (a0:b7:65:58:77:6c)
I (7205) wifi:enable tsf
I (7209) wifi:Set ps type: 1, coexist: 0

E (7313) my_app: Connected to SSID: WIFI_1
I (7313) my_app: .
I (7325) wifi:new:<2,1>, old:<1,0>, ap:<255,255>, sta:<2,1>, prof:1, snd_ch_cfg:0x0
I (7326) wifi:state: init -> auth (0xb0)
I (7351) wifi:state: auth -> assoc (0x0)
I (7358) wifi:state: assoc -> run (0x10)
I (7397) wifi:connected with WIFI_1, aid = 3, channel 2, 40U, bssid = 74:da:88:9a:c5:d6
I (7397) wifi:security: WPA2-PSK, phy: bgn, rssi: -59
I (7402) wifi:pm start, type: 1

I (7403) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (7422) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (7813) my_app: .
I (7922) wifi:<ba-add>idx:0 (ifx:0, 74:da:88:9a:c5:d6), tid:0, ssn:0, winSize:64
I (8314) my_app: .
I (8815) my_app: .
I (8922) wifi:<ba-add>idx:1 (ifx:0, 74:da:88:9a:c5:d6), tid:6, ssn:2, winSize:64
I (9315) my_app: .
I (9816) my_app: .
I (10316) my_app: .
I (10816) my_app: .
I (11316) my_app: .
I (11816) my_app: .
I (12316) my_app: .
I (12816) my_app: .
I (13316) my_app: .
I (13816) my_app: .
I (14316) my_app: .
I (14816) my_app: .
I (15316) my_app: .
I (15345) esp_netif_handlers: sta ip: 192.168.1.124, mask: 255.255.255.0, gw: 192.168.1.1
I (15347) my_app: WiFi connected, IP address: 192.168.1.124
I (15350) my_app: Start main application.
I (15377) mdns_mem: mDNS task will be created from internal RAM
I (15375) gpio: GPIO[21]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (15390) uart: queue free spaces: 20
I (15391) gpio: GPIO[19]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (15403) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated
I (15745) esp_netif_handlers: sta ip: 169.254.109.119, mask: 255.255.0.0, gw: 0.0.0.0
I (15748) my_app: WiFi connected, IP address: 169.254.109.119
I (15816) my_app: Connected to WIFI_1
I (15817) my_app: Ready, IP address: 
I (15817) my_app: 169.254.109.119
I (15818) my_app: Starting network connection

``` 

It connected and get IP first but later get IP: 169.254.109.119. This is a complex project, but it working fine on 3.1.0. I don't know if it is relate to ESP IDF 5.4.1 or Arduino ESP32 itself. There are some issue relate like https://github.com/espressif/arduino-esp32/issues/11294 

### Sketch

```cpp
WiFi.begin(ssid.c_str(), pwd.c_str());
  ESP_LOGE(TAG, "Connected to SSID: %s", ssid.c_str());

  wifi_timeout = millis() + 30000;
  while (WiFi.status() != WL_CONNECTED && millis() < wifi_timeout)
  {
    ESP_LOGI(TAG, ".");
    //flashing the blue LED to indicate WiFi connecting...
    ledIndicatorBlink();
  }
  if (WiFi.status() != WL_CONNECTED)
  {
    ESP_LOGI(TAG, "Failed to connect to wifi");
    return false;
  }
  ESP_LOGI(TAG, "Connected to %s", ap_ssid.c_str());
  ESP_LOGI(TAG, "Ready, IP address: ");
  unsigned long dhcpStartTime = millis();
  while ((WiFi.localIP().toString() == "0.0.0.0" || WiFi.localIP().toString() == "") && millis() - dhcpStartTime < 5000)
  {
    ESP_LOGI(TAG, ".");
    delay(500);
  }
  if (WiFi.localIP().toString() == "0.0.0.0" || WiFi.localIP().toString() == "")
  {
    ESP_LOGI(TAG, "Failed to get IP address");
    return false;
  }
  ESP_LOGI(TAG, "%s", WiFi.localIP().toString().c_str());
  ledIndicatorOff();
  // Auto reconnected
  WiFi.setAutoReconnect(true);
  WiFi.persistent(true);
```

### Debug Message

```plain
Log above, I use ESP IDF and Arduino as component
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-05-07:
Seems there is a missing code. Please provide a minimal compilable example sketch to reproduce

dzungpv, 2025-05-07:
> Seems there is a missing code. Please provide a minimal compilable example sketch to reproduce

It is complex project, I will try to get minimal code.

TD-er, 2025-07-01:
@dzungpv I think me-no-dev means the posted sketch cannot be just copy/paste into a new test build to verify your problem.
It is missing a few other things like what is done during `setup()` and I do see at least 1 variable which isn't declared in this piece of code you posted.

Also have you tried turing WiFi off and on when a reconnect is needed?
Or at least make sure to call for a disconnect to make sure you're in a known state.

Another good practice I use myself is to always add some `delay(..)` calls after a `WiFi.xxx` call which does change something, to make sure it actually is applied before going further.

For example add a `delay(100);` after turning WiFi and after `WiFi.begin`.

dzungpv, 2025-07-01:
> [@dzungpv](https://github.com/dzungpv) I think me-no-dev means the posted sketch cannot be just copy/paste into a new test build to verify your problem. It is missing a few other things like what is done during `setup()` and I do see at least 1 variable which isn't declared in this piece of code you posted.
> 
> Also have you tried turing WiFi off and on when a reconnect is needed? Or at least make sure to call for a disconnect to make sure you're in a known state.
> 
> Another good practice I use myself is to always add some `delay(..)` calls after a `WiFi.xxx` call which does change something, to make sure it actually is applied before going further.
> 
> For example add a `delay(100);` after turning WiFi and after `WiFi.begin`.
> 
> N.B. Some access points, like some FritzBox (not limited to these) may need upto 1000 msec delay after `WiFi.begin`. Not 100% sure why, but it seems the ESP may need its full attention to quickly respond or else it might miss some packets from the AP.

It is the real world application. I just revert back old version (using git) and it work fine, It is very complex project (~20k line of code, include IP V4, IP V6, MQTT, Http server, ESP matter stack) that why create a minimal project from it to reproduce the issue take so much time, so I just post code start Wi-Fi connection. 

What I see in the log seem that first the router provide the working IP for the ESP then after some second later it change to the 169.x.x.x like in the PC when you disable DHCP, it may relate to lwip. When in that state the device complete offline and stuck it forever. I have the callback to listen network event but not help. So from the time I stay with stable version 3.1.x. 
For the delay I using:
```
  while ((WiFi.localIP().toString() == "0.0.0.0" || WiFi.localIP().toString() == "") && millis() - dhcpStartTime < 5000)
  {
    ESP_LOGI(TAG, ".");
    delay(500);
  }
```
To waiting for it provide a valid IP, so it is much more time than your suggest delay(100)

TD-er, 2025-07-01:
> For the delay I using:
> 
> ```
>   while ((WiFi.localIP().toString() == "0.0.0.0" || WiFi.localIP().toString() == "") && millis() - dhcpStartTime < 5000)
>   {
>     ESP_LOGI(TAG, ".");
>     delay(500);
>   }
> ```
> 
> To waiting for it provide a valid IP, so it is much more time than your suggest delay(100)

No I meant right after the `WiFi.begin`, so before you even try to evaluate the IP.

```c++
WiFi.begin(....);
delay(100); // Or even 1000 for some access points
while ((WiFi.localIP().toString() == "0.0.0.0" || WiFi.localIP().toString() == "") && millis() - dhcpStartTime < 5000)
{ ... }
```

If you didn't specifically try to enable the STA mode, then I think it is done in the `WiFi.begin` call.
And like I said, maybe even do a `WiFi.disconnect()` (or whatever the function is called) before calling `WiFi.begin`. And of course also add a `delay(100);` between the `disconnect` and `begin`, just to make sure you're in a known state as much as possible.

Jason2866, 2025-10-02:
please try latest Arduino core 3.3.1

Parsaabasi, 2025-10-29:
Hi @dzungpv 

According to the comments above and using an older Arduino version, I close this issue for now. 

Please try the latest Arduino version. In case the issue persisted, please feel free to reopen it or create a new one. 

Kind regards.

closed 2025-10-29 by Parsaabasi