espressif/arduino-esp32 #11343: PHY and GPIO error after 3.1

ricardourio, 2025-05-08:
### Board

ESP32-WROOM-32E

### Device Description

DS3231 and BME280

### Hardware Configuration

GPIO21 & 22 to I2C devices

### Version

v3.2.0

### IDE Name

Arduino 2.3.5

### Operating System

Windows 11

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

921600

### Description

WiFi.h seems not 100% compatible with versions 3.1 and above, and return error on boot, while it didn´t happen with versions 3.0.7 or 2.0.17 . The error may have some consequences with Bluetooth communication as well when using 3.2.0. For now the sketch bellow has the following behavior on boot :
3.0.7 No error
3.1.3 E (509) wifi_init_default: netstack cb reg failed with 12308
3.2.0 E (89) phy_comm: gpio[0] number: 2 is reserved

### Sketch

```cpp
#include <WiFi.h>

void setup() {
  Serial.begin(115200);
  Serial.println("Starting...");

  WiFi.mode(WIFI_STA);     
  WiFi.disconnect(true);  
  delay(100);
}

void loop() {
  delay(1);
}
```

### Debug Message

```plain
3.1.3 :
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4916
load:0x40078000,len:16492
load:0x40080400,len:4
load:0x40080404,len:3524
entry 0x400805b8
Starting...
E (374) wifi_init_default: netstack cb reg failed with 12308

3.2.0 :
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4888
load:0x40078000,len:16516
load:0x40080400,len:4
load:0x40080404,len:3476
entry 0x400805b4
Starting...
E (90) phy_comm: gpio[0] number: 2 is reserved
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2025-05-12:
I can't reproduce such issue.
Runing the sketch from this issue using Arduino 3.2.0, I see no error message.
May it be some Arduino Core installation problem?

### This is the VEBOSE log output:

```
07:03:31.428 -> ets Jul 29 2019 12:21:46
07:03:31.428 -> 
07:03:31.428 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
07:03:31.476 -> configsip: 0, SPIWP:0xee
07:03:31.476 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
07:03:31.476 -> mode:DIO, clock div:1
07:03:31.476 -> load:0x3fff0030,len:4888
07:03:31.476 -> load:0x40078000,len:16516
07:03:31.476 -> load:0x40080400,len:4
07:03:31.476 -> load:0x40080404,len:3476
07:03:31.476 -> entry 0x400805b4
07:03:31.730 -> [     1][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400d955c
07:03:31.778 -> [    12][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400d9540
07:03:31.778 -> [    26][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400d9524
07:03:31.778 -> [    39][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400d9508
07:03:31.778 -> [    53][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400d955c
07:03:31.825 -> [    66][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400d9540
07:03:31.825 -> [    80][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400d9524
07:03:31.825 -> [    93][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400d9508
07:03:31.874 -> [   107][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400d955c
07:03:31.874 -> [   120][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400d9540
07:03:31.874 -> [   134][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400d9524
07:03:31.874 -> [   147][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400d9508
07:03:31.921 -> [   177][D][esp32-hal-cpu.c:263] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
07:03:31.921 -> [   194][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 3 successfully set to type UART_RX (2) with bus 0x3ffbdb6c
07:03:31.969 -> [   205][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 1 successfully set to type UART_TX (3) with bus 0x3ffbdb6c
07:03:31.969 -> =========== Before Setup Start ===========
07:03:31.969 -> Chip Info:
07:03:31.969 -> ------------------------------------------
07:03:31.969 ->   Model             : ESP32
07:03:31.969 ->   Package           : D0WD-Q5
07:03:32.017 ->   Revision          : 3.00
07:03:32.017 ->   Cores             : 2
07:03:32.017 ->   CPU Frequency     : 240 MHz
07:03:32.017 ->   XTAL Frequency    : 40 MHz
07:03:32.017 ->   Features Bitfield : 0x00000032
07:03:32.017 ->   Embedded Flash    : No
07:03:32.017 ->   Embedded PSRAM    : No
07:03:32.017 ->   2.4GHz WiFi       : Yes
07:03:32.017 ->   Classic BT        : Yes
07:03:32.017 ->   BT Low Energy     : Yes
07:03:32.064 ->   IEEE 802.15.4     : No
07:03:32.064 -> ------------------------------------------
07:03:32.064 -> INTERNAL Memory Info:
07:03:32.064 -> ------------------------------------------
07:03:32.064 ->   Total Size        :   336232 B ( 328.4 KB)
07:03:32.064 ->   Free Bytes        :   294800 B ( 287.9 KB)
07:03:32.064 ->   Allocated Bytes   :    34200 B (  33.4 KB)
07:03:32.064 ->   Minimum Free Bytes:   289228 B ( 282.4 KB)
07:03:32.112 ->   Largest Free Block:   110580 B ( 108.0 KB)
07:03:32.112 -> ------------------------------------------
07:03:32.112 -> Flash Info:
07:03:32.112 -> ------------------------------------------
07:03:32.112 ->   Chip Size         :  8388608 B (8 MB)
07:03:32.112 ->   Block Size        :    65536 B (  64.0 KB)
07:03:32.112 ->   Sector Size       :     4096 B (   4.0 KB)
07:03:32.112 ->   Page Size         :      256 B (   0.2 KB)
07:03:32.112 ->   Bus Speed         : 80 MHz
07:03:32.160 ->   Bus Mode          : QIO
07:03:32.160 -> ------------------------------------------
07:03:32.160 -> Partitions Info:
07:03:32.160 -> ------------------------------------------
07:03:32.160 ->                 nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
[... 29 lines cut ...]
07:03:32.614 -> [   872][V][NetworkEvents.cpp:117] _checkForEvent(): Network Event: 110 - STA_START
07:03:32.614 -> [   879][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 110 - STA_START
07:03:32.614 -> [   888][V][STA.cpp:190] _onStaEvent(): STA Stopped
07:03:32.614 -> [   893][V][NetworkEvents.cpp:117] _checkForEvent(): Network Event: 111 - STA_STOP
07:03:32.647 -> [   900][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 111 - STA_STOP
07:03:32.647 -> [   911][V][NetworkEvents.cpp:117] _checkForEvent(): Network Event: 100 - WIFI_OFF
07:03:32.647 -> [   919][V][NetworkInterface.cpp:262] destroyNetif(): Unregistered event handler
07:03:32.777 -> =========== After Setup Start ============
07:03:32.777 -> INTERNAL Memory Info:
[... cut ...]
07:03:32.777 ->   Minimum Free Bytes:   242080 B ( 236.4 KB)
07:03:32.777 ->   Largest Free Block:   110580 B ( 108.0 KB)
07:03:32.777 -> ------------------------------------------
07:03:32.825 -> GPIO Info:
07:03:32.825 -> ------------------------------------------
07:03:32.825 ->   GPIO : BUS_TYPE[bus/unit][chan]
07:03:32.825 ->   --------------------------------------  
07:03:32.825 ->      1 : UART_TX[0]
07:03:32.825 ->      3 : UART_RX[0]
07:03:32.825 -> ============ After Setup End =============

```

ichbaum, 2025-05-12:
Same issue.
Arduino IDE 2.3.6
ESP32-WROOM-32D

esp32 3.2.0.
E (116) phy_comm: gpio[0] number: 2 is reserved

esp32 3.1.3.
WiFi connected

SuGlider, 2025-05-12:
@ricardourio @ichbaum - Please execute the sketch with Debug Level set to VERBOSE and report here the log in order to allow us to check exactly what ESP32 revision version it is. Please sue Arduino Core 3.2.0.

SuGlider, 2025-05-12:
Good point @TD-er! Erasing the flash before moving from a core to another (with respective IDF Versions) may solve this issue.
IDF WiFi driver saves many information into NVS (flash partition) which changes from one IDF version to another.

ricardourio, 2025-05-12:
I tried the following :
- Arduino 2.3.5 + 3.2.0 Flashing with Arduino
- Arduino 2.3.5 + 3.2.0 Flashing Espressif Flash Download Tool (Erasing before flash) 
- Arduino 2.3.6 + 3.2.0 Flashing with Arduino
- Arduino 2.3.6 + 3.2.0 Flashing Espressif Flash Download Tool (Erasing before flash) 
- Arduino 2.3.6 + 3.2.0 Flashing with Arduino - Partition scheme Minimal with SPDIFF
- Arduino 2.3.6 + 3.2.0 Flashing Espressif Flash Download Tool (Erasing before flash) - Partition scheme Default 4 MB

All gave me same error. Here is the verbose for Arduino 2.3.6 + 3.2.0 - Partition scheme Default 4 MB :

`ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4888
load:0x40078000,len:16516
load:0x40080400,len:4
load:0x40080404,len:3476
entry 0x400805b4
[     1][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400d9410
[    12][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400d93f4
[    26][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400d93d8
[    39][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400d93bc
[    53][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400d9410
[    66][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400d93f4
[    80][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400d93d8
[    93][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400d93bc
[   107][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400d9410
[   120][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400d93f4
[   134][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400d93d8
[   147][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400d93bc
[   163][D][esp32-hal-cpu.c:263] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   177][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 3 successfully set to type UART_RX (2) with bus 0x3ffbdb6c
[   187][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 1 successfully set to type UART_TX (3) with bus 0x3ffbdb6c
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32
  Package           : D0WD-Q5
  Revision          : 3.00
  Cores             : 2
  CPU Frequency     : 240 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000032
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : Yes
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   336196 B ( 328.3 KB)
  Free Bytes        :   295532 B ( 288.6 KB)
  Allocated Bytes   :    33560 B (  32.8 KB)
  Minimum Free Bytes:   289748 B ( 283.0 KB)
  Largest Free Block:   110580 B ( 108.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         :  4194304 B (4 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : QIO
------------------------------------------
[... 36 lines cut ...]
[   864][E][WiFiGeneric.cpp:633] mode(): Dual Antenna Config failed!
[   885][W][STA.cpp:543] disconnect(): STA already disconnected.
[   892][V][STA.cpp:190] _onStaEvent(): STA Stopped
[   897][V][NetworkEvents.cpp:117] _checkForEvent(): Network Event: 111 - STA_STOP
[   904][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 111 - STA_STOP
[   915][V][NetworkEvents.cpp:117] _checkForEvent(): Network Event: 100 - WIFI_OFF
[   923][V][NetworkInterface.cpp:262] destroyNetif(): Unregistered event handler
=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   336196 B ( 328.3 KB)
  Free Bytes        :   274164 B ( 267.7 KB)
  Allocated Bytes   :    53912 B (  52.6 KB)
  Minimum Free Bytes:   242344 B ( 236.7 KB)
  Largest Free Block:   110580 B ( 108.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
     1 : UART_TX[0]
     3 : UART_RX[0]
============ After Setup End =============`

Ricardo Urio

SuGlider, 2025-05-13:
IDF thinks that the SoC has dual antenna.
That seems to be the origin of the issue.

What is the ESP32 that is being used?
Could you send us a clear picture of it?

me-no-dev, 2025-05-13:
At least the Dual Antenna board has been selected in the menu. If you are not using that module, please select the default ESP32 Dev Module

ricardourio, 2025-05-13:
I'm using this one. Not original Espressif module probably, but it's an ESP32-WROOM-32E and I'm selecting ESP32-WROOM-DA, and now I could see what's mean DA.

[image]

ricardourio, 2025-05-13:
Now selecting ESP32 Dev Module as @me-no-dev suggested, using Arduino 2.3.5 + 3.2.0 Flashing with Arduino, no more errors

`-> -> ets Jul 29 2019 12:21:46
-> 
-> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
-> configsip: 0, SPIWP:0xee
-> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
-> mode:DIO, clock div:1
-> load:0x3fff0030,len:4888
-> load:0x40078000,len:16516
-> load:0x40080400,len:4
-> load:0x40080404,len:3476
-> entry 0x400805b4
-> [     1][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400d9160
-> [    12][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400d9144
-> [    26][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400d9128
-> [    39][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400d910c
-> [    53][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400d9160
-> [    66][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400d9144
-> [    80][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400d9128
-> [    93][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400d910c
-> [   107][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400d9160
-> [   120][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400d9144
-> [   134][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400d9128
-> [   147][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400d910c
-> [   163][D][esp32-hal-cpu.c:263] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
-> [   179][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 3 successfully set to type UART_RX (2) with bus 0x3ffbdb6c
-> [   190][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 1 successfully set to type UART_TX (3) with bus 0x3ffbdb6c
-> =========== Before Setup Start ===========
-> Chip Info:
-> ------------------------------------------
->   Model             : ESP32
->   Package           : D0WD-Q5
->   Revision          : 3.00
->   Cores             : 2
->   CPU Frequency     : 240 MHz
->   XTAL Frequency    : 40 MHz
->   Features Bitfield : 0x00000032
->   Embedded Flash    : No
->   Embedded PSRAM    : No
->   2.4GHz WiFi       : Yes
->   Classic BT        : Yes
->   BT Low Energy     : Yes
->   IEEE 802.15.4     : No
-> ------------------------------------------
->   Classic BT        : Yes
->   BT Low Energy     : Yes
->   IEEE 802.15.4     : No
-> ------------------------------------------
-> INTERNAL Memory Info:
-> ------------------------------------------
->   Total Size        :   336232 B ( 328.4 KB)
->   Free Bytes        :   294756 B ( 287.8 KB)
->   Allocated Bytes   :    34228 B (  33.4 KB)
->   Minimum Free Bytes:   289192 B ( 282.4 KB)
->   Largest Free Block:   110580 B ( 108.0 KB)
-> ------------------------------------------
-> Flash Info:
-> ------------------------------------------
->   Chip Size         :  4194304 B (4 MB)
->   Block Size        :    65536 B (  64.0 KB)
->   Sector Size       :     4096 B (   4.0 KB)
->   Page Size         :      256 B (   0.2 KB)
->   Bus Speed         : 80 MHz
->   Bus Mode          : QIO
-> ------------------------------------------
-> Partitions Info:
-> ------------------------------------------
->                 nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
->             otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
[... 26 lines cut ...]
-> [   851][W][STA.cpp:543] disconnect(): STA already disconnected.
-> [   852][V][STA.cpp:186] _onStaEvent(): STA Started
-> [   861][V][NetworkEvents.cpp:117] _checkForEvent(): Network Event: 110 - STA_START
-> [   869][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 110 - STA_START
-> [   878][V][STA.cpp:190] _onStaEvent(): STA Stopped
-> [   882][V][NetworkEvents.cpp:117] _checkForEvent(): Network Event: 111 - STA_STOP
-> [   890][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 111 - STA_STOP
-> [   900][V][NetworkEvents.cpp:117] _checkForEvent(): Network Event: 100 - WIFI_OFF
-> [   908][V][NetworkInterface.cpp:262] destroyNetif(): Unregistered event handler
-> =========== After Setup Start ============
-> INTERNAL Memory Info:
-> ------------------------------------------
->   Total Size        :   336232 B ( 328.4 KB)
->   Free Bytes        :   273784 B ( 267.4 KB)
->   Allocated Bytes   :    54264 B (  53.0 KB)
->   Minimum Free Bytes:   242084 B ( 236.4 KB)
->   Largest Free Block:   110580 B ( 108.0 KB)
-> ------------------------------------------
-> GPIO Info:
-> ------------------------------------------
->   GPIO : BUS_TYPE[bus/unit][chan]
->   --------------------------------------  
->      1 : UART_TX[0]
->      3 : UART_RX[0]
-> ============ After Setup End =============`

lbernstone, 2025-05-13:
This is not the first issue we've seen like this. Perhaps DA should be un-abbreviated in the list?

SuGlider, 2025-05-15:
ESP32-WROOM-E is single antenna. 
`ESP32 Dev board` is the right board to select.

Closing this issue as completed.
Feel free to open it again, if necessary.

closed 2025-05-15 by SuGlider