espressif/arduino-esp32 #11351: ESP32 and BMI160 / ICM-20948 connection I2C issues with version 3.2.0

CancianMatteo, 2025-05-11:
### Board

ESP32C6, ESP32C3

### Device Description

XIAO ESP32C6, ESP32C3 Super Mini

### Hardware Configuration

I2C not working when IMU connected to SDA and SCL pins (22,23 for XIAO and 8,9 for Super Mini)

### Version

v3.2.0

### IDE Name

Arduino IDE

### Operating System

MacOS 13.3.1, Windows 11

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

With version 3.2.0, I2C stopped working on my ESP32C6 and ESP32C3

See full explanation here: https://www.reddit.com/r/esp32/comments/1kj5shf/esp32_and_bmi160_icm20948_connection_i2c_issues/

E (1351) i2c.master: I2C transaction unexpected nack detected 
E (1351) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed 
E (1352) i2c.master: i2c_master_transmit_receive(1220): I2C transaction failed

Reverting to v3.1.3 solved the issue.

### Sketch

```cpp
#include <Wire.h>
 
// I2C Configuration for ESP32
#define BMI160_I2C_ADDRESS 0x68  // I2C address for BMI160 (with SAO pin → GND), connect to 3V3 for 0x69
#define BMI160_SDA_PIN 22     // I2C D4(GPIO22) → SDA Pin for XIAO ESP32C6
#define BMI160_SCL_PIN 23     // I2C D5(GPIO23) → SCL Pin for XIAO ESP32C6
#define ACCEL_SENSITIVITY 16384.0 // Sensitivity for ±2g in LSB/g (adjust based on your configuration)
 
void setup() {
  Serial.begin(115200); // Initialize Serial communication
  Wire.begin(BMI160_SDA_PIN, BMI160_SCL_PIN);         // Initialize I2C communication
 
  // Initialize BMI160 accelerometer
  Wire.beginTransmission(BMI160_I2C_ADDRESS);
  Wire.write(0x7E); // Command register
  Wire.write(0x11); // Set accelerometer to normal mode
  Wire.endTransmission();
  delay(100);
 
  // Perform accelerometer auto-calibration
  autoCalibrateAccelerometer();
 
  Serial.println("BMI160 Initialized and Calibrated");
}
 
void loop() {
  int16_t ax, ay, az;
 
  // Read accelerometer data
  Wire.beginTransmission(BMI160_I2C_ADDRESS);
  Wire.write(0x12); // Start register for accelerometer data
  Wire.endTransmission(false);
  Wire.requestFrom(BMI160_I2C_ADDRESS, 6);
 
  if (Wire.available() == 6) {
    ax = (Wire.read() | (Wire.read() << 8));
    ay = (Wire.read() | (Wire.read() << 8));
    az = (Wire.read() | (Wire.read() << 8));
  }
 
  // Convert raw accelerometer values to m/s^2
  float ax_mps2 = ax /ACCEL_SENSITIVITY *9.81;
  float ay_mps2 = ay /ACCEL_SENSITIVITY *9.81;
  float az_mps2 = az /ACCEL_SENSITIVITY *9.81;
 
  // Print accelerometer values in m/s^2
  Serial.print("Acceleration (m/s^2): ");
  Serial.print(ax_mps2-0.1, 2);
  Serial.print(", ");
  Serial.print(ay_mps2+0.7, 2);
  Serial.print(", ");
  Serial.print(az_mps2, 2);
  Serial.print(". \t");

  // Convert raw accelerometer values to g
  float ax_g = ax / ACCEL_SENSITIVITY;
  float ay_g = ay / ACCEL_SENSITIVITY;
  float az_g = az / ACCEL_SENSITIVITY;
 
  // Calculate tilt angles (pitch and roll) in degrees
  float pitch = atan2(ay_g, sqrt(ax_g * ax_g + az_g * az_g)) * 180.0 / PI;
  float roll = atan2(-ax_g, az_g) * 180.0 / PI;
 
  // Print tilt angles
  Serial.print("Pitch: ");
  Serial.print(pitch+4.5, 2);
  Serial.print("°, Roll: ");
  Serial.print(roll+178.6, 2);
  Serial.println("°");
 
  delay(500);
}

void autoCalibrateAccelerometer() {
  // Configure accelerometer for auto-calibration
  Wire.beginTransmission(BMI160_I2C_ADDRESS);
  Wire.write(0x7E); // Command register
  Wire.write(0x37); // Start accelerometer offset calibration
  Wire.endTransmission();
  delay(100);
 
  // Wait for calibration to complete
  delay(1000);
  Serial.println("Accelerometer Auto-Calibration Complete");
}
```

### Debug Message

```plain
E (1351) i2c.master: I2C transaction unexpected nack detected 
E (1351) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed 
E (1352) i2c.master: i2c_master_transmit_receive(1220): I2C transaction failed
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-05-13:
We know that there is currently a problem with the new ESP-IDF I2C driver and are working on resolving it. Closing as duplicate of #11228

closed 2025-05-13 by me-no-dev

JakubAndrysek, 2025-10-02:
Hi, I'm investigating this issue and related ones (e.g., [Arduino ESP32 #11228](https://github.com/espressif/arduino-esp32/issues/11228)). To clarify:

Are you reporting that when using the XIAO ESP32C6 or ESP32C3 Super Mini boards with ESP Arduino version 3.1.3, the I2C communication with the BMI160/ICM-20948 works fine? Meaning, you can read accelerometer data without errors when the device is connected (using the code you provided).

However, after updating to the ESP32 board version 3.2.0 (which uses ESP-IDF 5.4.1), you start seeing continuous I2C error messages on the serial monitor, and you are not able to get any further data from the sensor with the same code and wiring.

sorka-34, 2025-10-03:
@JakubAndrysek 
Hello, if I may share my input here. I'm using the XIAO ESP32C3 by Seeed Studio, with an SHT20 weatherproof module manufactured by Liyuan Technology. When using version 3.1.3 it all works fine with no error messages, and accurate temperature and humidity readings. Updating to verion 3.3.1 results in I2C NACK error messages experienced above, the readings remain accurate though. These error messages can be suppressed using:
  esp_log_level_set("i2c.master", ESP_LOG_NONE);
Hope that helps