espressif/arduino-esp32 #11361: Release notes fail to mention older MBEDTLS version

dirkx, 2025-05-13:
### Board

All

### Device Description

any

### Hardware Configuration

any

### Version

v3.2.0

### IDE Name

any

### Operating System

any

### Flash frequency

any

### PSRAM enabled

yes

### Upload speed

any

### Description

From the migration documentation

        https://github.com/espressif/arduino-esp32/blob/543fad2/docs/en/migration_guides/2.x_to_3.0.rst

it is not readily apparent that the MBED TLS version has changed (backwards).

And that matters, for example in 2.0.17,  mbedtls_sha256_starts_ret() replaces the legacy mbedtls_sha256_starts()

While the MBED TLS version in 3.2.0 does not yet have mbedtls_sha256_starts_ret() but only mbedtls_sha256_starts().

### Sketch

```cpp
#include "mbedtls/sha256.h"

void setup() {
    mbedtls_sha256_context sha_ctx;
    mbedtls_sha256_init(&sha_ctx);
    mbedtls_sha256_starts_ret(&sha_ctx, 0);
}
void loop() { }
```

### Debug Message

```plain
na
```

### Other Steps to Reproduce

na

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2025-05-13:
What makes you think that is newer?  AFAICT, esp-idf is running the latest release of mbedtls (3.6.3), while arduino-esp32 v2.0.17 is running 2.28.8

dirkx, 2025-05-14:
Hm - that is odd - afaiks mbedtls_sha256_starts() was marked depcrecated; and renamed to mbedtls_sha256_starts_ret() as of.after 2.7.0:

[image]

Did it revert back then in 3.0 somehow ?

Jason2866, 2025-06-06:
Arduino Core 3.2.0 uses IDF 5.4.1 and this mbedtls version is exactly used https://github.com/espressif/esp-idf/tree/release/v5.4/components/mbedtls

closed 2025-06-06 by Jason2866