espressif/arduino-esp32 #11364: ESP32 Fails to Connect to Enterprise WiFi (PEAP) Due to TLS 1.0 Incompatibility — WL_NO_SSID_AVAIL Reported

Tahir-Innok, 2025-05-14:
### Board

ESP32 Dev Module

### Device Description

DevKit

### Hardware Configuration

NO

### Version

latest stable Release (if not listed below)

### IDE Name

VSCode

### Operating System

Ubuntu

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

I'm attempting to connect an ESP32-based board to an enterprise WiFi network using WPA2-Enterprise with PEAP (Protected EAP) and MSCHAPv2 authentication. However, the connection fails.

The enterprise network supports only TLS 1.0, which is now deprecated and appears to be unsupported by current ESP32 firmware (ESP-IDF / Arduino core).

When initiating a connection, the ESP32 reports a WL_DISCONNECTED status for several seconds, then transitions to WL_NO_SSID_AVAIL, even though the network is visible and broadcasting.

### Sketch

```cpp
status = WiFi.begin(
      network_config.ssid.c_str(),
      WPA2_AUTH_PEAP,
      network_config.identity.c_str(),
      network_config.username.c_str(),
      network_config.password.c_str()
    );
```

### Debug Message

```plain
After debugging alittle I understand The TLS version negotiation is handled internally by the ESP-IDF WPA2 Enterprise stack (based on wpa_supplicant), and it uses the highest possible TLS version supported by both the client and server. 

But still my question is it possible to change the TLS version to 1.0 only for WiFi stack. As for my use case its almost impossible to change/upgrade the hardware.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2025-05-15:
Correct no support for TLS 1.0. If you want to use espressif devices either use older core or updating the wifi hardware is needed.

Tahir-Innok, 2025-05-15:
Thanks for your response @Jason2866 , can you please share which version/release have the support for TLS1.0.