espressif/arduino-esp32 #11368: ESP32 C3 Preferences lib conflict

ITstreet1, 2025-05-16:
### Board

ESP32 C3 WROOM 02

### Device Description

ESP32 C3

### Hardware Configuration

Nothing

### Version

latest stable Release (if not listed below)

### IDE Name

Arduino IDE

### Operating System

Windows

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Hello,

I have some conflict with Preferences lib as the C3 goes into continuous restarts.

Here is the sketch.

What am I doing wrong?

### Sketch

```cpp
#include <Preferences.h>
Preferences preferences;
int rly = 12;
int led = 0;

void setup(){
  Serial.begin(9600);
  Serial.println();
  preferences.begin("data", false);
  pinMode(led, OUTPUT);
  pinMode(rly, OUTPUT);
  digitalWrite(rly, HIGH);
  digitalWrite(led, HIGH);
}
void loop(){
  Serial.println("test");
  delay(1000);
}
```

### Debug Message

```plain
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380000
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1148
load:0x403cc710,len:0xb40
load:0x403ce710,len:0x2f58
entry 0x403cc710
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2025-05-16:
@ITstreet1 - please enable the `Core Debug Level: "Verbose` and `Erase All Flash Before Sketch Upload: "Enabled"`
Please send back here the log output in order to allow us to understand better what may be happening.

ITstreet1, 2025-05-16:
Here is the output:

> ESP-ROM:esp32c3-api1-20210207
> Build:Feb  7 2021
> rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fcd5820,len:0x1148
> load:0x403cc710,len:0xb40
> load:0x403ce710,len:0x2f58
> entry 0x403cc710
> [     0][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42003bfc
> [    12][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42003bea
> [    23][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42003bd8
> [    34][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42003bc6
> [    46][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42003bfc
> [    57][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42003bea
> [    68][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42003bd8
> [    80][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42003bc6
> [    91][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type USB_DM (38) successfully set to 0x420020a8
> [   103][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type USB_DP (39) successfully set to 0x420020a8
> [   118][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 20 successfully set to type UART_RX (2) with bus 0x3fc8deac
> [   129][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 21 successfully set to type UART_TX (3) with bus 0x3fc8deac
> [   140][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 18 successfully set to type USB_DM (38) with bus 0x3fc8f974
> [   151][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 19 successfully set to type USB_DP (39) with bus 0x3fc8f974

ITstreet1, 2025-05-16:
The answer was here:
rst:0x8 (TG1WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)

There is no such thing as GPIO12 available for users. It was my typo, as I am using GPIO10.

closed 2025-05-16 by ITstreet1

SuGlider, 2025-05-16:
@ITstreet1 - happy that you have found the issue and it is working fine now.