espressif/arduino-esp32 #11374: I2C driver behaviour after moving from 2.X.X to 3.2

handmade0octopus, 2025-05-17:
### Board

ESP32-S3 custom board

### Device Description

Not relevant, same HW works perfectly on 2.X.X

### Hardware Configuration

```cpp
	#define I2C_SCL 	42
	#define I2C_SDA 	41
```

### Version

latest stable Release (if not listed below)

### IDE Name

Platformio

### Operating System

w11

### Flash frequency

nr

### PSRAM enabled

yes

### Upload speed

nr

### Description

Randomly (eg after few minutes of querying) I get errors like this:
```
E (190508) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (190508) i2c.master: i2c_master_receive(1240): I2C transaction failed
[187037][E][esp32-hal-i2c-ng.c:311] i2cRead(): i2c_master_receive failed: [259] ESP_ERR_INVALID_STATE
[187046][E][Wire.cpp:523] requestFrom(): i2cRead returned Error 259
```

And after I have store prohibited:
```
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40379f6c  PS      : 0x00060033  A0      : 0x80377f34  A1      : 0x3fc9c570
A2      : 0x3c1f1460  A3      : 0xffffffff  A4      : 0x3c1f1764  A5      : 0x3c1f1764
A6      : 0x3fc9c570  A7      : 0x3c1f1494  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x60013000  A13     : 0x3fcc78bc
A14     : 0x00000001  A15     : 0x0000abab  SAR     : 0x00000000  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff

Backtrace: 0x40379f69:0x3fc9c570 0x40377f31:0x3fc9c5b0 0x4037a76d:0x3fc9c5d0 0x4200d8d3:0x3fcbca00 0x42009269:0x3fcbca30 0x4200d53f:0x3fcbca60 0x403807a6:0x3fcbca90
  #0  0x40379f69 in i2c_ll_read_rxfifo at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/hal/esp32s3/include/hal/i2c_ll.h:703
      (inlined by) i2c_isr_receive_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_driver_i2c/i2c_master.c:676
      (inlined by) i2c_master_isr_handler_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_driver_i2c/i2c_master.c:714
  #1  0x3fc9c570 in _xt_exception_table at ??:?
  #2  0x40377f31 in shared_intr_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/intr_alloc.c:464
  #3  0x3fc9c5b0 in _xt_exception_table at ??:?
  #4  0x4037a76d in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/xtensa/xtensa_vectors.S:1240
  #5  0x3fc9c5d0 in _xt_exception_table at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:67
  #6  0x4200d8d3 in PhysicalButtonHandle::handle() at src/hardware/Input.hpp:121 (discriminator 1)
  #7  0x42009269 in handleInput() at src/hardware/Input.hpp:155
  #8  0x4200d53f in inputTask(void*) at src/hardware/Input.hpp:174
  #9  0x403807a6 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

Line in question is..:
```cpp
button = !gpio_get_level((gpio_num_t)pin);
```
I used Arduino version before because I though this is the problem but it isn't.

Sometime after message I do not get store prohibited!

### Sketch

```cpp
I use ADXL from Adafruit library and I pass I2CBME that I setup beforehand. It worked fine for few years on old Arduino so not sure if new I2C is causing issues with Adafruit I2C device  but it was working great for ages in my setup.

Adafruit_ADXL345_Unified accel = Adafruit_ADXL345_Unified(&I2CBME);
// ...
I2CBME.begin((int) I2C_SDA, (int) I2C_SCL, 1000000UL);
I2CBME.setTimeOut(1);

accel.begin(0x1D);

accel.setDataRate(ADXL345_DATARATE_100_HZ);
accel.setRange(ADXL345_RANGE_4_G);

Then I call this every 10ms:

union AccelObject {
	uint8_t buff[6];
	struct {
		int16_t x;
		int16_t y;
		int16_t z;
	};
} accelObj;

// ...
loop {
// ...
accel.readBuff(ADXL345_REG_DATAX0, accelObj.buff, 6);
// ...
}
```

### Debug Message

```plain
Sometimes it doesn't even reach reading GPIO:

E (21656) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (21656) i2c.master: i2c_master_receive(1240): I2C transaction failed
[ 21667][E][esp32-hal-i2c-ng.c:311] i2cRead(): i2c_master_receive failed: [259] ESP_ERR_INVALID_STATE
[ 21676][E][Wire.cpp:523] requestFrom(): i2cRead returned Error 259
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40379f6c  PS      : 0x00060033  A0      : 0x80377f34  A1      : 0x3fc9c570
A2      : 0x3c1f1460  A3      : 0xffffffff  A4      : 0x3c1f1764  A5      : 0x3c1f1764
A6      : 0x3fc9c570  A7      : 0x3c1f1494  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x60013000  A13     : 0x3fcc78bc
A14     : 0x00000001  A15     : 0x0000abab  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff

Backtrace: 0x40379f69:0x3fc9c570 0x40377f31:0x3fc9c5b0 0x4037a76d:0x3fc9c5d0 0x4200d53a:0x3fcbca60 0x403807a6:0x3fcbca90
  #0  0x40379f69 in i2c_ll_read_rxfifo at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/hal/esp32s3/include/hal/i2c_ll.h:703
      (inlined by) i2c_isr_receive_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_driver_i2c/i2c_master.c:676
      (inlined by) i2c_master_isr_handler_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_driver_i2c/i2c_master.c:714
  #1  0x3fc9c570 in _xt_exception_table at ??:?
  #2  0x40377f31 in shared_intr_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/intr_alloc.c:464
  #3  0x3fc9c5b0 in _xt_exception_table at ??:?
  #4  0x4037a76d in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/xtensa/xtensa_vectors.S:1240
  #5  0x3fc9c5d0 in _xt_exception_table at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:67
  #6  0x4200d53a in inputTask(void*) at src/hardware/Input.hpp:173
  #7  0x403807a6 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

CancianMatteo, 2025-05-21:
Seems to be a known issue, maybe duplicate of [#11351](https://github.com/espressif/arduino-esp32/issues/11351) and [#11228](https://github.com/espressif/arduino-esp32/issues/11228)

psykokwak-com, 2025-06-06:
Also have this error:
```
E (190508) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (190508) i2c.master: i2c_master_receive(1240): I2C transaction failed
[187037][E][esp32-hal-i2c-ng.c:311] i2cRead(): i2c_master_receive failed: [259] ESP_ERR_INVALID_STATE
[187046][E][Wire.cpp:523] requestFrom(): i2cRead returned Error 259
```

AcuarioCat, 2025-06-15:
On ESP32.com there is a note that this can be fixed as below. It seems that this needs to be fixed in a file that, in the Arduino implementation, is pre-compiled as it is part of the esp-idf so without compiling that and updating the relevant files the only solution seems to be to revert to a previous build.

`To resolve the bug.You have to update i2c_master.c and change from

    I2C_CLOCK_SRC_ATOMIC() {
        i2c_hal_set_bus_timing(hal, i2c_dev->scl_speed_hz, i2c_master->base->clk_src, i2c_master->base->clk_src_freq_hz);
    }

    // Set the timeout value
    i2c_hal_master_set_scl_timeout_val(hal, i2c_dev->scl_wait_us, i2c_master->base->clk_src_freq_hz);
to

    // Set the timeout value
    i2c_hal_master_set_scl_timeout_val(hal, i2c_dev->scl_wait_us, i2c_master->base->clk_src_freq_hz);

    I2C_CLOCK_SRC_ATOMIC() {
        i2c_hal_set_bus_timing(hal, i2c_dev->scl_speed_hz, i2c_master->base->clk_src, i2c_master->base->clk_src_freq_hz);
    }`

psykokwak-com, 2025-06-15:
Sounds great! So we have to wait the next release.
@me-no-dev ?

everslick, 2025-07-22:
I can provide some anecdata that might be related to this issue. We have ~800 deployed devices, all permanently online. Approximately 4–5% of devices eventually (at any given time) lose the ability to communicate over I2C (I2C devices: AT24C32 (EEPROM), RV-3028-C7 (RTC)). The bus appears completely unresponsive. SDA and SCL lines remain HIGH — no low-level stuck condition detected.

And the real bummer is: I2C bus recovery or re-initialization in software does not help. Only a hardware reset or power cycle restores functionality. A software reboot (`esp_restart()`) does not fix the issue. That means that we cannot recover without user intervention and have to deal with this condition in software. You can imagine how annoying that is.

Everything else, is extremely stable and reliable. No reboots, not crashes.

Hardware:

ESP32-WROVER-E (16MB flash), good power supply, 2x100uF bypass cap. 

```
// I2C pins
#define HWPIN_SCL         22
#define HWPIN_SDA         23
```

Pullups: 1K5
BusClock: 100kHz

Software:

PIN_ESP32_3.0.0 ?= "release/v3.0.x" # 2024/12/18

Notes:

    This issue has been observed across multiple firmware builds / Core versions and IDF versions.
    Seems independent of clock stretching or bus load.
    Devices use Arduino framework under the ESP32 Arduino core.

I'm inclined to believe that there is a hardware bug in the ESP32 I2C state machine. I **KNOW** that its not one of the i2c devices hogging the bus and that its not the devices that are unresponsive (otherwise resetting via ESP32 EN pin would not recover communication). I also **KNOW**, that `esp_restart()` does not help. It happens randomly with no obvious external factors (at least non which I could observe).

schneeer, 2025-07-22:
@everslick 
Du hast ja wenigstens Ahnung.
Ich bin ein fast 60 Jähriger alter Knacker mit elektrotechnicher Vorbelastung, der seit 4 Jahren immer verücktere Sachen zusammen bastelt. Ich hatte von Programmiersprachen bis dahin keinen blassen Schimmer. Bissel Basic habe ich vorher mal versucht und tasächlich ein Programm hinbekommen, welches 5 Leuten pro Jahr 10 € Wert ist. (Es ist kostenlos, für spezielle Extras wird eine personalisierte Seriennummer benötigt.)
Was andere studieren habe ich krümelweise im Internet zusammengesucht, versucht zu verstehen, für mich passend gemacht und mich ansonsten selbst versucht.
Ich bekomme da eine Kriese, wenn plötzlich Probleme auftauchen und nix mehr geht. Ich gehe ja dann erstmal von einem Hardwardefekt aus. Da muss man erstmal darauf kommen, das eins von mehreren Updates schuld ist und das ist auch noch zu finden!

@me-no-dev 
It doesn't matter whether it's the driver or not.
The IDF, like the ESP32s, are from Expressif, and the drivers also ultimately come from Expressif.
And the fact is that several problems arise with version 3.1.20 and higher: I2C, Neopixel...
And another fact is: downgrading to 3.1.13 helps.
Who else besides Expressif messes with that?!?

I can't provide an example sketch, but I2C works in my current project.
I had problems with a second project using Neopixels and Adafruit libraries, but downgrading the ESP32 board extension to 3.1.13 also helped.
(Thanks Google Translate)

schneeer, 2025-09-19:
I have to correct myself.
My errors were apparently somewhat different.
I reported this under "Problems" here and included the error codes. My error reports were closed with the note that it wasn't Espressif's fault and also because they were written in German.
Everything is also in my [previous post](https://github.com/espressif/arduino-esp32/issues/11374#issuecomment-3104366448).

I had error messages about unreachable slaves. The slaves actually don't exist yet, so logically, not all of them are reachable. I didn't realize that right away; as a German-speaking idiot, I don't understand the cryptic error messages.

The errors don't appear with version 3.1.3, only with higher versions.
I now check beforehand whether the slaves are reachable and only query the existing devices. This has eliminated the error messages.

This could have been sent to me in my error report or as a reply here. The experts here should have been able to recognize that from my error messages.

user-friendly-app, 2025-09-24:
I found a solution! I used a VL53L0X sensor and checked the voltage at its I2C outputs. I also desoldered the voltage regulator from the sensor board, as it was already receiving 3.3 volts, and everything worked. **The voltage at the sensor I2C outputs was low**.

me-no-dev, 2026-10-01:
Closing this as stale (no new reports have been filed). We can reopen if it's experienced again on the latest version

closed 2026-10-01 by me-no-dev