espressif/arduino-esp32 #11375: Bad HTTP response

ITstreet1, 2025-05-17:
### Board

ESP32 C3

### Device Description

Devkit

### Hardware Configuration

Nothing

### Version

latest stable Release (if not listed below)

### IDE Name

Arduino IDE

### Operating System

Win10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I tried to use my older code but always get a bad HTTP response. Can someone point me what to change?
The URL is ok, as I get the valid data over the browser with HTTP code 200.

### Sketch

```cpp
#include <WiFi.h>
#include <HTTPClient.h>
WiFiClient client;

const char* ssid = "SSID";
const char* password = "PASS";
const int deviceID = 2;
String server = "www.somesite.com";

void setup(){
  Serial.begin(9600);
  Serial.println();
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while((WiFi.status() != WL_CONNECTED)) { 
    delay(500);
    Serial.print(".");
  }
  Serial.println();
  Serial.println("Connected!");

}
void loop(){
  Serial.println("test");
  if(WiFi.status()== WL_CONNECTED){
      HTTPClient http;
      String serverPath = server;
      Serial.println(serverPath);
      Serial.println("-----");
      http.begin(client, serverPath.c_str());
      int httpResponseCode = http.GET();
      if(httpResponseCode>0){
        Serial.print("HTTP Response code: ");
        Serial.println(httpResponseCode);
      }else{
        Serial.print("Error code: ");
        Serial.println(httpResponseCode);
      }
      http.end();
    }else{
      Serial.println("WiFi Disconnected");
    }
  delay(5000);
}
```

### Debug Message

```plain
I always get -1 as HTTP code. Verbose output this:

[  6031][V][HTTPClient.cpp:259] beginInternal(): url: www.somesite.com
[  6315][D][HTTPClient.cpp:314] beginInternal(): protocol: http, host: www.somesite.com port: 80 url: ....//link is not important
[  6505][D][HTTPClient.cpp:621] sendRequest(): request type: 'GET' redirCount: 0

[  6593][D][NetworkManager.cpp:83] hostByName(): Clearing DNS cache
[  6670][D][NetworkManager.cpp:127] hostByName(): DNS found IPv4 *****
[  6776][E][NetworkClient.cpp:293] connect(): Setsockopt 'SO_SNDTIMEO'' on fd 48 failed. errno: 22, "Invalid argument"
[  6901][D][HTTPClient.cpp:1248] connect(): failed connect to www.somesite.com:80
[  6983][W][HTTPClient.cpp:1604] returnError(): error(-1): connection refused
Error code: -1
[  7066][D][HTTPClient.cpp:434] disconnect(): tcp is closed
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

ITstreet1, 2025-05-18:
HTTPClient.h line 47 clearly says #define HTTPC_ERROR_CONNECTION_REFUSED  (-1), which is what I get on the output, but why? What is changed in v3.x.x? What should be changed in begin to work properly?

From what I have seen, it all started from V3.x.x. There are a bunch of similar problems on the internet.

lbernstone, 2025-05-18:
I'd recommend you use `curl -v <url>` or your browser's network tools to see what the transaction looks like when it completes.  There is likely a redirect there to https or perhaps http 2.0. You could also check the verbose log of an earlier version of arduino-esp32 to see what the server response looks like.

ITstreet1, 2025-05-19:
When I use the very same URL address in a browser, I get what I expect, and get 200 response code.

When I use this same URL here, I get -1

I tired your suggestion on my PC and get this:

> * Host site.com:80 was resolved.
> * IPv6: (none)
> * IPv4: *****
> *   Trying *****:80...
> * Connected to site.com(*****) port 80
> > GET /index.php?try=in HTTP/1.1
> > Host: site.com
> > User-Agent: curl/8.9.1
> > Accept: */*
> >
> * Request completely sent off
> < HTTP/1.1 406 Not Acceptable
> < Date: Mon, 19 May 2025 08:14:01 GMT
> < Server: Apache/2.0.64 (Unix)
> < Content-Length: 404
> < Content-Type: text/html; charset=iso-8859-1
> <
> <!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
> <html><head>
> <title>406 Not Acceptable</title>
> </head><body>
> <h1>Not Acceptable</h1>
> <p>An appropriate representation of the requested resource /index.php could not be found on this server.</p>
> <p>Additionally, a 404 Not Found
> error was encountered while trying to use an ErrorDocument to handle the request.</p>
> </body></html>
> * Connection #0 to host site.com left intact
> 'deviceID' is not recognized as an internal or external command,
> operable program or batch file.
> 'rlyStatOff1' is not recognized as an internal or external command,
> operable program or batch file.
> 'rlyStatOff2' is not recognized as an internal or external command,
> operable program or batch file.

I downgraded ESP32 board definitions to 2.0.16 and get:

> [  3258][V][HTTPClient.cpp:259] beginInternal(): url: http://site.com/index.php?try=in&deviceID=2&rlyStatOff1=0&rlyStatOff2=0
> [  3552][D][HTTPClient.cpp:314] beginInternal(): protocol: http, host: site.com port: 80 url: /index.php?try=in&deviceID=2&rlyStatOff1=0&rlyStatOff2=0
> [  3742][D][HTTPClient.cpp:621] sendRequest(): request type: 'GET' redirCount: 0
> 
> [  3852][E][WiFiClient.cpp:282] connect(): Setsockopt 'SO_SNDTIMEO'' on fd 48 failed. errno: 22, "Invalid argument"
> [  3974][D][HTTPClient.cpp:1248] connect(): failed connect to site.com:80
> [  4057][W][HTTPClient.cpp:1604] returnError(): error(-1): connection refused
> Error code: -1
> [  4139][D][HTTPClient.cpp:434] disconnect(): tcp is closed

This same code works on ESP8266. Different libs, though.

ITstreet1, 2025-05-19:
I tried 2.0.2 and it works on ESP32.
The versions I mentioned above didn't worked even on a barebone ESP32. But this 2.0.2 don't have support for C3...

So, it is not the problem with the server, it is with the libraries. I am not sure what I wrote wrong in the code, but obviously it need to be changed to adopt the newer packages of support for the ESP32 series.

me-no-dev, 2025-05-19:
I have one suggestion. Comment out `return 0;` here: https://github.com/espressif/arduino-esp32/blob/master/libraries/Network/src/NetworkClient.cpp#L290 and try again. Please also post the verbose logs

ITstreet1, 2025-05-19:
The newest ESP32 packet version from the Arduino IDE.
When I comment out the line you suggested, it couldn't compile.

me-no-dev, 2025-05-19:
Since it's part of `#define`, you can just delete the whole line and that should do it

me-no-dev, 2025-05-20:
@ITstreet1 please post the verbose debug log from serial with that change

ITstreet1, 2025-05-20:
Hello,
The verbose for ESP32 C3 outputs all of this:

> [  4841][V][HTTPClient.cpp:259] beginInternal(): url: http://site.com/index.php?try=in&deviceID=2&rlyStatOff1=0&rlyStatOff2=0
> [  5125][D][HTTPClient.cpp:314] beginInternal(): protocol: http, host: site.com port: 80 url: /index.php?try=in&deviceID=2&rlyStatOff1=0&rlyStatOff2=0
> [  5315][D][HTTPClient.cpp:621] sendRequest(): request type: 'GET' redirCount: 0
> 
> [  5403][D][NetworkManager.cpp:83] hostByName(): Clearing DNS cache
> [  5484][D][NetworkManager.cpp:127] hostByName(): DNS found IPv4 194.106.162.117
> [  5587][E][NetworkClient.cpp:292] connect(): Setsockopt 'SO_SNDTIMEO'' on fd 48 failed. errno: 22, "Invalid argument"
> [  5712][E][NetworkClient.cpp:293] connect(): Setsockopt 'SO_RCVTIMEO'' on fd 48 failed. errno: 22, "Invalid argument"
> [  5837][D][HTTPClient.cpp:1255] connect():  connected to site.com:80
> [  5916][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  6309][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  6410][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  6512][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  6613][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  6714][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  6816][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  6917][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  7018][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  7119][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  7221][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  7322][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  7423][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  7524][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  7626][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  7727][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  7828][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  7929][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  8031][V][HTTPClient.cpp:1357] handleHeaderResponse(): RX: 'HTTP/1.1 200 OK'
> [  8114][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  8216][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  8317][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  8418][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  8519][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  8620][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  8722][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  8823][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  8924][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  9025][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  9127][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  9228][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  9329][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  9430][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  9532][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  9633][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  9734][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  9836][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [  9937][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [ 10038][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [ 10139][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [ 10240][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [ 10342][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [ 10443][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [ 10544][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [ 10645][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [ 10747][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
> [ 10848][E][NetworkClient.cpp:326] setSocketOption(): fail on 48, errno: 22, "Invalid argument"
[... cut ...]
      String serverPath = server+"?try=in&deviceID="+deviceID+"&rlyStatOff1="+rlyStatOff1+"&rlyStatOff2="+rlyStatOff2;
      Serial.println(serverPath);
      Serial.println("-----");
      http.begin(client, serverPath.c_str());
      int httpResponseCode = http.GET();
      if(httpResponseCode>0){
        Serial.print("HTTP Response code: ");
        Serial.println(httpResponseCode);
      }else{
        Serial.print("Error code: ");
        Serial.println(httpResponseCode);
      }
      http.end();
    }else{
      Serial.println("WiFi Disconnected");
    }
  delay(5000);
}
```

ITstreet1, 2025-05-20:
And I must repeat myself. This code worked with zero issues on older versions of ESP32 support packages. This code works with ESP8266, too. With ESP8266 libs and support, though.

me-no-dev, 2025-05-20:
You do not need to repeat yourself... see my output of your code below
```
test
http://site.com/index.php?try=in&deviceID=2&rlyStatOff1=0&rlyStatOff2=0
-----
[  3980][V][HTTPClient.cpp:242] beginInternal(): url: http://site.com/index.php?try=in&deviceID=2&rlyStatOff1=0&rlyStatOff2=0
[  3998][D][HTTPClient.cpp:293] beginInternal(): protocol: http, host: site.com port: 80 url: /index.php?try=in&deviceID=2&rlyStatOff1=0&rlyStatOff2=0
[  4011][D][HTTPClient.cpp:574] sendRequest(): request type: 'GET' redirCount: 0

[  4018][D][NetworkManager.cpp:83] hostByName(): Clearing DNS cache
[  4052][D][NetworkManager.cpp:127] hostByName(): DNS found IPv4 34.211.108.46
[  4370][D][HTTPClient.cpp:1112] connect():  connected to site.com:80
[  4679][V][HTTPClient.cpp:1201] handleHeaderResponse(): RX: 'HTTP/1.1 301 Moved Permanently'
[  4687][V][HTTPClient.cpp:1201] handleHeaderResponse(): RX: 'Date: Tue, 20 May 2025 11:17:50 GMT'
[  4696][V][HTTPClient.cpp:1201] handleHeaderResponse(): RX: 'Transfer-Encoding: chunked'
[  4705][V][HTTPClient.cpp:1201] handleHeaderResponse(): RX: 'Connection: keep-alive'
[  4713][V][HTTPClient.cpp:1201] handleHeaderResponse(): RX: 'Cache-Control: no-cache,must-revalidate,max-age=0,no-store,private'
[  4724][V][HTTPClient.cpp:1201] handleHeaderResponse(): RX: 'X-Content-Type-Options: nosniff'
[  4733][V][HTTPClient.cpp:1201] handleHeaderResponse(): RX: 'Expires: Thu, 01 Jan 1970 00:00:00 GMT'
[  4742][V][HTTPClient.cpp:1201] handleHeaderResponse(): RX: 'Set-Cookie: CookieConsentPolicy=1:1; path=/; expires=Wed, 20-May-2026 11:17:50 GMT; Max-Age=31536000; secure'
[  4758][V][HTTPClient.cpp:1201] handleHeaderResponse(): RX: 'Set-Cookie: LSKey-c$CookieConsentPolicy=1:1; path=/; expires=Wed, 20-May-2026 11:17:50 GMT; Max-Age=31536000; secure'
[  4774][V][HTTPClient.cpp:1201] handleHeaderResponse(): RX: 'Location: https://site.com/index.php?try=in&deviceID=2&rlyStatOff1=0&rlyStatOff2=0'
[  4787][V][HTTPClient.cpp:1201] handleHeaderResponse(): RX: 'Content-Security-Policy: upgrade-insecure-requests'
[  4798][V][HTTPClient.cpp:1201] handleHeaderResponse(): RX: 'Server: sfdcedge'
[  4805][V][HTTPClient.cpp:1201] handleHeaderResponse(): RX: 'X-SFDC-Request-Id: 6f709a681f53407729018a246267c8b0'
[  4815][V][HTTPClient.cpp:1201] handleHeaderResponse(): RX: 'X-Request-Id: 6f709a681f53407729018a246267c8b0'
[  4825][V][HTTPClient.cpp:1201] handleHeaderResponse(): RX: ''
[  4831][D][HTTPClient.cpp:1257] handleHeaderResponse(): code: 301
[  4837][D][HTTPClient.cpp:1264] handleHeaderResponse(): Transfer-Encoding: chunked
[  4845][D][HTTPClient.cpp:618] sendRequest(): sendRequest code=301

HTTP Response code: 301
[  4851][D][HTTPClient.cpp:373] disconnect(): still data in buffer (5), clean up.

[  4860][D][HTTPClient.cpp:378] disconnect(): tcp keep open for reuse
```

It's not the same as yours (and I did not make the edit that I asked you to do) so what gives?

ITstreet1, 2025-05-20:
Mate, with the exact code with site.com, I get -1 as HTTP response.
With the erased line you suggested, I get 200, but there are other problems. It looks like http.getString(); returns nothing. I will try on other PC, just to see if there is some lib conflict here. I have no idea what is the problem.
But the older package versions work. So it makes me a bit confused.

ITstreet1, 2025-05-20:
I believe here is nothing much to see:

> Sketch uses 1003690 bytes (76%) of program storage space. Maximum is 1310720 bytes.
> Global variables use 34712 bytes (10%) of dynamic memory, leaving 292968 bytes for local variables. Maximum is 327680 bytes.
> esptool.py v4.8.1
> Serial port COM13
> Connecting....
> Chip is ESP32-C3 (QFN32) (revision v0.3)
> Features: WiFi, BLE
> Crystal is 40MHz
> MAC: 7c:df:a1:b2:bc:b4
> Uploading stub...
> Running stub...
> Stub running...
> Changing baud rate to 921600
> Changed.
> Configuring flash size...
> Flash will be erased from 0x00000000 to 0x00004fff...
> Flash will be erased from 0x00008000 to 0x00008fff...
> Flash will be erased from 0x0000e000 to 0x0000ffff...
> Flash will be erased from 0x00010000 to 0x00105fff...
> Compressed 19696 bytes to 12723...
> Writing at 0x00000000... (100 %)
> Wrote 19696 bytes (12723 compressed) at 0x00000000 in 0.5 seconds (effective 303.2 kbit/s)...
> Hash of data verified.
> Compressed 3072 bytes to 146...
> Writing at 0x00008000... (100 %)
> Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.1 seconds (effective 440.4 kbit/s)...
> Hash of data verified.
> Compressed 8192 bytes to 47...
> Writing at 0x0000e000... (100 %)
> Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 562.6 kbit/s)...
> Hash of data verified.
> Compressed 1003840 bytes to 605873...
> Writing at 0x00010000... (2 %)
> Writing at 0x0001d4a8... (5 %)
> Writing at 0x00028c91... (8 %)
> Writing at 0x00030b03... (10 %)
> Writing at 0x0003aa4a... (13 %)
> Writing at 0x00040764... (16 %)
> Writing at 0x00047394... (18 %)
> Writing at 0x0004de03... (21 %)
> Writing at 0x000542f5... (24 %)
> Writing at 0x0005abf9... (27 %)
> Writing at 0x00060b46... (29 %)
> Writing at 0x00066af0... (32 %)
> Writing at 0x0006cdcd... (35 %)
> Writing at 0x00073170... (37 %)
> Writing at 0x00079109... (40 %)
> Writing at 0x0007f17b... (43 %)
> Writing at 0x00084e41... (45 %)
> Writing at 0x0008a839... (48 %)
> Writing at 0x0009009e... (51 %)
> Writing at 0x00095a6a... (54 %)
> Writing at 0x0009c3dd... (56 %)
> Writing at 0x000a1cc5... (59 %)
> Writing at 0x000a7a0a... (62 %)
> Writing at 0x000ad486... (64 %)
> Writing at 0x000b3433... (67 %)
> Writing at 0x000b91cf... (70 %)
> Writing at 0x000bf636... (72 %)
> Writing at 0x000c5772... (75 %)
> Writing at 0x000cbc9a... (78 %)
> Writing at 0x000d1c4a... (81 %)
> Writing at 0x000d7d0c... (83 %)
> Writing at 0x000de348... (86 %)
> Writing at 0x000e4241... (89 %)
> Writing at 0x000ea1e8... (91 %)
> Writing at 0x000f3419... (94 %)
> Writing at 0x000f8b44... (97 %)
> Writing at 0x000fed2a... (100 %)
> Wrote 1003840 bytes (605873 compressed) at 0x00010000 in 16.9 seconds (effective 475.6 kbit/s)...
> Hash of data verified.
> 
> Leaving...
> Hard resetting via RTS pin...
>

me-no-dev, 2025-05-20:
I tried with C3 and ESP32 on Arduino 3.2.0 with the same exact results as I posted above. Really strange why you see the different output and require the line to be removed. Even then, you get 200, but I get 302... what is different? How you get different HTTP result code and why can't you set the timeout of the client?

ITstreet1, 2025-05-21:
I just tried it as is on another PC. The same board, the newest package support, everything went fine. The HTTP response, the getString(), so far zero problems

So, the problem must be in some conflict with the packages of other boards or libraries I have here. What causes it, I have no idea. I will try to solve it by erasing what I don't use.
Still, what bothers me the most is the fact that it works with v2.x.

me-no-dev, 2025-05-21:
> Still, what bothers me the most is the fact that it works with v2.x.

I understand but at the same time, you see that I get a very different result than you when running on recent versions. It's only on that particular computer of yours that you have the problem. It must be some mix of core versions and external libraries getting mixed in.

Closing now as resolved

Bms7, 2025-06-23:
Hello,
same problem with ESP32 Dev Arduino IDE 2.3.6 and commenting out

#define ROE_WIFICLIENT(x, msg)                                                                           \
  {                                                                                                      \
    if (((x) < 0)) {                                                                                     \
      log_e("Setsockopt '" msg "'' on fd %d failed. errno: %d, \"%s\"", sockfd, errno, strerror(errno)); \
      return 0;                                                                                          \
    }                                                                                                    \
  }
  ROE_WIFICLIENT(setsockopt(sockfd, SOL_SOCKET, SO_SNDTIMEO, &tv, sizeof(tv)), "SO_SNDTIMEO");
  ROE_WIFICLIENT(setsockopt(sockfd, SOL_SOCKET, SO_RCVTIMEO, &tv, sizeof(tv)), "SO_RCVTIMEO");

did the trick.

But the file was 
,\.arduino15\packages\esp32\hardware\esp32\2.0.17\libraries\WiFi\src\WiFiClient.cpp
in line 280

Thanks