espressif/arduino-esp32 #11377: In stream.cpp add readln, to simplify reading Windows and Linux text files

zingzog, 2025-05-20:
### Related area

ESP32 Reading text files and html files created in Windows or Linuxfrom LittleFS

### Hardware specification

Any ESP with LittleFS

### Is your feature request related to a problem?

Text files can be read using readBytesUntil with the LineFeed character but with Windows files the CR character has to be dealt with. readln would terminate on LF and ignore CR so that the user just sees c strings representing the text on each line.

### Describe the solution you'd like

I have been using this code as a solution:
`// as readBytes but terminates on LF (13), ignores CR (10)
// terminates if length characters have been read, timeout, or if the terminator character  detected
// returns the number of characters placed in the buffer (0 means no valid data found)

int Stream::readln(char *buffer, int length) {
  int index = 0;
  int c = 0;
  while (index < length) {
    c = timedRead();
    if (c < 0 || c == 13) {
      break;
    }
    if (c != 10){
      *buffer++ = (char)c;
      index++;
    }
  }
  *buffer = '\0';
  if (c < 0){ 
    return -1;
  }else{
    return index;  // return number of characters, not including null terminator
  }
}
`

### Describe alternatives you've considered

As mentioned earlier, text files can be read using readBytesUntil with the LineFeed character but with Windows files the CR character has to be dealt with. The advantages of readln are that:
- it neatly encapsulates the issues 
- it works with both LF and CRLF terminated strings
- no additional cleanup work needed by the programmer

### Additional context

A quick Google search shows a number of questions in Stack Overflow, etc. requesting a solution to reading lines from text files.
I am using the function with LittleFS to read configuration key=value pairs from text files, and to read lines from html files for modification, and removal.
For me it is simple and works reliably.

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2025-05-20:
Proper place to suggest this is in the official Arduino API repository here: https://github.com/arduino/ArduinoCore-API

Please submit the feature request there and if approved, we can add it on our end then.

zingzog, 2025-05-20:
Okidoki, I'll do that.

zingzog, 2025-05-20:
Done, with the title and serial number:
In stream.cpp add readln, to simplify reading Windows and Linux text files #251 
[https://github.com/arduino/ArduinoCore-API/issues/251](url)

Parsaabasi, 2026-01-06:
Dear @zingzog 

Seems like you have already reported this issue properly as suggested. I close this for ticket. 

Kind regards,

closed 2026-01-06 by Parsaabasi