espressif/arduino-esp32 #11378: ESP32 Matter Arduino not resuming session

theonlytechnohead, 2025-05-20:
### Board

ESP32 Dev Module

### Device Description

Plain module connected with jumpers

### Hardware Configuration

GPIO 12 & 13 are connected to LED strips with 60 SK6812 LEDs per strip
LEDs are powered externally

### Version

v3.2.0

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Hello,
Thank you for such an excellent experience setting up a Matter device!
I followed the ColorLight example to get started, however when the device reboots, I see some errors in the console.
The errors seem to appear roughly when `Matter.begin()` or `ColorLight.updateAccessory()` is called.

The device also produces errors when responding to some commands sent by routines in the Google Home app:
`(317140) chip[DMG]: Endpoint=1 Cluster=0x0000_0300 Command=0x0000_000A status 0x81 (no additional context)`
The Google Home app reports that an action failed.

Please let me know if there's anything else I can do to help debug these issues!

### Sketch

```cpp
#include <Matter.h>
#include <WiFi.h>

const char *ssid = "ssid ";
const char *password = "password ";

MatterColorLight ColorLight;

#include <FastLED.h>

#define LEFT_PIN 12
#define RIGHT_PIN 13

#define NUM_LEDS 60

CRGB left[NUM_LEDS];
CRGB right[NUM_LEDS];

TaskHandle_t matterTask;

bool state = true;
espHsvColor_t colour = { 0, 0, 0 };
espHsvColor_t target = { 140, 254, 127 };

void setup() {
  Serial.begin(115200);

  FastLED.addLeds<WS2812, LEFT_PIN, GRB>(left, NUM_LEDS).setRgbw(RgbwDefault());
  FastLED.addLeds<WS2812, RIGHT_PIN, GRB>(right, NUM_LEDS).setRgbw(RgbwDefault());

  // clear to black
  set_all(CRGB::Black);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    centre_wipe(CRGB::Yellow);
    delay(500);
    centre_wipe(CRGB::Black);
  }

  ColorLight.begin(state, target);

  ColorLight.onChange(update);

  xTaskCreatePinnedToCore(
    matter,        // Function to implement the task
    "matterTask",  // Name of the task
    8192,          // Stack size in words (causes stack overflow (DUH!!!!) if too low
    NULL,          // Task input parameter
    0,             // Priority of the task, 0 is lowest
    &matterTask,   // Task handle
    0);            // Core where the task should run, code runs on core 1 by default
}

void matter(void *parameter) {
  Matter.begin();
  vTaskDelay(500);
  // this may be a restart of an already commissioned Matter accessory
  if (Matter.isDeviceCommissioned()) {
    while (!Matter.isDeviceConnected()) {
      vTaskDelay(500);
      centre_wipe(CRGB::Orange);
      vTaskDelay(500);
      centre_wipe(CRGB::Black);
    }
    // configure the light based on initial state and colour
    ColorLight.updateAccessory();
  }

  for (;;) {
    vTaskDelay(10);
    if (!Matter.isDeviceCommissioned()) {
      Serial.println("");
      Serial.println("Matter Node is not commissioned yet.");
      Serial.println("Initiate the device discovery in your Matter environment.");
      Serial.println("Commission it to your Matter hub with the manual pairing code or QR code");
      Serial.printf("Manual pairing code: %s\r\n", Matter.getManualPairingCode().c_str());
      Serial.printf("QR code URL: %s\r\n", Matter.getOnboardingQRCodeUrl().c_str());
      // wait for Matter Light Commissioning.
      uint32_t timeCount = 0;
      while (!Matter.isDeviceCommissioned()) {
        while (WiFi.status() != WL_CONNECTED) {
          vTaskDelay(500);
          centre_wipe(CRGB::Yellow);
          vTaskDelay(500);
          centre_wipe(CRGB::Black);
        }
        vTaskDelay(500);
        centre_wipe(CRGB::Blue);
        vTaskDelay(500);
        centre_wipe(CRGB::Black);
        if ((timeCount++ % 50) == 0) {  // 50*100ms = 5 sec
          Serial.println("Matter Node not commissioned yet. Waiting for commissioning.");
          Serial.printf("Manual pairing code: %s\r\n", Matter.getManualPairingCode().c_str());
          Serial.printf("QR code URL: %s\r\n", Matter.getOnboardingQRCodeUrl().c_str());
        }
      }
    }
  }
}

void loop() {
  if (colour.h != target.h || colour.s != target.s || colour.v != target.v) {
    colour = { target.h, target.s, target.v };
    set_all(CHSV(colour.h, colour.s, colour.v));
    Serial.printf("loop colour changed to hsv(%d, %d, %d)\n", colour.h, colour.s, colour.v);
  }
}

bool update(bool s, espHsvColor_t c) {
  if (state != s) {
    Serial.printf("update state changed to %s\n", s ? "ON" : "OFF");
    state = s;
    if (state && 1 < c.v)
      target = { c.h, c.s, c.v };
    if (!state) {
      target = { 0, 0, 0 };
    }
    return true;
  }
  if (colour.h != c.h || colour.s != c.s || colour.v != c.v) {
    Serial.printf("update colour changed to hsv(%d, %d, %d)\n", c.h, c.s, c.v);
    if (state && 1 < c.v)
      target = { c.h, c.s, c.v };
    else
      target = { 0, 0, 0 };
    return true;
  }
  // This callback must return the success state to Matter
  return true;
}

void set_all(CRGB c) {
  for (int i = 0; i < NUM_LEDS; i++) {
    left[i] = c;
    right[i] = c;
  }
  FastLED.show();
}

void centre_wipe(CRGB c) {
  for (int i = 0; i < NUM_LEDS; i++) {
    left[i] = c;
    right[i] = c;
    FastLED.show();
  }
}
```

### Debug Message

```plain
E (11114) chip[DMG]: Endpoint 0, Cluster 0x0000_0031 not found in IncreaseClusterDataVersion!
E (11115) chip[DMG]: Endpoint 0, Cluster 0x0000_0031 not found in IncreaseClusterDataVersion!
update color changed to hsl(0, 0, 0)
E (12611) chip[IN]: SendMessage() to UDP:[REDACTED-IPV6-ADDRESS]:5540 failed: 3000004
E (12612) chip[-]: Error LwIP:0x03000004 at ./managed_components/espressif__esp_matter/connectedhomeip/connectedhomeip/src/app/OperationalSessionSetup.cpp:254
E (12623) chip[DMG]: Failed to establish CASE for subscription-resumption with error '3000004'
E (12648) chip[DMG]: Failed to establish CASE for subscription-resumption with error '3000004'
```

### Other Steps to Reproduce

The device is provisioned within Google Home via a Google Nest Mini

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2025-05-22:
`Endpoint=1 Cluster=0x0000_0300 Command=0x0000_000` is about the MatterColorLight Endpoint.
Cluster 0x0300  is ColorControl Cluster for Color Setting, suported by Google controller.
Command 0x0000 is MoveToHue command.
Status 0x81 is CHIP_ERROR_UNKNOWN_KEY_TYPE_FROM_PEER.

It was supposed to work fine. 
Could you please set Debug Level to Verbose and report the log when the Google Routine tries to chane the light color?

SuGlider, 2025-05-22:
When the device reboots, it may take some time to restabish the CASE session with the Google Matter Controller.
```
E (11114) chip[DMG]: Endpoint 0, Cluster 0x0000_0031 not found in IncreaseClusterDataVersion!
E (11115) chip[DMG]: Endpoint 0, Cluster 0x0000_0031 not found in IncreaseClusterDataVersion!
update color changed to hsl(0, 0, 0)
E (12611) chip[IN]: SendMessage() to UDP:[REDACTED-IPV6-ADDRESS]:5540 failed: 3000004
E (12612) chip[-]: Error LwIP:0x03000004 at ./managed_components/espressif__esp_matter/connectedhomeip/connectedhomeip/src/app/OperationalSessionSetup.cpp:254
E (12623) chip[DMG]: Failed to establish CASE for subscription-resumption with error '3000004'
E (12648) chip[DMG]: Failed to establish CASE for subscription-resumption with error '3000004'
```

It shall work after a couple minutes in the worst case.
If possible, reboot the ESP32 and wait for about 5 minutes.
In case it doesn't recover, please report the DEBUG logs for investigation.

SuGlider, 2025-05-22:
Google Controller seems to demand some updating. Please make sure that the Google Device has its firmware in the latest version.

Veletax, 2025-05-24:
Same issue and error message in my case with ESP32- C6 Dev Module  with the example MatterColorLigth (using RGB in-builtin LED) and using Home Assistant. It seems error is not related to the hub.

SuGlider, 2025-05-25:
@Veletax - When you say same error, what is the error message that you get?
There are a few `chip[DMG}:` messages that are ok, because those attributes are not mandatory.

But if the issue is about not restablishing the CASE session and the Matter Accessory is offline for long, this is an issue.

Please post the whole log output, description of the process to produce the issue and information about app/version/controller for good investigation.

theonlytechnohead, 2025-05-26:
Hi @SuGlider - thanks for taking interest in my issue!

My Google Nest Mini is running the [latest firmware](https://support.google.com/googlenest/answer/7365257?hl=en).

After looking through the verbose logs, I noticed that it takes much longer for the ESP32 to log that it has an IPv6 address, compared to IPv4.

I managed to rectify this by calling `WiFi.enableIPv6();` right after `WiFi.begin();`, and accordingly waiting until both the link-local and global IPv6 addresses are resolved.

This reduces the startup errors down to:

<details>
<summary>Log</summary

```
  ESP-IDF Version   : v5.4.1-1-g2f7dcd862a-dirty
  Arduino Version   : 3.2.0
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32_DEV
  Arduino Variant   : esp32
  Arduino FQBN      : esp32:esp32:esp32:UploadSpeed=921600,CPUFreq=240,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=min_spiffs,DebugLevel=verbose,PSRAM=disabled,LoopCore=1,EventsCore=1,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=default
============ Before Setup End ============
[   768][V][NetworkEvents.cpp:117] _checkForEvent(): Network Event: 101 - WIFI_READY
[   843][V][STA.cpp:186] _onStaEvent(): STA Started
[   847][V][NetworkEvents.cpp:117] _checkForEvent(): Network Event: 110 - STA_START
[   857][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 110 - STA_START
[  1732][V][STA.cpp:206] _onStaEvent(): STA Connected: SSID: <REDACTED>, BSSID: <REDACTED>, Channel: <REDACTED>, Auth: WPA3_PSK
[  1742][V][NetworkEvents.cpp:117] _checkForEvent(): Network Event: 112 - STA_CONNECTED
[  1752][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 112 - STA_CONNECTED
[  1762][V][STA.cpp:127] _onStaArduinoEvent(): Enabled IPv6 Link Local on sta
[  3244][V][NetworkInterface.cpp:78] _onIpEvent(): sta Got New IP: 192.168.1.15 MASK: 255.255.255.0 GW: 192.168.1.1
[  3255][V][NetworkEvents.cpp:117] _checkForEvent(): Network Event: 115 - STA_GOT_IP
[  3264][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 115 - STA_GOT_IP
[  3274][V][STA.cpp:171] _onStaArduinoEvent(): STA IP: 192.168.1.15, MASK: 255.255.255.0, GW: 192.168.1.1
[  3757][V][NetworkInterface.cpp:120] _onIpEvent(): IF sta Got IPv6: Interface: 0, IP Index: 0, Type: LINK_LOCAL, Zone: 2, Address: fe80:0000:0000:0000:3698:7aff:febb:b70c
[  3773][V][NetworkEvents.cpp:117] _checkForEvent(): Network Event: 116 - STA_GOT_IP6
[  3783][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 116 - STA_GOT_IP6
[  4756][V][NetworkInterface.cpp:120] _onIpEvent(): IF sta Got IPv6: Interface: 0, IP Index: 1, Type: GLOBAL, Zone: 0, Address: <REDACTED>
[  4772][V][NetworkEvents.cpp:117] _checkForEvent(): Network Event: 116 - STA_GOT_IP6
[  4781][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 116 - STA_GOT_IP6
[  5256][I][MatterColorLight.cpp:193] begin(): RGB Color Light created with endpoint_id 1
E (10657) chip[DMG]: Endpoint 0, Cluster 0x0000_0031 not found in IncreaseClusterDataVersion!
E (10658) chip[DMG]: Endpoint 0, Cluster 0x0000_0031 not found in IncreaseClusterDataVersion!
=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   276848 B ( 270.4 KB)
  Free Bytes        :   154372 B ( 150.8 KB)
  Allocated Bytes   :   107648 B ( 105.1 KB)
  Minimum Free Bytes:   151436 B ( 147.9 KB)
  Largest Free Block:   102388 B ( 100.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
     1 : UART_TX[0]
     3 : UART_RX[0]
============ After Setup End =============
[  6755][V][NetworkInterface.cpp:120] _onIpEvent(): IF sta Got IPv6: Interface: 0, IP Index: 0, Type: LINK_LOCAL, Zone: 2, Address: fe80:0000:0000:0000:3698:7aff:febb:b70c
[  6771][V][NetworkEvents.cpp:117] _checkForEvent(): Network Event: 116 - STA_GOT_IP6
[  6772][I][Matter.cpp:92] app_event_cb(): Interface IPV6 Address changed
[  6787][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 116 - STA_GOT_IP6
```
</details

theonlytechnohead, 2025-05-27:
<details>
<summary>a) Using Google Home app to toggle on/off:</summary

Logs:
```
[2556910][D][Matter.cpp:39] app_attribute_update_cb(): Attribute update callback: type: 0, endpoint: 1, cluster: 8, attribute: 0, val: 1
[2556922][V][Matter.cpp:44] app_attribute_update_cb(): Attribute update callback: PRE_UPDATE
[2556932][D][MatterColorLight.cpp:92] attributeChangeCB(): RGB Color Attr update callback: endpoint: 1, cluster: 8, attribute: 0, val: 1, type: 136
[2556947][D][MatterColorLight.cpp:115] attributeChangeCB(): RGB Color Light Brightness changed to 1
[2556959][D][Matter.cpp:39] app_attribute_update_cb(): Attribute update callback: type: 1, endpoint: 1, cluster: 8, attribute: 0, val: 1
[2556973][V][Matter.cpp:50] app_attribute_update_cb(): Attribute update callback: POST_UPDATE
[2556983][D][Matter.cpp:39] app_attribute_update_cb(): Attribute update callback: type: 0, endpoint: 1, cluster: 6, attribute: 0, val: 0
[2556997][V][Matter.cpp:44] app_attribute_update_cb(): Attribute update callback: PRE_UPDATE
[2557008][D][MatterColorLight.cpp:92] attributeChangeCB(): RGB Color Attr update callback: endpoint: 1, cluster: 6, attribute: 0, val: 0, type: 1
[2557023][D][MatterColorLight.cpp:101] attributeChangeCB(): RGB Color Light On/Off State changed to 0
[2557037][D][Matter.cpp:39] app_attribute_update_cb(): Attribute update callback: type: 1, endpoint: 1, cluster: 6, attribute: 0, val: 0
[2557049][V][Matter.cpp:50] app_attribute_update_cb(): Attribute update callback: POST_UPDATE
[2557061][D][Matter.cpp:39] app_attribute_update_cb(): Attribute update callback: type: 0, endpoint: 1, cluster: 8, attribute: 0, val: 254
[2557074][V][Matter.cpp:44] app_attribute_update_cb(): Attribute update callback: PRE_UPDATE
[2557084][D][MatterColorLight.cpp:92] attributeChangeCB(): RGB Color Attr update callback: endpoint: 1, cluster: 8, attribute: 0, val: 254, type: 136
[2557100][D][MatterColorLight.cpp:115] attributeChangeCB(): RGB Color Light Brightness changed to 254
[2557111][D][Matter.cpp:39] app_attribute_update_cb(): Attribute update callback: type: 1, endpoint: 1, cluster: 8, attribute: 0, val: 254
[2557125][V][Matter.cpp:50] app_attribute_update_cb(): Attribute update callback: POST_UPDATE
```

Outcome: light turns off, no errors seen
</details>

<details>
<summary>b) Using Google Home app to trigger a routine that toggles on/off:</summary>

Logs:
```
[2645785][D][Matter.cpp:39] app_attribute_update_cb(): Attribute update callback: type: 0, endpoint: 1, cluster: 6, attribute: 0, val: 1
[2645797][V][Matter.cpp:44] app_attribute_update_cb(): Attribute update callback: PRE_UPDATE
[2645807][D][MatterColorLight.cpp:92] attributeChangeCB(): RGB Color Attr update callback: endpoint: 1, cluster: 6, attribute: 0, val: 1, type: 1
[2645822][D][MatterColorLight.cpp:101] attributeChangeCB(): RGB Color Light On/Off State changed to 1
[2645837][D][Matter.cpp:39] app_attribute_update_cb(): Attribute update callback: type: 1, endpoint: 1, cluster: 6, attribute: 0, val: 1
[2645849][V][Matter.cpp:50] app_attribute_update_cb(): Attribute update callback: POST_UPDATE
E (2651186) chip[DMG]: Endpoint=1 Cluster=0x0000_0300 Command=0x0000_000A status 0x81 (no additional context)
```

Outcome: Light turns on, device logs an error, Google Home app notification says something went wrong with the device
</details>

<details>
<summary>c) Using Google Nest Mini voice commands to toggle on/off:</summary>

Logs:
```
[2744815][D][Matter.cpp:39] app_attribute_update_cb(): Attribute update callback: type: 0, endpoint: 1, cluster: 8, attribute: 0, val: 1
[2744827][V][Matter.cpp:44] app_attribute_update_cb(): Attribute update callback: PRE_UPDATE
[2744838][D][MatterColorLight.cpp:92] attributeChangeCB(): RGB Color Attr update callback: endpoint: 1, cluster: 8, attribute: 0, val: 1, type: 136
[2744853][D][MatterColorLight.cpp:115] attributeChangeCB(): RGB Color Light Brightness changed to 1
[2744864][D][Matter.cpp:39] app_attribute_update_cb(): Attribute update callback: type: 1, endpoint: 1, cluster: 8, attribute: 0, val: 1
[2744878][V][Matter.cpp:50] app_attribute_update_cb(): Attribute update callback: POST_UPDATE
[2744889][D][Matter.cpp:39] app_attribute_update_cb(): Attribute update callback: type: 0, endpoint: 1, cluster: 6, attribute: 0, val: 0
[2744903][V][Matter.cpp:44] app_attribute_update_cb(): Attribute update callback: PRE_UPDATE
[2744913][D][MatterColorLight.cpp:92] attributeChangeCB(): RGB Color Attr update callback: endpoint: 1, cluster: 6, attribute: 0, val: 0, type: 1
[2744928][D][MatterColorLight.cpp:101] attributeChangeCB(): RGB Color Light On/Off State changed to 0
[2744943][D][Matter.cpp:39] app_attribute_update_cb(): Attribute update callback: type: 1, endpoint: 1, cluster: 6, attribute: 0, val: 0
[2744955][V][Matter.cpp:50] app_attribute_update_cb(): Attribute update callback: POST_UPDATE
[2744967][D][Matter.cpp:39] app_attribute_update_cb(): Attribute update callback: type: 0, endpoint: 1, cluster: 8, attribute: 0, val: 254
[2744980][V][Matter.cpp:44] app_attribute_update_cb(): Attribute update callback: PRE_UPDATE
[2744990][D][MatterColorLight.cpp:92] attributeChangeCB(): RGB Color Attr update callback: endpoint: 1, cluster: 8, attribute: 0, val: 254, type: 136
[2745006][D][MatterColorLight.cpp:115] attributeChangeCB(): RGB Color Light Brightness changed to 254
[2745017][D][Matter.cpp:39] app_attribute_update_cb(): Attribute update callback: type: 1, endpoint: 1, cluster: 8, attribute: 0, val: 254
[... cut ...]
[3047994][D][MatterColorLight.cpp:92] attributeChangeCB(): RGB Color Attr update callback: endpoint: 1, cluster: 8, attribute: 0, val: 254, type: 136
[3048010][D][MatterColorLight.cpp:115] attributeChangeCB(): RGB Color Light Brightness changed to 254
[3048021][D][Matter.cpp:39] app_attribute_update_cb(): Attribute update callback: type: 1, endpoint: 1, cluster: 8, attribute: 0, val: 254
[3048035][V][Matter.cpp:50] app_attribute_update_cb(): Attribute update callback: POST_UPDATE
```

Outcome: light turns off, no errors seen
</details>

SuGlider, 2026-09-12:
@theonlytechnohead - it seems that the sketch should use `MatterEnhancedColorLight ColorLight;` instead of  `MatterColorLight ColorLight;` in order to solve the `E (446700) chip[DMG]: Endpoint=1 Cluster=0x0000_0300 Command=0x0000_000A status 0x81 (no additional context)` issue and return success to Google. This would solve the on/off routine that you have reported.

Please try it and let me know. Thanks!

theonlytechnohead, 2026-10-16:
Thanks for the update, I've updated my code accordingly.

Using `3.3.10` I still get:
`E (7838) chip[DMG]: Endpoint 0, Cluster 0x0000_0031 not found in IncreaseClusterDataVersion!`

Using `3.3.11` and `3.3.12` it's gone.

Thank you so much!

closed 2026-10-16 by theonlytechnohead