espressif/arduino-esp32 #11381: Zigbee end device "randomly" factory resets; seems due to handling of ESP_ZB_ZDO_SIGNAL_LEAVE

tofurky, 2025-05-21:
### Board

dfrobot_beetle_esp32c6

### Device Description

Small esp32c6 board powered via USB-C

### Hardware Configuration

I2C pins are wired up to DF Robot O2 sensor; powered via USB-C port

### Version

v3.2.0

### IDE Name

arduino-cli

### Operating System

Linux

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

460800

### Description

The device is on a fairly busy Zigbee network with tens of bulbs from different vendors including Ikea, Cree, Sylvania/OSRAM. Using Homeassistant 2025.4 with ZHA on a SkyConnect / ZBT-1 dongle (silabs stack).

Somewhat randomly, it seems another device on the network (Ikea bulbs, I believe - I actually had a capture of them doing the same to some Lutron remotes) won't like something the ESP32C6 does and tells it to "leave" the network - generating a ESP_ZB_ZDO_SIGNAL_LEAVE signal. It seems it happens when the device sends a reading; some other device didn't like it. Last set of logs attached below shows it was in the middle of updating the clusters.

The result due to the default handler of this in ZigbeeCore.cpp is that the device will factory reset itself and need to be manually rejoined to the network.

Relevant code: https://github.com/espressif/arduino-esp32/blob/13cd0d3c3fb6d6f719c55cdb671d6807874427b5/libraries/Zigbee/src/ZigbeeCore.cpp#L373-L379

Per docs, there is an enum that can be checked to determine which type of reset this is. I have a feeling that the proper handling of these messages coming from other nodes on the network (i.e. not the coordinator) should just be to reset the zigbee stack or reset the device, without fully wiping the stored config and requiring it to be manually re-joined to the network.

### Sketch

```cpp
Basically, Zigbee_Temperature_Sensor.ino. Can post full sketch if needed, but I believe this issue is due to the catch-all handling of ESP_ZB_ZDO_SIGNAL_LEAVE signals in ZigbeeCore.cpp
```

### Debug Message

```plain
Here are older logs I left in a comment a while ago (https://github.com/espressif/arduino-esp32/pull/10676#issuecomment-2564487117)

It's happening as well on 3.2.0, I just didn't have the device running with verbose logs this time.

[26523678][V][ZigbeeTempSensor.cpp:112] setHumidity(): Updating humidity sensor value...
[26523678][D][ZigbeeTempSensor.cpp:114] setHumidity(): Setting humidity to 2090
[26523679][V][ZigbeeTempSensor.cpp:71] setTemperature(): Updating temperature sensor value...
[26523679][D][ZigbeeTempSensor.cpp:73] setTemperature(): Setting temperature to 2430
[26524179][I][o2.ino:140] zigbee_reading_handler(): Sent average of readings of oxygen=20.9%, temperature=24.3 C (60 total readings)
[26524677][D][o2.ino:57] sensor_read(): oxygen=20.9% temperature=24.3 C (2518 us)
[26525677][D][o2.ino:57] sensor_read(): oxygen=20.9% temperature=24.4 C (2416 us)
[26526677][D][o2.ino:57] sensor_read(): oxygen=20.9% temperature=24.3 C (2485 us)
[26526711][I][ZigbeeCore.cpp:225] esp_zb_app_signal_handler(): Device started up in non factory-reset mode
[... 120 lines cut ...]
[   339][D][ZigbeeCore.cpp:152] zigbeeInit(): Register all Zigbee EPs in list
[   341][I][ZigbeeCore.cpp:160] zigbeeInit(): List of registered Zigbee EPs:
[   341][I][ZigbeeCore.cpp:162] zigbeeInit(): Device type: Temperature Sensor, Endpoint: 10, Device ID: 0x0302
[   345][I][ZigbeeCore.cpp:241] esp_zb_app_signal_handler(): Zigbee stack initialized
[   346][D][ZigbeeCore.cpp:242] esp_zb_app_signal_handler(): Zigbee channel mask: 0x07fff800
[   347][I][ZigbeeCore.cpp:248] esp_zb_app_signal_handler(): Device started up in  factory-reset mode
[   348][I][ZigbeeCore.cpp:255] esp_zb_app_signal_handler(): Start network steering
[   350][I][o2.ino:275] setup_reporting(): Setting reporting intervals
[   353][I][o2.ino:311] setup_clusters(): Setting initial cluster values
[   356][D][ZigbeeTempSensor.cpp:134] setHumidity(): Setting humidity to 2070
[   357][D][ZigbeeTempSensor.cpp:81] setTemperature(): Setting temperature to 2250
[   359][I][o2.ino:339] connect_zigbee(): Waiting for Zigbee to connect...
[  2359][I][o2.ino:339] connect_zigbee(): Waiting for Zigbee to connect...
[  2577][I][ZigbeeCore.cpp:317] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)
[  4360][I][o2.ino:339] connect_zigbee(): Waiting for Zigbee to connect...
[  5811][I][ZigbeeCore.cpp:317] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)
[  6362][I][o2.ino:339] connect_zigbee(): Waiting for Zigbee to connect...
[  8363][I][o2.ino:339] connect_zigbee(): Waiting for Zigbee to connect...
[  9042][I][ZigbeeCore.cpp:317] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)
[ 10363][I][o2.ino:339] connect_zigbee(): Waiting for Zigbee to connect...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2025-05-21:
Hi @tofurky, Thank you for reporting this issue. 
I will open a PR today with a proper leave signal handling, which will depend on a leave type.
Leave without rejoin (will factory reset) or Leave and rejoin (only reboot ESP).

tofurky, 2025-05-21:
> Hi [@tofurky](https://github.com/tofurky), Thank you for reporting this issue. I will open a PR today with a proper leave signal handling, which will depend on a leave type. Leave without rejoin (will factory reset) or Leave and rejoin (only reboot ESP).

Great, thanks for the quick response. I can test it locally, but I unfortunately don't have a mechanism to trigger the leave; it sometimes takes a week or 2.

tofurky, 2025-05-21:
I've reflashed using the ZigbeeCore.cpp changes from #11385.

It'll be a bit more difficult to tell when it happens now, as it was quite obvious when the device was factory reset, but I'm saving serial output to a .txt file so I can grep it for that other (new) case.