espressif/arduino-esp32 #11383: OV2640 Camera Not Detected on ESP32-S3 WROOM — Error 0x105 (ESP_ERR_NOT_FOUND)

Sujjiiiya, 2025-05-21:
### Related area

Board support, Camera peripheral (esp_camera), I2C/SCCB initialization

### Hardware specification

ESP32-S3 WROOM (N16R8), OV2640 camera module

### Is your feature request related to a problem?

Yes. I am unable to get the OV2640 camera working on my ESP32-S3 WROOM board using Arduino IDE (2.3.6).

I tested this on **two different ESP32-S3 WROOM (N16R8)** boards and a  **OV2640** camera module. I am using the `esp_camera` library and the I do not know if correct GPIO configuration 
Despite this, I consistently get the following error in the Serial Monitor:
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2a (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x118c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xc20
load:0x403cb700,len:0x30e0
entry 0x403c88b8
E (207) esp_core_dump_flash: Core dump data check failed:
Calculated checksum='995aede6'
Image checksum='ffffffff'
E (17) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
E (38) gdma: gdma_disconnect(319): no peripheral is connected to the channel

This is my code:
### 🔧 Minimal Reproducible Code

```cpp
#include "esp_camera.h"
#include <WiFi.h>

// === Camera Pin Configuration (for ESP32-S3 WROOM) ===
#define PWDN_GPIO_NUM     -1
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM     10
#define SIOD_GPIO_NUM     40
#define SIOC_GPIO_NUM     39
#define Y9_GPIO_NUM       48
#define Y8_GPIO_NUM       11
#define Y7_GPIO_NUM       12
#define Y6_GPIO_NUM       14
#define Y5_GPIO_NUM       16
#define Y4_GPIO_NUM       18
#define Y3_GPIO_NUM       17
#define Y2_GPIO_NUM       15
#define VSYNC_GPIO_NUM    38
#define HREF_GPIO_NUM     47
#define PCLK_GPIO_NUM     13

const char* ssid = "iptime";
const char* password = "";

void startCameraServer(); // Placeholder

void setup() {
  Serial.begin(115200);
  delay(1000);
  Serial.println("📷 Initializing camera...");

  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sccb_sda = SIOD_GPIO_NUM;
  config.pin_sccb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.pixel_format = PIXFORMAT_JPEG;
  config.frame_size = FRAMESIZE_QQVGA;
  config.fb_location = CAMERA_FB_IN_DRAM;
  config.jpeg_quality = 12;
  config.fb_count = 1;
  config.grab_mode = CAMERA_GRAB_WHEN_EMPTY;

  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf(" Camera init failed with error 0x%x\n", err);
  } else {
    Serial.println("Camera initialized successfully");
  }
}

void loop() {
  delay(10000);
}

### Describe the solution you'd like

I would like esp_camera or related drivers to reliably detect OV2640 camera modules on ESP32-S3 WROOM boards using the standard GPIO pin mapping, even when PSRAM is disabled.

If the issue lies in internal I2C pull-ups, SCCB timings, or clock configurations specific to ESP32-S3 WROOM, I would appreciate a code-level patch or configuration suggestion to fix the camera detection issue and prevent ESP_ERR_NOT_FOUND (0x105).

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2025-05-21:
SIOC and SIOD must be wrong for the camera to not be detected. Check your pins and hardware connection (power to the module)

closed 2025-05-21 by me-no-dev

Sujjiiiya, 2025-05-21:
@me-no-dev ,Thanks for the suggestion!

I tried swapping the SIOC and SIOD pins to several alternatives (GPIOs 26/27 and also 1/2), but unfortunately the ESP32-S3 either entered a reboot loop (`rst:0x8 (TG1WDT_SYS_RST)`) or still gave me the same `ESP_ERR_NOT_FOUND` error.

I've reverted to the originally used GPIOs:
```cpp
#define SIOD_GPIO_NUM 40  // SDA
#define SIOC_GPIO_NUM 39  // SCL

Sujjiiiya, 2025-05-21:
These seem to be the most commonly documented pins for ESP32-S3 camera boards. However, the camera is still not detected on two different ESP32-S3 WROOM boards, both using a verified OV2640 module.

If anyone has verified working GPIO mappings for the OV2640 with ESP32-S3 WROOM (N16R8) using the camera ribbon connector, I’d really appreciate it if you could share them.

I’m happy to test further — just trying to narrow down whether this is a board-specific routing issue or something else in the ESP32-S3 + OV2640 flow. Thanks again!

Sujjiiiya, 2025-05-22:
Hi again!

Just to be 100% sure, I’m attaching a picture of my board with the OV2640 camera ribbon inserted. The ribbon does not go more in than this TT

Could someone please take a look and let me know if the ribbon is inserted properly? If you notice anything wrong with the orientation or connection, I’d really appreciate your advice.

Thanks in advance!

[image]

Sujjiiiya, 2025-05-23:
Final Update — RESOLVED

Hi everyone! Just wanted to share that my issue is finally resolved

After trying many things (pin remapping, Wi-Fi switching, PSRAM tweaks, minimal test sketches, etc.), the real fix turned out to be:

Changing the "Upload Mode" setting in Arduino IDE to USB OTG CDC / TinyUSB

Previously, I was using UART / Hardware CDC and even though uploads were successful, the code wasn't running afterward — only the ESP-ROM bootloader logs appeared. No camera output, no Wi-Fi connection, and no debug prints. After switching to USB OTG CDC, everything started working instantly!

Big thanks to everyone who gave suggestions — your help really kept me going :)