espressif/arduino-esp32 #11386: Cannot Flash ESP32 C5-Devkit

weoiss1998, 2025-05-21:
### Board

ESP32 C5 Devkit

### Device Description

ESP32 C5 Wroom1

### Hardware Configuration

equal

### Version

other

### IDE Name

Arduino IDE

### Operating System

Win 11

### Flash frequency

40&80

### PSRAM enabled

no

### Upload speed

115200, 921600

### Description

I cannot flash a C5 module with the development version.
I tried different C5 and different USB-C cable.
I tried UART and USB aswell

### Sketch

```cpp
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
Der Sketch verwendet 272183 Bytes (20%) des Programmspeicherplatzes. Das Maximum sind 1310720 Bytes.
Globale Variablen verwenden 15616 Bytes (4%) des dynamischen Speichers, 312064 Bytes für lokale Variablen verbleiben. Das Maximum sind 327680 Bytes.
esptool.py v4.8.1
Serial port COM7
Connecting....
Chip is ESP32-C5 (revision v1.0)
Features: WiFi 6, BT 5, IEEE802.15.4
Crystal is 48MHz
MAC: 30:ed:a0:ff:fe:e4:21:18
BASE MAC: 30:ed:a0:e4:21:18
MAC_EXT: ff:fe
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Erasing flash (this may take a while)...
Chip erase completed successfully in 1.5 seconds.
Compressed 19808 bytes to 12485...

A fatal error occurred: Packet content transfer stopped (received 4072 bytes)
Fehlgeschlagenes Hochladen: Hochladefehler: exit status 2
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-05-21:
You probably have the ECO1 version of C5 and you can use that ONLY with 3.3.0-alpha1 version installed through the board manager. If you have ECO2 chip somehow, we do not officially support it yet.

weoiss1998, 2025-05-21:
I use the 3.3.0-alpha1. I don't know if I have ECO1 or ECO2. It's the devkit which is sold by Espressif

justcallmekoko, 2025-05-21:
I am also getting the same error using the dev kit sold by espressif on [aliexpress](https://www.aliexpress.us/item/3256808604473710.html?spm=a2g0o.order_list.order_list_main.5.c5521802CvtioE&gatewayAdapt=glo2usa#nav-specification). I am looking for markings to indicate which version of the C5 it is but I have found nothing to indicate ECO1 or ECO2. The devkit board version is `ESP32-C5-DevKitC-1 V1.2`

Nigdzie, 2025-05-21:
Its is ECO2, we can see this in ESP-IDF menuconfig

[image]

Lucia-Rubio-E, 2025-05-22:
@weoiss1998 @me-no-dev @justcallmekoko @Nigdzie @kumekay 

I'm currently working on a project that requires Wi-Fi FTM, so I need to use an ESP32-C5 chip or module with ECO2 (revision v1.2) silicon.

Does anyone know where I can buy a confirmed ECO2/v1.2 version, either as a DevKit or a standalone chip/module?
Official distributors, AliExpress, or links from Espressif would all be very helpful.

Thanks in advance!

weoiss1998, 2025-05-22:
> I am also getting the same error using the dev kit sold by espressif on [...]

@Lucia-Rubio-E here you have the link :)

Nigdzie, 2025-05-22:
> [@weoiss1998](https://github.com/weoiss1998) [@me-no-dev](https://github.com/me-no-dev) [@justcallmekoko](https://github.com/justcallmekoko) [@Nigdzie](https://github.com/Nigdzie) [...]

https://www.aliexpress.com/item/1005008790788462.html

Nigdzie, 2025-05-22:
> I imagine these ECO2 revision chips are going to be the mainstream considering they were created to add stability fixes over the previous chip version.

well ... i was able to flash it ;)
now testing if we can go forward

`ESP-ROM:esp32c5-eco2-20250121
Build:Jan 21 2025
rst:0x8 (TG1_WDT_HPSYS),boot:0x18 (SPI_FAST_FLASH_BOOT)
Core0 Saved PC:0x40801f1e
SPI mode:DIO, clock div:1
load:0x40855ab0,len:0x1274
load:0x4084bfa0,len:0xa88
load:0x4084e9a0,len:0x3004
entry 0x4084bfa0`

`esptool.py v4.8.1
Serial port COM4
Connecting....
Chip is ESP32-C5 (revision v1.0)
Features: WiFi 6, BT 5, IEEE802.15.4
Crystal is 48MHz
MAC: xxx
BASE MAC: xxx
MAC_EXT: ff:fe
ROM expects crystal freq: 48 MHz, detected 48 MHz
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00002000 to 0x00006fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00010000 to 0x0004bfff...
Erasing flash...
Compressed 19808 bytes to 12485...
Took 0.09s to erase flash block
Wrote 19808 bytes (12485 compressed) at 0x00002000 in 0.4 seconds (effective 361.5 kbit/s)...
Hash of data verified.
Erasing flash...
Compressed 3072 bytes to 146...
Took 0.02s to erase flash block
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.0 seconds (effective 1156.7 kbit/s)...
Hash of data verified.
Erasing flash...
Compressed 244528 bytes to 137042...
Took 0.60s to erase flash block
Wrote 244528 bytes (137042 compressed) at 0x00010000 in 4.9 seconds (effective 400.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...`

Nigdzie, 2025-05-22:
[image]

> [@Nigdzie](https://github.com/Nigdzie) What were you upload settings/args?

polishing gimme sec :)

Nigdzie, 2025-05-22:
> [@Nigdzie](https://github.com/Nigdzie) What were you upload settings/args?

and no luck mate ... bootloops bootloops only bootloops

here we have retro what was done and why ...
https://github.com/Nigdzie/WiFiP0rN

@me-no-dev i am wondering if we rlly could reopen this case or make a new one ?

Pybite, 2025-05-26:
to find if you have eco1 or 2 just put the device into download mode and then idf.py monitor

AZKKXL, 2025-07-07:
@weoiss1998 
Hello, I'd like to know how you solved this problem

loraadh, 2025-07-10:
Hello,
I used to flash esp32-C6 via VSC without any issues.
With ESP32-C5 devkit-1-V1.2 ECO2, however, I am not able to flash it. 
VCS -> show examples -> iperf shows that C5 targets are still not listed among the list of supported targets (picture below). 

Is it going to be supported soon?
Is there a workaround that allows me to flash the iperf project to my C5?

[image]

AZKKXL, 2025-07-14:
@Pybite 

Thank you for your reply. I have a small doubt, and I'm not sure if I understand it correctly: You mentioned using espidf/release v5.5, and during my testing, the program works normally on the ESP32C5 EOC2, and this is not a problem. However, when using ArduinoIDE to compile and upload, I still encounter the error "A fatal error occurred: Packet content transfer stopped (received 4072 bytes)." Does this mean I can only compile the program using the ESPIDF framework to make it work, or is it possible to modify some toolchain settings in ArduinoIDE so that the program compiled by ArduinoIDE can be uploaded to the ESP32C5 ECO2 development board?

Thank you for your reply, as using ESPIDF allows the program to run correctly.

Pybite, 2025-07-14:
@AZKKXL 

long story short, ArduinoIde ( ArduinoCore ) isnt updated to flash firmware on the newest SoCs as yet ( C5, C61 ).

in order to flash any program to those chips, you will need to use the esp idf framework +v5.5

weoiss1998, 2025-07-14:
Thanks for the Update @me-no-dev !